fairecasoimeme/ZiGate #1: Integration with Home Assistant

ruimarinho, 2017-11-19:
Hello @fairecasoimeme.

First of all, great work on ZiGate! I'm very excited to finally try out ZigBee devices without more and more cloud gateways.

Have you considered integration with the awesome open source automation platform [Home Assistant](https://home-assistant.io)? I see a lot of people looking to replace their Xiaomi Gateway on the community forums, so I can imagine this integration would give ZiGate a big boost.

KiwiHC16, 2017-12-07:
ZiGate interface is public and automation platform could use it to do the integration.

ruimarinho, 2017-12-18:
It would be great to have a custom component for ZiGate. I'm very excited about its possibilities :)

elric91, 2017-12-25:
I started a component here : https://github.com/elric91/hass_custom_components/blob/master/zigate.py
(python test code here : https://github.com/elric91/ZiGate)

loicortola, 2017-12-26:
Hello guys, how is this component going?
Need a hand?
I'm working on a java client to facilitate setup and pairing for now

elric91, 2017-12-26:
Hi
I just started to code it yesterday ... For the time being, it is "only" able to connect HASS to the Zigate through usb/serial and send / receive message in async mode. I plan to work next on HASS GUI integration to ease with pairing / monitoring without having to rely only on server logs 
I started this having in mind a community / collaborative approach, so any contribution / help / advice is welcome 
At the same time, we can share our experience with Akila an help improving technical documentation

NB : the python code already includes some decoding functions that are not yet in the HASS component

closed 2018-01-30 by fairecasoimeme