fairecasoimeme/ZiGate #2: Code de la zigate (firmware nxp5168)

gbrd, 2017-11-20:
Hi, 
Are there any plans to share the firmware code of zigate in this repository  ? (firmware for nxp 5168)

loicortola, 2017-12-25:
+1 . I am willing to contribute and spend some time on improving some Zigate code as long as it is OpenSource :)

gbrd, 2018-01-08:
no answer ? :)

fairecasoimeme, 2018-01-30:
I'm waiting to deliver a good version with some clean debug
It's for soon

lolorc, 2018-03-26:
Any news ?
by the way, why would you close issues before fixing them ? ;-)

intra-au, 2018-04-08:
I have few spare nxp5169 modules and would like to contribute too, however with 5169 binary not released and 5168 source not provided I'll probably end up making my own firmware from scratch using your protocol for compatibility with other modules

lolorc, 2018-04-12:
@intra-au @fairecasoimeme 
might be useful to re-open this issue or create another one.
That would be sad and a waste of time to start from scratch.

intra-au, 2018-04-19:
@fairecasoimeme any chance it can be done soon?

fairecasoimeme, 2018-04-19:
yep, soon... with the new firmware 3.0e

pdecat, 2018-06-09:
Any idea of the license terms under which the source code will be distributed?

pipiche38, 2020-09-30:
Suggest to close that one.
NXP code is provided as per NXP Copyright

Proper LIcense will be specify for this project

closed 2020-09-30 by fairecasoimeme