fairecasoimeme/ZiGate #4: More clarifications on the protocol

loicortola, 2017-12-25:
Bonjour,

bravo pour la Zigate, je suis en train de vous faire un joli GUI en JavaFX multi-plateforme pour ne plus avoir à passer par l'outil un peu dégueu de NXP, mais j'ai besoin de précisions sur le protocole, plutôt que de chercher à la main :

 * Ai-je bien toujours 6 bytes de headers (1b-start, 2b-msg-code, 2b-len, 1b-chk) ?
 * Quelle valeur peut prendre le RSSI? A quoi est-ce que cela correspond? (en général c'est une valeur en Db négative, donc difficile de voir)
 * Je n'ai pas saisi votre règle sur le transcodage, et typiquement si c'était sur les headers, sur le message, ou sur toute la trame?
 * Je n'ai pas saisi non plus si les headers de length prenaient en compte le transcodage de la trame ou non?
 * Sinon, je crois avoir vu sur les exemples mais pour confirmer, on est bien en little-endian pour les valeurs sur 2-bytes?

Merci de clarifier un peu la spec, ce serait top ^^.
En attendant je mets les travaux en pause pour m'amuser sur un firmware custom 5168

Loïc

fairecasoimeme, 2017-12-26:
Hi,
The encoding data way is only for 0x00 to 0x09 bytes. It's used to escape data which is not a command.
For example, the byte 0x05 is encoding : 0x02 0x15
the byte 0x01 is encoding :0x02 0x11

Concerning RSSI, if the packet come from a device the byte before the end of packet is the RSSI. The value of RSSI (in this case) could be between 0x01 and 0xff

When packet come from ZiGate the RSSI value is 0x00.

You have to check the [example in PHP](http://zigate.fr/wiki/exemple-api-en-php/), there are more informations.

I hope you find what you search.

Best regards.
Fairesoimeme

loicortola, 2017-12-26:
Hello,
Thanks for the response! I will update the message accordingly.

One last question : 
In my second example, I see no RSSI byte though... Or is RSSI part of the "data length"?
Is the checksum also transcoded?

fairecasoimeme, 2017-12-26:
Sorry I re read my spec and the encoding byte is from 0x00 to 0x0f, you're right.

RSSI byte is in data. So RSSI is part of the "data lenght". In your example RSSI value is 0x00. (just before 0x03)
The checksum is also transcoded

closed 2018-01-08 by loicortola

KiwiHC16, 2018-01-26:
Concerning RSSI, if the packet come from a device the byte before the end of packet is the RSSI. The value of RSSI (in this case) could be between 0x01 and 0xff

Is a field that you added in your firmware version as I don't see it in the NXP initial version ?
What is this RSSI ? Is it measured at equipment level ? Quid if the message is routed between the zigate and the equipment ?
Any docs on how this is measured and where ?

fairecasoimeme, 2018-01-26:
Yep I added this feature because it's usefull. This is not in the NXP initial version. The RSSI determine the reception power between ZiGate and device. When device is routed, you've got the RSSI of the last router. You can get the value of the RSSI between device and router with a command  LQI req (0x004E)

No docs. I noticed only that the quality is between 1 and 255

KiwiHC16, 2018-01-26:
Ok this is in line with my understanding ;-)
I'm lucky that you put it at the end as it has visibly no impact on my plugin.
Now that I have a alpha which ork I will look at the LQI info and si what I can do. I have some side script to draw a graph, but let see....