fairecasoimeme/ZiGate #6: Feedback with IKEA Tradfri remote, possible?

tjikkun, 2017-12-28:
I wrote (still work in progress) my own Zigate Python library and a plugin for Domoticz and I have various devices working, also the Ikea bulb. Now I can switch the bulb on/off with both Domoticz and the remote. However, when I use the remote, the status in Domoticz is not updated, no message is received from the Zigate. Is it possible to add this?

elric91, 2018-01-20:
The way I do it from home-assistant is pairing the remote with the ZiGate (not with the device to be switched) and the device as well. Home-assistant detects when I press the switch and fire an event to turn the device on or off

KiwiHC16, 2018-02-05:
"Home-assistant detects when I press the switch" 
Could you give a bit more information ? What is this "switch" ? A button on the IKEA remote ?
If HA detect, which message is it receiving ?

elric91, 2018-02-05:
Sorry : I should have written "the way I __would__ do it" as I don't own a TradFri remote yet.
Pairing the switch (indeed, remote button) to the Zigate should at least log info, but reading http://zigate.fr/ikea-tradfri-telecommande-e1524/ it is unclear what message would be published

(from HA, if you manage to pair the remote with the ZiGate, you can have a look at the logs to see what message - if any - can be used to detect press)

KiwiHC16, 2018-02-05:
I can tell you that IKEA remote doesn't send anything to zigate after pairing (no traffic on zigbee). So zigate can't send anything to HA. As tijkkun mentionned nothing comes from remote and as IKEA bulb doesn't report by default, zigate don't receive anything neither HA.
The only solution would be to do some specific configuration like the IKEA GW does, but it's not clear yet how to do so.

pipiche38, 2019-01-21:
Dont' know if the pre-release available on github https://github.com/fairecasoimeme/ZiGate/releases/tag/v3.0f-PR is able to handle the Ikea Remote , but Zigate crash with ZGUI when doing a Get Group Membership

To duplicate use ZGUI and do get Group with

Address: 0x0000
Source Ep: 01
Dest Ep: 01
Group Count: 0x00
Group : 0x0000

closed 2019-03-04 by ISO-B