fairecasoimeme/ZiGate #10: Problème avec le firmware 3.0b

dams74, 2018-01-21:
Bonjour
Pour commencer merci pour le travail effectué.
Lors de la mise à jour du firmware 3.0b, j'ai le message suivant: the firmware file was built for another processor version and might not run on the connected device.
Voir capure ci dessous.
[image]
J'ai essayé de mettre les firmware 3.0 et 3.0a et tout passe sans message d'erreur ou d'avertissement.
Damien

ioull, 2018-01-21:
Tu as essayé de re télécharger le firmware 3.0b ? J’ai eu le même soucis et ça venait de mon fichier corrompu.

dams74, 2018-01-21:
En effet le problème viens bien de mon fichier...
Pourquoi n'ai-je pas pensé à le télécharger de nouveau...
Merci bien en tout cas.
Damien

closed 2018-01-21 by dams74