fairecasoimeme/ZiGate #13: ZiGate should be transparent to Xiaomi messages 0x115F (see mail)

KiwiHC16, 2018-01-22:


KiwiHC16, 2018-01-23:
- Xiaomi Temperature Carre: zigate ne transmet pas le pacquet propriétaire de donc nous ne pouvons recevoir le niveau de batterie (Volt).

zaraki673, 2018-01-23:
Bonjour,
 j'ai mis a jour mon firmware hier sur la 3.0b, pas de soucis de se genre de mon coté, 
le niveau de batterie est bien remonté sur mes capteur de temperatures (carré et rond)
pas sur les autres, mais je ne me souviens pas les avoir déja vu (ou pas vu et du coup pas codé dans mon plugin domoticz pour le coup)

elric91, 2018-01-23:
Je confirme qu'il est également présent avec le firmware 2.x. Par contre, il n'est pas envoyé souvent

KiwiHC16, 2018-01-23:
L'info est envoyée dans 2 types de messages. Un passe la Zigate et pas l autre. J ai envoyé à Akila une proposition de correctif.

fairecasoimeme, 2018-01-30:
Pending for the next version

closed 2018-01-30 by fairecasoimeme