fairecasoimeme/ZiGate #14: ZiGate command Obj-> ZiGate	0x8015

KiwiHC16, 2018-01-25:
Answer 0x8015 to command 0x0015 reply with one zero too many.

Spec: http://zigate.fr/wiki/commandes-zigate/
Example: 
// <device list – data each entry is 13 bytes>
                        // <ID: uint8_t>
                        // <Short address: uint16_t>
                        // <IEEE address: uint64_t>
                        // <Power source: bool_t> 0 – battery 1- AC power
                        // <LinkQuality : uint8_t> 1-255
                        
                        // Id Short IEEE             Power  LQ
                        // 00 ffe1  00158d0001d5c421 00     a6
                        // 01 c4bf  00158d0001215781 00     aa
                        // 02 4f34  00158d00016d8d4f 00     b5
                        // 03 304a  00158d0001a66ca3 00     a4
                        // 04 cc0D  00158d0001d6c177 00     b3
                        // 05 3c58  00158d00019f9199 00     9a
                        // 06 7c3b  000B57fffe2c82e9 00     bb
                        // 07 7c54  00158d000183afeb 01     c3
                        // 08 3db8  00158d000183af7b 01     c5
                        // 32 553c  000B57fffe3025ad 01     9f
                        // 00 -> Pourquoi 00 ?

Is it an issue on my side or ZiGate send this 00 too many ?

tjikkun, 2018-01-25:
It is the RSSI, see http://zigate.fr/wiki/protocole-zigate/

KiwiHC16, 2018-01-26:
@Akila si tu passes par la, je ne comprends pas car si @tjikkun dit vrai un RSI de 00 n'est pas terrible et ta doc n'est pas en ligne avec la page 22 du doc JN-AN-1216. Aurais tu fais une modification ?

doudz, 2018-01-26:
Comme indiqué ici, c'est effectivement le RSSI
https://github.com/fairecasoimeme/ZiGate/issues/4#issuecomment-353952273

fairecasoimeme, 2018-01-30:
Indeed, this is the RSSI. When it's 00, it's not the real RSSI because the message come from ZiGate and not from radio devices.
I add link quality in data message because I find this usefull ;)

closed 2018-01-30 by fairecasoimeme