fairecasoimeme/ZiGate #15: Command 0x0024 doesn't reply with response 0x8024

doudz, 2018-01-30:
Sending command 0x0024 to start network doesn't reply with response 0x8024
According to doc, it should
----
La commande 0x0024 "Start Network" ne provoque pas la réponse 0x8024 par conséquent on ne sait pas si le réseau est démarré ou non
Il n'y a pas de commande permettant de vérifier cet état (mais ce n'est peut-être pas utile)

fairecasoimeme, 2018-02-03:
Indeed, the reply 0x8024 seems to come only when the network is starting from an erase PDM. A reset command open the network by default. No 0x8024 response in this case.
I will see but I don't know if it's an important issue to be usefull.

fairecasoimeme, 2018-02-17:
fixed with next version 3.0d

closed 2018-02-17 by fairecasoimeme

doudz, 2018-02-17:
it doesn't seem fix in 3.0d, no response 0x8024 after command 0x0024
and I don't really know how understand response 0x8009 (the new command 0x0009 ) since there's no status information in response

fairecasoimeme, 2018-02-19:
Sorry, I didn't fix really this issue but i thinked that with this new command, you can get the network state.
There is no status but if PAN ID = 0 the network is down else the network is up

doudz, 2018-02-19:
ok, thanks for explanation

pipiche38, 2018-11-06:
Could you kindly confirm that Short address and Extended Address are refered to the Zigate (Controller) address ?

  | <Short Address: uint16_t> |  
-- | -- | --
  |   |   | <Extended Address: uint64_t>