fairecasoimeme/ZiGate #16: set command equivalent to 0x015

handfreezer, 2018-01-30:
Hello,

  Is it planned to implement a set command in consequence of 0x015 command?
  The goal is simple : being able to export associated devices with zigate, in order to be able to restore it later (after some test for example, or in case of replacement if zigate gone broken and replaced)

note : what you've done in this project is really famous! Thanks

Best regards

elric91, 2018-01-30:
(related question : do you loose associations when you upgrade firmware ?)

handfreezer, 2018-01-30:
Not yet , but:
*it could happen or needed maybe with a change in firmware in futur, this feature could be nice,
*it is necessary to be able to do a backup in case of broken zigate, to change it easily without rejoining every sensors (which go with associating them with good name in your domotic software, a relay udge job when you have more than few sensors)
*could be a cool feature if you plan to split your first conf in two or more conf.

No?

fairecasoimeme, 2018-02-01:
Hi,
indeed, the feature is interesting but there is a problem. When you add a device to ZiGate there is a negociation between the two entities. The goal is to ensure the confidentiality and integrity. So, the negociation give a "network key" (stored in device) which is calculate from the ZiGate PAN (when it create the network), Link key and others stuffs ... .

The only solution would be to be able to save / load full EEPROM but I can't ensure that will be ok. I will try to see if it's possible but not right now

KiwiHC16, 2018-02-01:
I have also this in my todo including the IEEE address

handfreezer, 2018-02-02:
I understand, even if I think it is a major feature for testers to be confident during tests to go back to a working situation.

handfreezer, 2018-02-02:
it is in your plan if possible, so I guess this issue can be closed.

closed 2018-02-02 by handfreezer