fairecasoimeme/ZiGate #18: Wrong cluster count

tjikkun, 2018-02-06:
When my Ikea Tradfri remote sends a Simple Descriptor Response (0x8043) the length inside the data and cluster_out count seem incorrect:
Received data:
    01 80 43 02 10 26 8a 10 02 10 27 02 1a 22 02 11 02 11 02 14 02 18 10 02 12 02 16 02 10 02 10 02 10 02 11 02 10 02 13 02 10 02 19 02 1b 02 15 10 02 10 02 17 02 10 02 10 02 10 02 11 02 10 02 13 02 10 02 19 02 1b 02 15 10 02 10 6f 03
Decoded:
01804300268a1000270a220101040810020600000001000300090b0510000700000001000300090b0510006f03

01 < start
8043 < msg type
0026 < length
8a < chksum

DATA:

10 < seq nr
00 < status
270a < nwk_addr
22 < length  (=34), but only 32 follow below
01 < endpoint
0104 < profile
0810 < device id
02 < bit fields
06 < in count
0000 0001 0003 0009 0b05 1000 < 6 clusters
07 < out count
0000 0001 0003 0009 0b05 1000 < ************** 6 clusters, should be 7? ****************

6f < RSSI
03 < END

KiwiHC16, 2018-02-06:
I found a similar issue yesterday. Haven't investigated more yet.

tjikkun, 2018-02-07:
@fairecasoimeme I found the issue, it is already in JN-AN-1216-Zigbee-3-0-IoT-ControlBridge:
The patch is:
```diff
--- Source/ZigbeeNodeControlBridge/app_general_events_handler.c.orig	2018-02-07 10:04:37.316158274 +0100
+++ Source/ZigbeeNodeControlBridge/app_general_events_handler.c	2018-02-07 10:04:47.221059266 +0100
@@ -625,7 +625,7 @@
                             }
                             ZNC_BUF_U8_UPD   ( &au8LinkTxBuffer [u16Length] , sApsZdpEvent.uZdpData.sSimpleDescRsp.sSimpleDescriptor.u8OutClusterCount,    u16Length );
                             i =  0;
-                            while ( i < sApsZdpEvent.uZdpData.sSimpleDescRsp.sSimpleDescriptor.u8InClusterCount )
+                            while ( i < sApsZdpEvent.uZdpData.sSimpleDescRsp.sSimpleDescriptor.u8OutClusterCount )
                             {
                                 ZNC_BUF_U16_UPD  ( &au8LinkTxBuffer [u16Length] , sApsZdpEvent.uLists.au16Data [ i ],         u16Length );
                                 i++;
```

fairecasoimeme, 2018-02-07:
Great ! Thank you, I'll patch in the next version. 3.0d

closed 2018-02-07 by fairecasoimeme

tjikkun, 2018-02-07:
Sorry, after further reading it seems my patch is probably wrong. Since the zigate source is not yet public I cannot test myself, but with my patch you will get the correct size, but you will still get the input cluster listed instead of the output clusters.
Easy fix would *probably* be:
```patch
--- Source/ZigbeeNodeControlBridge/app_general_events_handler.c.orig	2018-02-07 10:04:37.316158274 +0100
+++ Source/ZigbeeNodeControlBridge/app_general_events_handler.c	2018-02-07 11:56:24.067034130 +0100
@@ -625,9 +625,9 @@
                             }
                             ZNC_BUF_U8_UPD   ( &au8LinkTxBuffer [u16Length] , sApsZdpEvent.uZdpData.sSimpleDescRsp.sSimpleDescriptor.u8OutClusterCount,    u16Length );
                             i =  0;
-                            while ( i < sApsZdpEvent.uZdpData.sSimpleDescRsp.sSimpleDescriptor.u8InClusterCount )
+                            while ( i < sApsZdpEvent.uZdpData.sSimpleDescRsp.sSimpleDescriptor.u8OutClusterCount )
                             {
-                                ZNC_BUF_U16_UPD  ( &au8LinkTxBuffer [u16Length] , sApsZdpEvent.uLists.au16Data [ i ],         u16Length );
+                                ZNC_BUF_U16_UPD  ( &au8LinkTxBuffer [u16Length] , sApsZdpEvent.uLists.au16Data [ i + sApsZdpEvent.uZdpData.sSimpleDescRsp.sSimpleDescriptor.u8InClusterCount ],         u16Length );
                                 i++;
                             }
                         }
```
But the documentation says "uLists is a union of the different arrays/tables which act as temporary storage for data elements used by the stack (and are therefore for internal use only)"

So I guess it is probably better to use the pointers *pu16InClusterList and *pu16OutClusterList

fairecasoimeme, 2018-02-07:
Ok, i'll do tests with your fix and I'll see if it's ok.
Thanks

fairecasoimeme, 2018-02-12:
It's OK. The modification works great. I add the fix in the new ZiGate version . 3.0d