fairecasoimeme/ZiGate #20: SPI MISO low via software

tjikkun, 2018-02-12:
@fairecasoimeme Is it possible to reset the device to the bootloader, without pressing the button?
There are Ftdi-Chip Serial board designs that put the Jennic chip into reset over the DTR/DSR lines. I was wondering if something like this is the case with Zigate.
I wrote https://github.com/tjikkun/zigate-flasher (not totally finished yet, but working) and it would be nice if I can use it without touching the hardware.

fairecasoimeme, 2018-02-12:
Hi, 
I know the DTR/DSR solution but now, ZiGate use a USB/TTL module and the DTR/DSR signal don't exit.
2 solutions for the next ZiGate version : 
* Integrate serial chip on ZiGate but you lose modularity
* Create a custom USB-TTL module with new broaching

So, It's no simple as I expect. I'll try to search a good compromise for the next version.
Thanks for the flasher tool. I'll try this too.
Best regards.

closed 2018-02-12 by fairecasoimeme

tjikkun, 2018-02-13:
Ok, I thought this was the case, wanted to make sure. Thanks.
Please let me know what you think of the flasher tool when you have tried it.