fairecasoimeme/ZiGate #21: Windows10 no response on GetVersion

qiou, 2018-02-12:
Hi.

I'm using windows 10. I've already installed the specific driver for the USB TTL.
I've also launched the GUI, selected COM3 (in my case), set the baud rate.
Opened the port (no error so far)

But the GetVersion does not give any answer.

I'm using a genuine Windows 10 pro edition, on a gamer PC which does not have much installed.
I'll probably switch to Linux at some point, on a RPi, but I wanted to give a try first.

Is windows 10 supported ?

fairecasoimeme, 2018-02-17:
Hi, 
please follow instructions in : 
http://zigate.fr/wiki/description-de-la-zigate/

Instructions pour les pilotes Windows 10 :

https://i1.wp.com/www.totalcardiagnostics.com/images/support/Prolific-USB-to-Serial-code-10-error-solution-11.png?w=750

    Téléchargez et enregistrez pilotes_PL2303.zip
    Débranchez tous les adaptateurs USB-série de votre ordinateur – et double-cliquez sur « PL2303_64bit_Installer.exe » ou « PL-2303_Driver_ Installer.exe » pour du 32bits
    Quand il vous le demande, branchez la ZiGate et cliquez sur « Continuer ».
    Redémarrez votre ordinateur. C’est tout!

closed 2018-02-17 by fairecasoimeme

maskimfr, 2018-02-18:
Bonjour,

J'ai le même problème, j'ai suivi les instructions à la lettre sans succès.
Avez-vous réussi à résoudre le problème ?