fairecasoimeme/ZiGate #23: documentation issue on complex descriptor ?

npenin, 2018-02-13:
Hi,

Thanks for the project, it is very handful. I have received my zigate end of last week and already started to build my own nodejs API (using typescript). However, I am stuck with the documentation on the Complex descriptor response and could not find any documentation on it. Why is there an xml tag as an int8 with a length in uint8 before ? Is this a type issue ?

Thanks,
Nicolas

pipiche38, 2018-12-17:
From what I saw in the Source code, this command is not implemented yet !

ISO-B, 2019-03-04:
Documentation has wrong command id(0x034) for this. Based on source code correct is 0x0531 which is E_SL_MSG_COMPLEX_DESCRIPTOR_REQUEST. Same thing with response should be 0x8531 instead on 0x8034

fairecasoimeme, 2019-03-07:
updated but I don't have explanations for Complex desc elements

closed 2019-03-07 by fairecasoimeme