fairecasoimeme/ZiGate #25: Support de la commande Move Stop Move 0x0083

esver, 2018-02-17:
Je n'ai pas réussi à utiliser la commande 0x0083 Move Stop Move et ai du utiliser à la place 0x0084 Move Stop with On Off

Le log :
Stop C779 (Salon 2) 0x0083 avec erreur
19:53:54.474 -> 01 02 10 83 02 10 02 15 3A 02 12 C7 79 02 11 02 11 03 
19:53:54.593 <- 01 80 00 00 05 04 02 00 00 83 00 03

esver, 2018-02-18:
Je pense, d'après le code JN-AN-1216, qu'une adaptation du style :
            case (E_SL_MSG_MOVE_STOP_MOVE):
            {
                tsCLD_LevelControl_StopCommandPayload    sPayload;

                sPayload.u8OptionsMask        =  au8LinkRxBuffer [ 6 ];
                sPayload.u8OptionsOverride    =  au8LinkRxBuffer [ 7 ];
                u8Status = eCLD_LevelControlCommandStopCommandSend ( au8LinkRxBuffer [ 3 ],
                                                                     au8LinkRxBuffer [ 4 ],
                                                                     &sAddress,
                                                                     &u8SeqNum,
                                                                     FALSE,
                                                                     &sPayload );
            }
            break;

            case (E_SL_MSG_MOVE_STOP_ONOFF):
            {
                tsCLD_LevelControl_StopCommandPayload    sPayload;

                sPayload.u8OptionsMask        =  au8LinkRxBuffer [ 6 ];
                sPayload.u8OptionsOverride    =  au8LinkRxBuffer [ 7 ];
                u8Status = eCLD_LevelControlCommandStopCommandSend ( au8LinkRxBuffer [ 3 ],
                                                                     au8LinkRxBuffer [ 4 ],
                                                                     &sAddress,
                                                                     &u8SeqNum,
                                                                     TRUE,
                                                                     &sPayload );
            }
            break;

doudz, 2019-02-08:
Effectivement la commande 0x0083 ne semble pas implémenté dans le firmware

Quel est la différence d'ailleurs entre les 2 commandes ? 0x0083 et 0x0084 ?

esver, 2019-02-08:
D'après la ZCL :
3.10.2.3.5 'With On/Off' Commands
The Move to Level (with On/Off), Move (with On/Off) and Step (with On/Off)
commands have identical payloads to the Move to Level, Move and Step
commands respectively. The also have the same effects, except for the following
additions.
• Before commencing any command that has the effect of increasing
CurrentLevel, the OnOff attribute of the On/Off cluster on the same endpoint, if
implemented, shall be set to On.
• If any command that decreases CurrentLevel reduces it to the minimum level
allowed by the device, the OnOff attribute of the On/Off cluster on the same
endpoint, if implemented, shall be set to Off.

doudz, 2019-02-08:
oui je suis d'accord pour les commandes move to level, move, move step, ça a un sens mais **move stop move** ça voudrait dire quoi ? stop ou move ? 
et pour le coup move stop onoff ? si c'est stop, c'est ni on, ni off...
bref c'est pas clair

esver, 2019-02-08:
Je me suis dit que c'était la bonne commande car elle est déjà dans le fichier https://github.com/fairecasoimeme/ZiGate/blob/master/Module%20Radio/Firmware/src/ZiGate/Source/ZigbeeNodeControlBridge/SerialLink.h et qu'elle correspondait à la fonction E_SL_MSG_MOVE_STOP_ONOFF sans le on/off

ISO-B, 2019-02-09:
From ZigBee Cluster Library specification:
>3.10.2.4.4 Stop Command
This command has no payload. Upon receipt of this command, any Move to Level, Move or Step command (and their 'with On/Off' variants) currently in process SHALL be terminated. The value of CurrentLevel SHALL be left at its value upon receipt of the Stop command, and RemainingTime SHALL be set to zero.
>
> This command has two entries in Table 3-5, one for the Move to Level, Move and Set commands, and one for their 'with On/Off' counterparts. This is solely for symmetry, to allow easy choice of one or other set of commands – the Stop commands are identical.

There is one extra command defined in SerialLink.h that is not needed.

Command code name | Command code | Command
-----------------------|------------------|-------------
E_SL_MSG_MOVE_TO_LEVEL | 0x0080 | Move
E_SL_MSG_MOVE_TO_LEVEL_ONOFF | 0x0081 | Move to Level
E_SL_MSG_MOVE_STEP | 0x0082 | Step
E_SL_MSG_MOVE_STOP_ONOFF | 0x0084 | Stop

Those commands are implemented to do something. Everyone of those commands have internally two different commands. Which internal command is used depends value of on/off parameter. Example: Move command with on/off parameter 0x00 will send Level control cluster command 0x01, but if on/off parameter is 0x01 then Level control cluster command sent is 0x05.

@fairecasoimeme I suggest that we remove command 0x0083(E_SL_MSG_MOVE_STOP_MOVE) from SerialLink.h and fdocumentation. It is only defined but not used anywhere. It would be logical to change id 0x0084(Stop) -> 0x0083, but it will break all libraries. Probably that one empty command between other commands doesn't make any difference, so I would only remove 0x0083.

doudz, 2019-02-09:
I agree, the command 0x0083 should be removed since it's not implemented.

esver, 2019-02-10:
I was talking about the 0x0083 command because I want 0x0084 without ON/OFF

ISO-B, 2019-02-10:
Then use 0x0084 with on/off parameter 0. It will use internally Stop without on/off. Also it should not really matter with stop if you use on/off or not since it is same command.

esver, 2019-02-10:
How can you use 0x0084 with on/off parameter 0 ? It need a new command like the code I wrote.

ISO-B, 2019-03-04:
ZiGate documentation has incorrect message format for command 0x0084.

Correct one is:
```
<Address fashion: uint8_t>
<Target short address: uint16_t>
<Source endpoint: uint8_t>
<Destination endpoint: uint8_t>
<Onoff: uint8_t>
<Options Mask: uint8_t> <- Optional
<Options Override: uint8_t> <- Optional
```

fairecasoimeme, 2019-03-07:
updated

closed 2019-03-07 by fairecasoimeme