fairecasoimeme/ZiGate #26: FlashExtensionProgrammer_JN5168.bin?

LouiS22, 2018-02-18:
Hello,

Where can I find the above bin (required for erasing EEPROM/PDM)?

fairecasoimeme, 2018-02-21:
New upload here.
https://github.com/fairecasoimeme/ZiGate/blob/master/Tools/FlashProgrammer/flashprogrammer.zip

closed 2018-02-21 by fairecasoimeme

lolorc, 2018-02-22:
Hi, how to achieve this on linux with JennicModuleProgrammer ?
Should I first flash FlashExtensionProgrammer_JN5168.bin then flash 3.0d ?

tjikkun, 2018-02-22:
My guess would be it is not possible. The EEPROM/PDM need to be cleared from the application (It would be nice if Zigate would support this, either through command or automatically, as is suggested in NXP docs)
The way I assume flashprogrammer does is, is that it loads the FlashExtensionProgrammer_JN5168.bin into RAM, and sets some info at a predefined address that will be read to know if it should erase the whole EEPROM or just PDM. I might be totally of the mark here though, it is mostly guesswork.

lolorc, 2018-02-23:
i've opened a separate issue for this, see #27