fairecasoimeme/ZiGate #29: Is remove device command 0x0026 working ?

doudz, 2018-02-25:
I tried to send the remove command and I always get status error code 169

Can you confirm the command is working ?

doudz, 2018-02-28:
Example :
short addr : bd3b
ieee : 158d0001de743c

I tried to send the command 0x0026 with the payload : bd3b00158d0001de743c => Error 169
full message : 0026000aa5bd3b00158d0001de743c

Since doc says that short addr is uint64_t I tried : 000000000000bd3b00158d0001de743c => Error 169
full message : 00260010bf000000000000bd3b00158d0001de743c

Am I wrong ?

pipiche38, 2018-08-14:
Is that a correct understanding from #32 that we have to used twice IEEE. 

In short:

"0026"+"IEEEE address"+"IEEE address" 

Is that correct ?

fairecasoimeme, 2018-11-14:
This command worked but there is a problem with devices on battery.
In fact, the first IEEE must be the ZiGate's IEEE and the second must be the device's IEEE.
When the command is done, you have to push the link button on the target device. Indeed, this is the target device which send to the ZiGate the leave command and active the processus to delete.

For the powered device, there is no problem

closed 2019-03-04 by ISO-B

pipiche38, 2019-08-19:
I just did some test today, and if you set the Zigate IEEE as the 1st IEEE, the device is not removed from the Zigate. Instead if you put TWICE the IEEE of the End Device then the device is removed from the Zigate.

I did the test with an Xiaomi Plug