fairecasoimeme/ZiGate #30: Problème association capteur Pluzzy

doudz, 2018-03-07:
Il semble y avoir un problème avec les capteurs Pluzzy
Soit il est impossible d'appairer le capteur, soit le capteur semble s'appairer à répétition
voir https://www.jeedom.com/forum/viewtopic.php?f=133&t=34490&start=80#p594273

Pour info il s'agit du capteur qui a aidé au développement de la zigate

rockynox, 2019-03-02:
Hello, je relance de thread, parce que je suis à la recherche d'informations sur la compatibilité de la Zigate (+ via le pluging Jeedom) et des prises Pluzzy. 
D'après le site de Pluzzy, la compatibilité est testée, mais impossible d'appairer les prises...

Si je peux aider, je suis partant, mais je sais pas trop vers ou chercher :) 

Merci ! :)

Franck69570, 2019-03-02:
Pour ma part, toujours impossible d'associer mes prises et mon capteur de T/H avec jeedom et le plugin Zigate, alors que ce dernier capteur avait été utilisé par le concepteur de la Zigate pour vérifier la compatibilité avec ces produits. Dommage car j'ai récupéré les prises inutilisées par un proche ; j'en ai donc 8 en ma possession, aux fonctionnalités intéressantes : télécommandables et remontée de la consommation...

pipiche38, 2019-04-17:
Sur le site Zigate.fr il est fait mention que le capteur n'utilise pas le meme Endpoint avec la dernière version.

https://zigate.fr/pluzzy-2-1-prise-pilotee/
https://zigate.fr/produits-pluzzy-compatibles-zigate/

Mukatai, 2019-04-17:
Sur le deuxième liens indiqué par pipiche il est mentionné :

> Pour que vous puissiez communiquer avec vos objets Pluzzy, il convient d’ouvrir votre réseau ZigBee sur le canal 19

fairecasoimeme, 2019-04-17:
Bonjour,
désolé pour le retard ...
Les appareils pluzzy ne sont malheureusement plus compatibles avec les dernières version du firmware. Les pluzzy utilisent une norme (ZHA1.1) qui ne peut plus coller avec les nouveaux capteurs.
Pour bien faire, il faut créer une nouvelle branche à ce firmware qui permettra de retrouver la compatibilité (même avec les plugins d'aujourd'hui, je ne change pas le protocole).
Il me faut juste un peu de temps (que je vais bientôt retrouver) pour vous fournir un firmware compatible pluzzy.

Par contre, ce sera difficile de rendre compatible pluzzy et ikea par exemple.

Désolé pour ce contre-temps.

Fred.

rockynox, 2019-04-17:
Bonjour Fred, 

Merci pour ta réponse ! Trop top si tu trouve un peu de temps pour re-modifier le firmware avec la norme ZHA1.1.
Et pour éviter d'avoir un fork pour pluzzy, ne serait il pas possible de s'appuyer sur un paramètre (a setter dans Jedom ou autre) pour dire au zigate quel norme utiliser ? (je ne me rends pas trop compte de la difficulté de l'affaire)

En tout cas si tu as besoins de testeurs j'ai tout une panoplie Pluzzy à tester ;) 

Léo

djuillard, 2019-04-29:
Up !
Particulièrement intéressé par le sujet, et pour aider si besoin également. 

De mon côté, sur Jeedom, je détecte les devices Pluzzy sur le canal 19, et ai une commande refresh dont je ne sais pas trop bien à quoi elle sert. Mais impossible d'ajouter une commande autre telle que On / Off pour allumer / éteindre la prise connectée par exemple. 

J'ai aussi essayé avec le soft  ZGWUI et je vois bien mes prises également. En revanche, je ne parviens pas à passer de commande encore à ce stade. Il faut je me plonge dans la doc des endpoint, cluster et groups, scenes.....

Merci !

damien

cleminfo, 2019-05-06:
Bonjour,
Moi aussi je suis intéressé, j'ai d'ailleurs acheter la zigate uniquement pour ça.
Merci

Nardo60, 2019-05-07:
Bonjour à tous,

Je possède plusieurs prise Pluzzy, j'ai aussi Domoticz sur un RPI, je souhaiterai pouvoir avoir l'état de mes prises et la consommation sur Domoticz.
Je n'ai pas encore la zigate

Avant de passer le cap j'aurai quelques questions 
comment connaitre la version de ses prises ZHA 1.1 ou ZHA 1.2
quelle clé acheter pour que celà fonctionne

Merci pour vos retours

PS : si vous souhaitez un beta testeur, je suis ok

fairecasoimeme, 2019-05-07:
Bonjour,
Je devrais mettre en ligne le firmware compatible ZiGate courant du mois prochain (Juin 2019).
Désolé pour le retard et merci de votre patience

Fred.

pipiche38, 2019-05-07:
Coté Domoticz, si il y a quelques chose à faire je vous propose d'utiliser cet Issue sur le github du plugin:
https://github.com/sasu-drooz/Domoticz-Zigate/issues/429

fairecasoimeme, 2019-05-10:
> Bonjour Fred,
>
> Merci pour ta réponse ! Trop top si tu trouve un peu de temps pour re-modifier le firmware avec la norme ZHA1.1.
> Et pour éviter d'avoir un fork pour pluzzy, ne serait il pas possible [...]

Je ne peux pas intégrer le ZHA 1.1 sur le SDK Zigbee 3.0 actuel. Par contre, je peux plus facilement intégrer le ZHA 1.1 sur le SDK ZHA1.2. Du coup, le firmware pluzzy sera aussi compatible avec les produits ZHA1.2 (genre xiaomi et autres) mais avec moins de fonctions et d'évolutivité que le firmware officiel de la ZiGate.

Pour les pluzzy, j'ai les capteurs de température/humidité / détection de mouvement / prises pilotées / ouverture fermeture de porte.
le capteur TIC n'est quant à lui pas compatible (jamais réussi à le faire fonctionner)

pipiche38, 2019-07-01:
> Bonjour à tous,
>
> Je possède plusieurs prise Pluzzy, j'ai aussi Domoticz sur un RPI, je souhaiterai pouvoir avoir l'état de mes prises et la consommation sur Domoticz.
> Je n'ai pas encore la zigate [...]

Salut, a priori Akila vient de pousse une pre-release du Firmware pour pluzzy.
Pour la partie Domoticz, je te propose d'utiliser l'issue suivant: https://github.com/sasu-drooz/Domoticz-Zigate/issues/429

cleminfo, 2019-07-08:
quelqu'un a essayé le firmawer zigate pour pluzzy ? Si oui quel est le fichier .bin à utiliser ?

pipiche38, 2019-07-08:
> quelqu'un a essayé le firmawer zigate pour pluzzy ? Si oui quel est le fichier .bin à utiliser ?

https://github.com/sasu-drooz/Domoticz-Zigate/issues/429#issuecomment-509169784

pipiche38, 2019-07-17:
Should be closed, as it is now handled : https://github.com/fairecasoimeme/ZiGate_HA1.2

pipiche38, 2019-07-25:
Pour information avec l'aide @cleminfo, nou savons le Capteur de Temp+Humi opérationnel sur Domoticz via Zigate et le plugin standard (pour l'instant branche Beta) ainsi que le Firmware specifique.

@fairecasoimeme bosse actuellement pour qu'on puisse avoir la prise pilotée avec remontée de Puissance...

pipiche38, 2019-07-25:
Is that issue should be transferred to : https://github.com/fairecasoimeme/ZiGate_HA1.2

closed 2020-09-30 by fairecasoimeme