fairecasoimeme/ZiGate #32: Fail to remove a device with TestGUI_ZiGate_NXP-4

lboue, 2018-03-10:
Hello,

I am trying to remove a device from the device list but I didn't succeed. I am getting an 'invalid parameter' error. 
[image]

Here is my device list:
```
Type: 0x8015 (Get Devices List)
  Length: 4
  ID: 0
  Short Address: 0xB26C
  IEEE Address: 0x158D0001D5C468
  Power Source: 0
  Link Quality: 149
  ID: 1
  Short Address: 0x1BD7
  IEEE Address: 0x158D000123DE6E
  Power Source: 0
  Link Quality: 164
  ID: 2
  Short Address: 0x7629
  IEEE Address: 0x158D000170EB24
  Power Source: 0
  Link Quality: 195
  ID: 50
  Short Address: 0x6F44
  IEEE Address: 0xB57FFFE0D820A
  Power Source: 1
  Link Quality: 93
```
[image]

Somebody can help me?

Regards,
Ludovic

lboue, 2018-03-10:
Actually I have to remove the '0x' prefix. So there is no more error from coming from TestGUI.
But this this error is from the zigate:
[image]

```
Type: 0x8000 (Status)
  Length: 5
  Status: 0xA9 (ZigBee Error Code)
  SQN: 0x00
  Message: 
```

Is anyone managed to remove a device on the Zigate?
How can I do a Zigate reset factory?

Regards,
Ludovic

doudz, 2018-03-20:
Duplicate of  https://github.com/fairecasoimeme/ZiGate/issues/29

doudz, 2018-04-12:
The code 0xA9 means 

> An APSDE-DATA.request with a destination
> addressing mode set to 0x03 failed due to no
> corresponding short address found in the
> address map table.

Using the IEEE address twice I got the code 0xAD

>  An APSDE-DATA.request requesting
> security has resulted in an error during the
> corresponding security processing.

is it normal ? (I tried on a Xiaomi aqara switch http://zigate.fr/xiaomi-bouton-switch-aqara-clusters/ )

closed 2019-01-10 by lboue