fairecasoimeme/ZiGate #37: Question diverses.

max5962, 2018-04-06:
Bonjour,

Je continue de jouer avec la Zigate et je trouve ça fascinant le boulot que tu as accomplie …
Du coup j’ai plusieurs questions, et je trouve pas mes réponses dans la documentation. ( une rubrique « analyse de problèmes » ou une sorte de FAQ peut être une bonne idée.

1/ Je joue avec la zigate depuis maintenant plusieurs mois , et j’arrive à certaines « limites ».
Je suis à 10 modules ( ouvertures de portes, capteur de présence, températures ) sur toute la surface de ma maison. ( tout sur pile ).
J’ai certains équipements qui régulièrement n’émette plus, du coup je me pose la question si il n’est pas trop loin de la zigate ou des interférences.
A partir de combien le « Link Quality » renvoyé par le getDevicesList, peut-on considérer qu’il est trop faible ( module trop loin ) ?

2/ as-tu déjà constaté que certains éléments génèrent de nombreuses interférences ? ( je soupçonne mon PC de bureau avec sa clef WIFI de créer des interférences ), un conseil pour éviter ça ? 

3/ la gateway xiaomi peut-elle servir de « routeur » d’un point de vue zigbee ?
SI oui, doit-on réaliser une opération pour forcer les « End-devices » à passer par le routeur et non plus direct vers le coordinateur ?

4/ J’ai par exemple parfois des réponses ( quelques dizaines de secondes plus tard ) :
2018-03-26 21:09:34 DEBUG (SyncWorker_27) [zigate] REQUEST : 0092 02b4b5010100
2018-03-26 21:09:34 DEBUG (SyncWorker_27) [zigate] # standard : 01 00 92 00 06 97 02 B4 B5 01 01 00 03
2018-03-26 21:09:34 DEBUG (SyncWorker_27) [zigate] # encoded : b’0102109202100216970212b4b502110211021003′
2018-03-26 21:09:34 DEBUG (SyncWorker_27) [zigate] (timestamp : 21:09:34)

Puis des « APS Data Confirm Fail » 🙂

Merci d’avance.
Maxime

Marc--, 2018-04-06:
Salut,

Intéressé également par cette question sur le LinkQuality... j'ai 9 modules a la maison, et frequement des pertes de liaison. Ma Zigate est dans la baie informatique avec routeur 4G / Wifi, le NAS, onduleur etc. Donc environnement potentiellement pollué !

Le truc assez dingue, c'est que pendant mes premiers tests, elle était sur un pc dans le salon, et la je captais super bien les sondes, même distantes ou en extérieur. Alors que maintenant certaines à quelques mètres sont limites.

Je rajoute une question : la possibilité de mettre une antenne était évoquée dans le blog. Toujours d'actualité ?

fairecasoimeme, 2018-04-11:
1/ le Link Quality est un chiffre de 1à 255. La qualité de réception n'est pas linéaire et par expérience en dessous de 30, la réception est en limite basse. Après le ZigBee reste de la radio ... les perturbations existent et peuvent polluer la réception du ZigBee.

2/ Je n'ai jamais observé d'interférences dues à des appareils WiFi. Les canaux sont normalement différents avec des bandes passantes différentes. Dans la théorie, le ZigBee vient s'intercaler entre les canaux du WiFi. Cependant, il est possible que des harmoniques puissent apparaitre et polluer les canaux ZigBee mais normalement une clé WiFi CE ne devrait pas produire cet effet. Mais ça reste de la théorie ...
La meilleure façon de tester votre théorie et de couper le WiFi pendant la période nécessaire de contrôle.

3/ A ma connaissance la Gateway Xiaomi ne peut jouer le rôle de routeur. C'est un coordinateur tout comme la ZiGate et n'a pas pour objectif de router les données puisque l'objectif d'un routeur et d'emmener le message vers le coordinateur.

4/ En effet, cela ressemble à des loupés de trames. Les APS DATA Confirm Fail sont des messages qui signalent (tout dépend du status dans la trames) un loupé d'informations en retour. Peux tu donner le status de retour pour préciser le problème ?

KiwiHC16, 2018-04-11:
1/ Link Quality: dans le cas d'un message router par plusieurs routeurs, que represente cette valeur ? Est ce la qualité du dernier lien radio (entre le dernier routeur et la zigate) ?

fairecasoimeme, 2018-04-11:
Bonne question ! Il me semble que le Link Quality de l'objet représente la qualité de réception avec son parent.

Si l'objet passe par plusieurs routeur, il faut donc contrôler la qualité de l'objet puis du parent du parent, etc... jusqu'au coordinateur.

Avec la commande LQI request (Management LQI request) 0x004E, il me semble qu'on peut avoir toutes les infos.

KiwiHC16, 2018-04-11:
Je suis entrain d'essayer de comprendre ces aspects car le routage peut être different de la structure objet/parent, ce qui ne simplifie pas la chose.

max5962, 2018-04-11:
Merci pour ces infos :)
1/ Si mon RSSI en hexa vaut 3C, la valeur decimal est 60 ou 195 ? ( bit de poid fort en premier ou en dernier ) ?
Je pose cette question car je suspect un bug dans le plugin d'home_assistant. il me remonte un RSSI de 60 pour un module à 1 mètre sans mur de la zigate.

2/ OK

3/ dommage, peut être un hack un jour :)

4/ 
```
2018-04-09 20:22:25 DEBUG (MainThread) [zigate] RESPONSE 8702 : APS Data Confirm Fail
2018-04-09 20:22:25 DEBUG (MainThread) [zigate]   - Status         : b'f0'
2018-04-09 20:22:25 DEBUG (MainThread) [zigate]   - Src endpoint   : b'01'
2018-04-09 20:22:25 DEBUG (MainThread) [zigate]   - Dst endpoint   : b'01'
2018-04-09 20:22:25 DEBUG (MainThread) [zigate]   - Dst mode       : b'02'
2018-04-09 20:22:25 DEBUG (MainThread) [zigate]   - Dst address    : b'0add01040006a701'
2018-04-09 20:22:25 DEBUG (MainThread) [zigate]   - Sequence       : b'e4'
2018-04-09 20:22:25 DEBUG (MainThread) [zigate]   # encoded : b'018702120210021ee6f0021102110212021add0211021402100216a70211e4021903'
2018-04-09 20:22:25 DEBUG (MainThread) [zigate]   # decoded : 0187 02 00 0E E6 F0 01 01 02 0A DD 01 04 00 06 A7 01 E4 0903
2018-04-09 20:22:25 DEBUG (MainThread) [zigate]   (@timestamp : 20:22:25)
2018-04-09 20:23:00 DEBUG (MainThread) [zigate] --------------------------------------
```
Voici ce que j'ai :)

fairecasoimeme, 2018-04-11:
1/ 3C = 60 --> pas terrible

4/ selon mes registres, le status 0xf0 correspond à l'erreur suivante : 
"Pending transaction has expired and data discarded"
Autrement dit, un timeout (réglé à 6sec) dans une transaction

max5962, 2018-04-11:
Du coup soit j'ai de sacrés interférences soit j'ai un souci avec la zigate soit Home assistant dit n'importe quoi.
Faut que je tente avec testGUi (toujours pas dispo sous linux ?) Histoire d'avoir une source fiable :)

Dans tous les cas, les APS erreurs, me confirme que j'ai des pertes de paquets.

Il y a t'il un mécanisme de rejeu dans le protocole zigbee ?

ruimarinho, 2018-04-15:
I get a lot of APS Data Confirm Fail reports as well. Here's an example:

```
2018-04-15 08:57:05 DEBUG (MainThread) [zigate] --------------------------------------
2018-04-15 08:57:05 DEBUG (MainThread) [zigate] RESPONSE 8702 : APS Data Confirm Fail
2018-04-15 08:57:05 DEBUG (MainThread) [zigate]   - Status         : b'f0'
2018-04-15 08:57:05 DEBUG (MainThread) [zigate]   - Src endpoint   : b'01'
2018-04-15 08:57:05 DEBUG (MainThread) [zigate]   - Dst endpoint   : b'01'
2018-04-15 08:57:05 DEBUG (MainThread) [zigate]   - Dst mode       : b'02'
2018-04-15 08:57:05 DEBUG (MainThread) [zigate]   - Dst address    : b'2c8201040000a701'
2018-04-15 08:57:05 DEBUG (MainThread) [zigate]   - Sequence       : b'b4'
2018-04-15 08:57:05 DEBUG (MainThread) [zigate]   # encoded : b'018702120210021ebef00211021102122c820211021402100210a70211b47e03'
2018-04-15 08:57:05 DEBUG (MainThread) [zigate]   # decoded : 0187 02 00 0E BE F0 01 01 02 2C 82 01 04 00 00 A7 01 B4 7E03
2018-04-15 08:57:05 DEBUG (MainThread) [zigate]   (@timestamp : 08:57:05)
```

max5962, 2018-04-26:
Hello,
Je viens de faire un test entre home-assistant et zigateGUI sur le résultat de getContactList et je trouve des valeurs vraiment bizarre : 

adresse | description | Distance Zigate | RSSI HASS | RSSI GUIZigate
0x841F | Temperature etage | loin | 48 | 255 => valeur abérante
0ADD | presence couloir | très proche | 47 | 48 
F3EB | Temperature Cave | proche | 28 | 134 
F261 | temperature salon | très proche | 63 | 161
980B | fenetre SDB | loin | 30 | 29
2240 | Temperature exterieur | moyen | 27 | 37
B4B5 | Presence etage | loin | 19 | 138
9693 | fenetre chambre | loin | 51 | 48
8331 | Fenetre cuisine | moyen-proche | 34 | 31

Je viens de regarder, les hexa remontés et je comprend pas comment il est possible d'avoir parfois des valeurs super proche ( cohérente ) et des fois des valeurs complètements différentes. 
Le plus étrange est le RSSI (47) de "présence couloir" qui est à 3 mètres de la zigate sans mur à traverser.

Merci pour tes éclaircissements. 
Maxime

max5962, 2018-04-26:
Édit 1 : sur mon équipement (ouverture de porte) le plus proche (1metre en direct) plus de nouvelles...
J'ai appuyé sur le bouton du module  relancé un join et hop il réapparaît sous un autre nom.

Je comprend plus rien :)

fairecasoimeme, 2018-04-27:
En effet, tout ça n'est pas très cohérent.
les mesures de la qualité se font à un instant T. La comparaison entre RSSI HASS et RSSI GUIZigate ne sont probablement mesurés au même moment. J'en conclu donc que la qualité varie énormément en fonction du temps.
Soit il y a un élément perturbateur (et ça va être difficile à vérifier sans matériel adéquat)
Soit il y a un PB sur l'étage radio de la ZiGate.

Pourrais-tu faire plusieurs mesures de la qualité sur un même produit espacé de quelques minutes pour voir si la qualité de réception pour un même produit varie beaucoup dans le temps ?

max5962, 2018-04-27:
Bonjour et merci pour la prise en compte de mon problème.
Les deux mesures rssi (Hass et zigateGUI) ont été faites à moins de 5 minutes d'intervalles (le temps de débrancher et allumer le software :))
Mais effectivement, je vais faire les mesures comme tu le souhaites et noter les valeurs sur 1 jour (intervalle : 1h). 
Je met ça en place et je reviens :)

handfreezer, 2018-04-27:
La disparition du périphérique et le retour en refaisant la procédure de "join" qui donne un nouvelle identifiant ressemble comme deux gouttes d'eau à mon ticket #38 , et je pense que ce sont deux problèmes indépendant. Je me demandais si la clé de sécurité avait une durée de vie max entre deux échange?

fairecasoimeme, 2018-04-28:
Pas à ma connaissance. :(

max5962, 2018-04-30:
Hello again,

Après plusieurs tests , j'ai gardé home-assistant pour être sur de garder les mêmes variables.
MAX(values_RSSI)-MIN(Value_RSSI) varie entre 5 et 30.
Donc c'est plutôt "stable".

Est-il possible que la même Zigate, au même endroit, mais sur 2 PC différents (Windows et linux ) restitues un getContactList différents ? 
Je vais tenter de récupérer la trame en hexa de chaque coté, pour voir si c'est pas un soucis d’interprétation.

handfreezer, 2018-04-30:
Comme jai deux zigate (1 pour la maison, l'autre pour des essais). Veux tu que je fasse un essai particulier? 

Questions relatives au sujet: jai vu qu'une commande exustait pour avoir la liste des nodes appaires. Y a t il la cle et le timestamp de derniere trame? Est ce pertienent de faire un relevé regulier?

KiwiHC16, 2018-04-30:
"jai vu qu'une commande exustait pour avoir la liste des nodes appaires" laquelle ? Peut être que je ne la connais pas...

max5962, 2018-05-09:
Hello Again,
J'ai continué mes investigations. J'ai développé un plugin pour home-assistant afin faire un getDevicesList tous les 5 minutes, comme ça ds qu'un module "parle" à la Zigate, l'info est poussé sur un influxDB ( rendu grafana ) : 

[image]

On peut faire plusieurs remarques : 

- Valeurs aberrantes plusieurs fois ( les deux détecteurs de mouvements : sensor 1 et 2 ) sont à 255
-  mis à part mon sensor 1, les valeurs semblent stables.

Sur ce graphe j'ai malheureusement pas tous mes devices car la zigate semble en avoir perdu quelques uns en route .... Mon getDevicesList me donne les info pour les devicesID ( les espaces représentes les modules perdu : {1,2,3,4, , ,7,8,9, , , }

On dit que dès qu'un module n’émets pas depuis un certains temps, la zigate la supprime ( pour détecteur de fenêtre c'est un comportement très gênant ) 

Merci pour avoir pris le temps de me lire.
Maxime

max5962, 2018-05-10:
Hello again,
Après 1 jour d'utilisation, voici ce que ça donne : 
[image]

ça fluctue pas mal.
D'autres constatent le même schéma ? ( afin de déterminer si c'est l'une des raisons suivantes : perturbations, module radio Zigate ect ) 

Ce qu'on peut en tirer déjà, c'est que la valeur par défaut semble être 255 ce qui donne des valeurs abérantes. Mes modules très très proches ( < 1 mètre en direct ) me donnent des valeurs très très basses : 76 ! Le Zigbee peut-il être perturbé par les fréquences en 400/800 Mhz ? ( alarme deltadore ) ?

fairecasoimeme, 2018-05-18:
Salut,
Je n'ai à vrai dire pas beaucoup de retour sur ce problème. 
Ce que je te propose, c'est que je t'envoie une autre ZiGate pour discriminer le module radio. (ou pas :))
Envoie moi ton num de commande en privée et je t'en envoie une Mardi.

max5962, 2018-05-18:
Hello @fairecasoimeme, il n'est plus possible d'envoyer un message privée ni sur tweeter, ton site ou ici :)
Mon numéro de commande est : "Commande n°713".
Merci beaucoup !

fairecasoimeme, 2018-05-18:
erf ok je m'occupe de te renvoyer une ZiGate...

max5962, 2018-05-28:
Hello Akila,
J'ai bien reçu ma zigate. (Encore merci ! ) 
Tu as changé de fournisseur pour le boitier ? (Personnellement ça donne encore plus l'impression de "fairecasoimeme" c'est top ) ! 
Au bout de 20 minutes tout a été réinstallé. J'attends de voir les premiers résultats en fin de journée.( les premiers résultats semblent encourageant.
Juste à des fins de tests, peux-ton transformer ma première Zigate en routeur ? 
Si oui comment ? 
merci

fairecasoimeme, 2018-05-28:
Bonjour,
Je croise les doigts pour que les mesures de qualité de réception continuent sur cette lancée. Ça à l'air vraiment mieux ...

Pour le boitier, non c'est le même fabricant. Pour tout dire, la transparence du boitier permet avant tout d'observer l'état de la LED bleue :)

On peut transformer la ZiGate en routeur mais je n'ai jamais eu le temps de tester et surtout .... l'appairer à une autre ZiGate.
Pour la config du mode routeur, tu peux utiliser l'outil TestGui sous windows. Mais j'ai pas assez étudié le truc pour pouvoir l'appairer à une autre ZiGate.

Si test se révèlent concluant, je suis preneur pour que tu me retournes la ZiGate défectueuse pour que j'analyse pourquoi elle ne fonctionne pas correctement au niveau de l'étage radio. Je te paierai les frais de port bien sûr...

Merci pour le retour

max5962, 2018-06-13:
Bonjour,
J'ai pas pu répondre avant j'étais à l'étranger.
Je confirme que les résultats sont bien meilleurs : 

[image]

On remarque plusieurs choses : 

1. Le module radio est bien en cause sur mon soucis de RSSI erratique
2. La moyenne des RSSI a quasiment 70 points.
3. L'activité humaine fait fluctuer grandement ces valeurs. ( période ultra lisse correspond à la période où j'étais à l'étranger.

J'essayerais de faire "mumuse" avec l'autre Zigate pour la passer en mode router et je te tiendrais au courant. Ensuite je te l'enverais.

maxime

fairecasoimeme, 2018-06-13:
Merci pour le retour,
En effet, c'est flagrant. 
Ok pour le module. J'attends tes retours constructifs avec impatience :)

Fred

closed 2018-06-13 by fairecasoimeme

max5962, 2018-06-13:
De rien  : )

En revanche je constate certains comportements etranges depuis quelques jours (maintenant que je suis sur que le module fonctionne) : 
Par moment, la clef ne reagit plus.
(Detecteur de presence et temperature)
Il faut que je stop le docker, enleve la clef, reboot le docker pour que je recoit a nouveau des ordres.

Là je vais recevoir en pagaille des "devices annonce". Comme-ci la clef est tombé en "segmentation fault" et venait de se reveiller. C'est possible d'avoir un mode "verbose" de la zigate  ?  Histoire d'avoir de quoi bosser et d'etre autonome sur nos soucis  : ) 
( je suis ingenieur en informatique c'est un defaut professionnel ^^ )

KiwiHC16, 2018-06-13:
Quel soft as tu dans ton docker en interface de ta zigate ?

KiwiHC16, 2018-06-13:
home-assistant: je suppose, c est plus haut dans la page.

KiwiHC16, 2018-06-13:
Je viens de lancer la capture des mêmes infos sous jeedom avec abeille pour voir ce que cela donne et si c'est comparable.

max5962, 2018-06-14:
Hello @KiwiHC16 
Oui effectivement c'est du home-assistant.
Du coup je sais pas vraiment si c'est due à la clef ou la fiabilité des modules aqara : 

[image]

Ce graph est assez flagrant. Nous avons une communication constante puis coupure du lien ou plantage du module.
Tu as déjà observé ce comportement ?

KiwiHC16, 2018-06-14:
Hello max5962,
Non jamais observé ce comportement mais en même temps je n'ai jamais fouillé ce point. Dans ta courbe, je ne sais pas si c'est l'échelle mais je suis surpris par le nombre de points qui seraient remontés par le capteur de temperature. Quel modèle est ce ? Je ne sais pas si le capteur envoie selon  un periode ou sur variation de temperature,...
Moi je suis sous Jeedom qui fait des moyennes dans l historique donc le phénomè est peut être caché. Ce que je remarque c est que normalement, je devrai avoir une valeur moyénnée toutes les heures, hors il me manque des heures. Je vais mettre un capteur en sniff pour voir la fréquence d'envoie des informations. Je vais essayer de faire la même chose avec ma GW Xiaomi pour comparer.

KiwiHC16, 2018-06-14:
Le graphes LQI commencent à montrer des infos. Les valeurs sont stables et le pires a un bruit de +/-15 sur le LQI de 60. Je ne vois pas de pic comme toi pour l instant. A suivre.

max5962, 2018-06-14:
@KiwiHC16
Cest le capteur de temperateur V2 (carre).
J'ai remarqué qu'un capteur a l'exterieur ou les variations de température/pression/humidité sont frequentes, émet plus souvent.
Donc oui, sur ce graph j'ai beaucoup de points (1 tous les 15/20 minutes contre 1heure pour mes autres capteurs).

Mes graphes correspondent aux RSSI et non au LQI. Toutes les 20 minutes, j'interroge la zigate qui me donne les valeurs (rssi) et je log tout ca dans un grafana.
Les PIC sont en realités des valeurs aberantes (rssi=255) qui peuvent soulever des questions.
Lorsque la zigate nous renvoit 255, j'ai l'impression que c'est le signe d'un module qui n'a pas été "vue" (pas recu de message depuis un certains temps).

max5962, 2018-06-27:
@fairecasoimeme 
Un RSSI = 255 peut-il être considéré comme un élément qui est sorti du réseaux ?

KiwiHC16, 2018-06-27:
Essayons de ne pas melanger les sujets.
RSSI:
Le RSSI mesuré ne transite jamais dans les échanges Zigbee. Donc cette mesure ne peut venir que de la zigate. Cette info a été ajoutée par Akila dans les trames zigate. La zigate mesure probablement le RSSI (Quand ? Pendant la reception d'une trame ?) et l'ajoute à la trame.
"In telecommunications, received signal strength indicator (RSSI) is a measurement of the power present in a received radio signal." : C'est donc "uniquement" l'information de puissance du signal pendant cette période. Avec tout ce qui est dans la bande de 2,4GHz la puissance peut certainement bouger énormément. 255 voudrait dire que tu as un émetteur qui pose problème. Dans mes historiques moyennes, je ne trouve pas plus de 200. As tu essayé de couper le wifi pendant 24 heures pour voir si il y a une différence ?
Peux tu confirmer que dans toutes les mesures ci dessus, tu ne parles que de RSSI ?
Comme cette information est mesurée par la zigate, elle ne represente que ce que voit la zigate (confirmé par l'amélioration de tous les courbes après changement de zigate).
Ce qui veut aussi dire que si des équipements sont routés, nous n'avons aucune info sur eux, nous n'avons que des infos sur les derniers routeurs qui relaient les messages.
Ci dessous des graphes sur le RSSI.

LQI:
Le LQI tente de mesurer la qualité du lien ce qui est interessant car cela mesure la possibilité de communiquer entre deux équipements. Malheuresement chaque constructeur est libre de faire ce qu'il veut pour mesurer cette valeur. Cette info peut être récupérée de tous les équipements routeurs. Nous faisons des graphs sous Jeedom/Abeille avec cette info. 
Je n'ai pas de courbe d'historique. Il faudrait que j y réfléchisse.

KiwiHC16, 2018-06-27:
[image]

Cette courbe montre le RSSI d'une ampoule à 2m puis à 3m avec la mise en service au même moment de mon capteur de temperature (qui émet des trames toutes les secondes) . La zigate est toujours à vue de lampoule. La courbe de signal fort à plus faible (éloignement ?) et de stable à instable (Capteur 1s ?).

KiwiHC16, 2018-06-27:
Dans cette courbe un bouton xiaomi qui passe par une prise xiaomi (routeur). Les courbes sont identiques ce qui "confirme" ma comprehension.

[image]

KiwiHC16, 2018-06-27:
Dans tes commentaires tu indiquais que l activité humaine impactait le RSSI, il semblerait que l activité humaine provoque les echanges d infos plus souvent (capteur presence, door,...) et donc impacte le RSSI...

max5962, 2018-06-27:
@KiwiHC16 
Merci pour ces explications très complètes, j'en espérais pas tant :+1: 
je parle toujours de RSSI, la valeur LQI n'ayant pas encore été implémentée dans Home assistant ( j'ai d'ailleurs récupéré ta classe dans Abeille pour l'éplucher et développer l'équivalent dans mon système)

Je n'ai aucun router, ce qui implique que la connexion est toujours Zigate <-> module.

Donc si je comprend bien un RSSI à 255 n'est pas possible, mon wifi en serait la cause ( Akila a d'ailleurs indiqué que cela ne devrait pas poser problème), j'ai quand même modifié mes fréquences 2.4 et 5 Ghz pour ne pas marcher sur celles du zigbee. 
Je te rejoins sur le fait que le wifi doit jouer un rôle. 

Tu indiques que l'activité humaine joue sur le RSSI car mon activité va déclencher plus de modules et ainsi plus polluer mon environnement. Le module aura alors plus de mal à émettre. Ai-je raison ? 

La valeur du RSSI est pour moi un indicateur à d'éventuels problèmes.
Ce midi, je constate que depuis 2h je n'ai plus de remontée. Je reboot mon système domotique, idem, je reboot mon pc, idem.
Je débranche/rebranche la zigate, et je commence à récupérer quelques valeurs. 
Faut que je continue à creuser sur la rootCause à tous mes problèmes.

KiwiHC16, 2018-06-27:
Pour moi un RSSI a 255, c'est que la puissance reçue  sur le canal est tellement forte qu'elle sature la reception de la zigate. Qui est l'émetteur qui fait ca ? Si je me souviens bien, j'avais collé une lampe Ikea a une prise xiaomi et j'avais de valeur de 230 max (il faudrait que je fasse le test de nouveau).
2.4 est la bande partagée par le wifi et zigbee. Tu trouveras des schemas montrant les différents canaux et les correspondances entre canaux zigbee et canaux wifi. Normalement au démarrage de la zigate (la premiere fois), elle choisi un canal sans trop de puissance pour ne pas être perturbée. Mon experience est que les canaux 11, 20, 26 et 15 reviennent souvent (comme l indique la doc et la répartition des canaux). (il n'y a pas de correspondences entre le numero de canaux wifi et zigbee).
Le Wifi 5GHz est normalement trop loin en fréquence pour jouer le moindre role.

La mesure RSSI indique le niveau de puissance sur le canal. Pour faire simple et par analogie, ca mesure le niveau sonore dans une pièce. S'il n'y a personne, il n'y a pas de bruit (RSSI = 0). Plus tu mets de personnes qui parlent, plus le bruit sera fort (RSSI fort) mais personne ne n'entend personne. Normalement le zigbee fait en sorte qu'une seule personne parle à un instant pour qu'ils s'entendent. Mais je suppose que si les gents parlent en continu les uns après les autres le niveau mesuré sera plus fort qui si quelqu'un parle de temps en temps. Donc dans ton cas, plus il y a de capteur qui parlent plus le RSSI sera haut. En tout cas c'est ce que je vois avec mon capteur qui parle tout le temps et qui est en plus répété par deux routeurs.

Quand tu dis "Ce midi, je constate que depuis 2h je n'ai plus de remontée", veux tu dire que tu as perdu tous les capteurs en même temps ? (perso ca ne met jamais arrivée). Es tu sure du plugin HA ?

max5962, 2018-06-27:
Pour moi RSSI=255 n'est pas possible enfin en pratique.
Actuellement, j'ai eu 3 modules (2 temperatures et 1 presence) qui m'ont repondu en rssi=255.
Depuis  les capteurs de temperatures ne repondent plus, le capteur de presence est quant a lui revenu d'entre les morts apres 4H d'interruption.

Pour le wifi en phase, j'avais trouvé une image qui montrait les differents canaux, ca m'avait beaucoup aidé.

Pour ton analogie, je suis egalement en phase. Cela signifie donc qu'un RSSI fort ne signifie pas une bonne connexion. Faut vraiment que je me penche sur le LQI.

Pour le plugin HA oui, j'avais le meme soucis sur domoticz. En resumé ca m' a donné  : 
10h  :  Ok
11h  :  6 modules /12 ne repondent plus 
11h10  :  restart docker , no effect
11h15  :  restart pc no effect
11h30  :  debranchement et branchement zigate, capteur de presence reviennent peu a peu.
15h00  :  jappuie sur le bouton d'un capteur temperature, je vois rien arriver. Dans les logs.
16h00  :  hop une remonte de temperature.
Depuis (23h30) toujours 3 modules temperatures OUT, 1 module ouverture Out.
J'ai l'impression qu'ils sont sorti du reseaux.

KiwiHC16, 2018-06-27:
Sans faire de pub pour mon bout de soft, il me serait plus facile d'investiguer si tu avais Jeedom/Abeille, et le top du top avec une sniffer ZigBee. Aurais tu la possibilité de faire cela ?

max5962, 2018-06-28:
Je comprend tout à fait, et rien que le fait que tu prennes le temps de m'aider t'excuse quand bien même c'était de la pub :)
J'ai une zigate dont le module radio ne fonctionne pas bien, je vais regarder si il est possible de la mettre en mode sniff.

Pour info, mes modules se sont mis à réémettre à 00H46 (1X) ou à 1H46 (2X).
Le module d'ouverture est toujours out, je pense que j'ai dû déclenche le reset en jouant avec le bouton :)