fairecasoimeme/ZiGate #39: Response 0x8140 : not enough information

doudz, 2018-04-23:
The response 0x8140 should include short addr or ieee , endpoint id and cluster id because without that we don't know what we are talking about
```

Obj-> ZiGate | 0x8140 | Attribute Discovery response | 
  |   |   | <complete: uint8_t> |  
  |   |   | <attribute type: uint8_t> |  
  |   |   | <attribute id: uint16_t> |  
  |   |   | Complete: |  
  |   |   | 0 – more attributes to follow |  
  |   |   | 1 – this was the last attribute

```

KiwiHC16, 2018-06-20:
Yep, I did the same request. One solution is to use the SQN (sequence number).

doudz, 2018-06-20:
Why not, but it would be easier to have the addr

KiwiHC16, 2018-06-20:
I agree

nouknouk, 2018-07-11:
+1
the same for me

doudz, 2018-08-20:
But it seems that many devices don't support Attribute Discovery, so is it really important ?

nouknouk, 2018-08-21:
for mebit is all the difference between:
- zigate software which can only work with "hard coded" list of known devices.
- zigate software which can also propose all standard features available on an newly included device, even if this device is not yet supported by the software.

So, yes, for me this is somethings that provides added value.

doudz, 2018-11-27:
Any update to this issue ? this can really make things smoother

doudz, 2018-11-27:
https://github.com/fairecasoimeme/ZiGate/pull/85

pipiche38, 2019-03-06:
I think that is cover as part of 3.0f

ISO-B, 2019-03-22:
Documentation seems to be update also. Someone open this again if this is still problem, but seems to be fixed.

closed 2019-03-22 by ISO-B