fairecasoimeme/ZiGate #40: Question about battery / refresh rate for Xiaomi Sensors

nerdosity, 2018-05-09:
Hi,

I would like to buy a Zigate, but I have to be sure that it will not cause me these kinds of issues:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/555

If it is not affected (expecially battery and refresh rate for temp sensors), I will get it ASAP.
Regards.

KiwiHC16, 2018-05-14:
"battery status: that's too random. Someone told me in another issue that it get "up to one hour" to get refreshed. But that's a bit optimistic. Sometimes it got battery status immediately after adding, sometimes it got after multiple removing/readding, keeping them near and separating, sometimes it NEVER got battery status"

Fact are:
- You have 2 versions of Xiaomi Temp (Rond one: V1 and square One: V2 ). They don't behave exactly in the same way.
- Batterie information are send either as per zigbee protocol or as per a proprietary message.
- ...
Any way for V2 version you can have the info every 50 minutes which is far more than enought for a batterie which is suppose to last months...

nerdosity, 2018-05-14:
Hi.
I just bought the Zigate, I hope it will behave better than the other device.
Yeah, I have 1x old version and 6x new version.
Both were advertised as supported, but they are not properly supported.

[image]

[image]

Both were removed and readded a week ago.

I don't need battery information hourly, I just know that I have a solid reading at least once a week.
But seeing that even after a week it doesn't have a battery value when it should have after an hour, that's a little bit too inconsistant for me.
About the temperature sensors, they will end up on eBay soon 😆

KiwiHC16, 2018-06-20:
I don't recognise those screenshot. Which SW do you use connected to the zigate for the graphical interface ?

doudz, 2018-06-20:
That's Home assistant

nerdosity, 2018-06-20:
Yeah, that's home assistant.
It has a pretty usable zigate module, is not working automagically (you join the device with a command, then you have to configure the device manually inside the configuration YAML file) like the Deconz one but at least is open 😉

doudz, 2018-06-20:
For information, I'm currently writing/testing a support for Home assistant more automagic :wink:

KiwiHC16, 2018-06-20:
I'm dev of a Jeedom / Abeille plugin so can't really help on HA.
There are 2 types of reports from Xiaomi Equipment. One is transferred by zigate, the second should be implemented in next release 3.0.E.
In my plugin Abeille, I manage first type without any issue. Waiting for zigate 3.0.E for the second type.
So I would imagine your issue is related to HA plugin.

max5962, 2018-06-20:
@doudz  why not working on existing Home assistant plugin ?

doudz, 2018-06-20:
Because I use a different python lib for zigate

nerdosity, 2018-06-20:
@doudz instead I would like to start a project of professional made Web UI, in a couple of months. 
I'm a web UI developer, I think that we really need a universal UI (more or less like the one of raspbee, but in a way so that can be really extended and used elsewhere) for using the zigate as a standalone.
That way it can be easily integrated in home assistant, jeedom or anything else.

ISO-B, 2019-03-04:
Has this been solved by newer firmwares?

sifus, 2019-03-28:
in my case, my xiaomi temp captor refreshing every 6 or 8 hour... and i'm not happy with that..

for exemple , the last temperature checking in my appartement is at 19h... and now it's 00h36

pipiche38, 2019-05-06:
Les équipements Xiaomi émettent une trames tous les 5 minutes avec les informations pertinentes en fonction de l’équipement.

closed 2020-09-30 by fairecasoimeme