fairecasoimeme/ZiGate #41: USB write/read error

edgd1er, 2018-05-14:
Hi, thanks for your work.

I have a recurring problem with the usb stick zigate. ( version 1.0D, latest as of 13/05/2018). The only way I ve found to reenable zigate communication is to disconnect/reconnect the usb stick. restart rpi or remove/reload modules pl2303/usbserial does not correct the problem.

I'm using Abeille plugin in Jeedom which is based on your demo php app. ( usb -> fifo + MQTT -> 3 deamons php to handle messages)

It seems I'm the only one to have this. Just in case, It gives a lead for another problem.

``[148947.346296] pl2303 1-1.5:1.0: pl2303_vendor_write - failed to write [0000]: -32
[148947.346501] pl2303 ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[148947.346646] pl2303 ttyUSB0: pl2303_set_control_lines - failed: -32
[148947.346738] pl2303 ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[148947.347429] pl2303 ttyUSB0: pl2303_get_line_request - failed: -32
[148947.347782] pl2303 ttyUSB0: pl2303_set_line_request - failed: -32
[148947.348012] pl2303 1-1.5:1.0: pl2303_vendor_write - failed to write [0000]: -32
[148947.461338] pl2303 ttyUSB0: pl2303_set_control_lines - failed: -32
[148947.461643] pl2303 ttyUSB0: error sending break = -32
[148947.463694] pl2303 1-1.5:1.0: pl2303_vendor_write - failed to write [0008]: -32
[148947.463898] pl2303 1-1.5:1.0: pl2303_vendor_write - failed to write [0009]: -32
[148947.464168] pl2303 ttyUSB0: pl2303_get_line_request - failed: -32
[148947.464421] pl2303 ttyUSB0: pl2303_set_line_request - failed: -32
[148947.464639] pl2303 1-1.5:1.0: pl2303_vendor_write - failed to write [0000]: -32
[148947.464896] pl2303 ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[148947.464903] pl2303 ttyUSB0: pl2303_set_control_lines - failed: -32
[148947.464963] pl2303 ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[270144.872368] usb 1-1.5: USB disconnect, device number 6
[270144.872946] pl2303 ttyUSB0: pl2303_set_control_lines - failed: -19
[270144.872977] pl2303 ttyUSB0: error sending break = -19
[270144.873611] pl2303 ttyUSB0: pl2303 converter now disconnected from ttyUSB0
[270144.873695] pl2303 1-1.5:1.0: device disconnected
[270149.275382] usb 1-1.5: new full-speed USB device number 7 using dwc_otg
[270149.408125] usb 1-1.5: New USB device found, idVendor=067b, idProduct=2303
[270149.408147] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[270149.408179] usb 1-1.5: Product: USB-Serial Controller
[270149.408190] usb 1-1.5: Manufacturer: Prolific Technology Inc.
[270149.409801] pl2303 1-1.5:1.0: pl2303 converter detected
[270149.414783] usb 1-1.5: pl2303 converter now attached to ttyUSB0
[289811.885838] pl2303 ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[289811.885928] pl2303 ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[363510.632668] usb 1-1.5: USB disconnect, device number 7
[363510.636889] pl2303 ttyUSB0: pl2303_set_control_lines - failed: -19
[363510.636918] pl2303 ttyUSB0: error sending break = -19
[363510.638643] pl2303 ttyUSB0: pl2303 converter now disconnected from ttyUSB0
[363510.638734] pl2303 1-1.5:1.0: device disconnected
[363515.816580] usb 1-1.5: new full-speed USB device number 8 using dwc_otg
[363515.949032] usb 1-1.5: New USB device found, idVendor=067b, idProduct=2303
[363515.949054] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[363515.949065] usb 1-1.5: Product: USB-Serial Controller
[363515.949075] usb 1-1.5: Manufacturer: Prolific Technology Inc.
[363515.950566] pl2303 1-1.5:1.0: pl2303 converter detected
[363515.960049] usb 1-1.5: pl2303 converter now attached to ttyUSB0
[363549.137222] pl2303 ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[363549.137383] pl2303 ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
``

KiwiHC16, 2018-05-14:
Je viens d'avoir un probleme tres similaire mais sur un autre composant USB, la solution a été de changer le cable USB entre le RPI et le composant...

lolorc, 2018-06-07:
d'apres http://faire-ca-soi-meme.fr/domotique/2018/05/07/zigate-laventure-continue/#comments
il y pourrait y avoir des soucis avec le module usb/serial.
As tu reproduit sur une autre machine ? le rpi se plaint-il de son alimentation 5V ? (dmesg affiche Under-voltage detected!, ou sinon ya l'éclair jaune sur l'écran)

edgd1er, 2018-06-15:
Depuis le changement du module, je n'ai eu aucun problème en 6 jours. J'ai reçu un nouveau module TTL (pilote pl2303) que je viens d'installer. Aucun soucis depuis 2 jours. Pas de problème de pilote 2303 sur raspberry donc mais une panne matériel sur le module TTL. Je pense que l'issue peut être fermée.

closed 2018-06-15 by fairecasoimeme

Turgon37, 2021-07-06:
Bonjour, je rencontre le même souci que vous, est-ce que ce modèle d'adapteur est compatible avec la zigate v1 (celle qui etait livré avec le convertisseur à carte "bleu")
https://zigate.fr/produit/module-usb-ttl-cp2102/ ?

pipiche38, 2021-07-07:
Oui le module rouge est compatible avec la zigate v1