fairecasoimeme/ZiGate #42: FlashGUI erreur

phenates, 2018-05-23:
Bonjour,

Je rencontre une erreur lors du lancement de FlashGUI.exe sous windows 10 et je ne peux donc pas mettre a jour la ZiGate.

Détails de FlashGUI.exe.log:

Traceback (most recent call last):
  File "FlashProgrammer.pyw", line 2860, in OnExit
  File "FlashProgrammer.pyw", line 378, in __init__
IndexError: list index out of range
Traceback (most recent call last):
  File "FlashProgrammer.pyw", line 2897, in <module>
  File "FlashProgrammer.pyw", line 2892, in GUIMain
  File "wx\_core.pyo", line 7971, in __init__
  File "wx\_misc.pyo", line 6210, in SetInstallPrefix
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe9 in position 4: ordinal not in range(128)

Si vous avez une idée, je suis preneur.
Merci d'avance.

doudz, 2018-05-23:
Je ne connais pas le logiciel mais à mon avis il doit y avoir un caractère "spécial" dans le chemin
c'est une erreur classique de programmation
Par caractère spécial, je veux dire un caractère non ascii : éèêëàùâ ....

fairecasoimeme, 2018-05-28:
Cette erreur m'est déjà arrivée lorsque j'ai tenté de lancer le logiciel à partir d'un chemin réseau...

phenates, 2018-05-29:
Merci pour les retours.
C'est effectivement un pb de caractère "spécial" dans le chemin du répertoire. Le terme "ascii" du message d’erreur  aurait du me faire tilt.

closed 2018-05-29 by phenates