fairecasoimeme/ZiGate #44: Zigate is "eating" Xiaomi devices currently associated with Xiaomi Gateway

nerdosity, 2018-06-11:
Hi,

since I am not yet ready to switch to Zigate for my home assistant configuration, I am still using the Xiaomi Gateway for my proper config, slowly expanding the Zigate config.

https://community.home-assistant.io/t/new-custom-component-zigate/36820
This component is pretty stable, I would say that it's usable.

The funny thing is that, after associating my first device (a Xiaomi thermometer), I had a couple of notification that "New device is paired!"
Maybe I tried the zigate in the past with those devices, but now they are officially paired with my official Xiaomi Gateway.
How can the zigate steal them and auto-add if no button pressed?

fairecasoimeme, 2018-06-13:
Hi,
Normally If ZiGate is not in "Permit Join" it can't pair a device. On ZiGate, there are no button to press to add devices.
When a device leave and rejoin, It change its ID (short address) so it's not possible to get a shadow.
I don't understand your issue. For sure, it's seems to be a manipulation mistake.

Best regards,
Fred

nerdosity, 2018-06-20:
It did not happen again, but I only associated one device (that wasn't associated before with the Xiaomi Gateway) with the Zigate. 
Now, after the "New device is paired" message in home assistant, in the verbose log I can see three devices 😃 

There is a specific "zigate.permit_join" command to send in home assistant, but it's valid for 30 seconds, and is in specific developer panel, so it's very difficult to trigger it in a mistake.

I will update you on this.

closed 2019-04-27 by nerdosity