fairecasoimeme/ZiGate #45: Wifi module issue.

KiwiHC16, 2018-06-20:
https://github.com/KiwiHC16/Abeille/issues/282

doudz, 2018-06-20:
Pour ma part, je pense réécrire le firmware wifi en micropython, (je suis pas très à l'aise en Arduino)

doudz, 2018-06-20:
Looking to the firmware code, I saw that it use SoftwareSerial, maybe it's a timing issue when the module spend time to talk over wifi some serial data are lost or corrupt

KiwiHC16, 2018-06-20:
Request to bulb to get it's name:
echo -ne '\x01\x02\x11\x02\x10\x02\x10\x02\x1e\xd5\x02\x12\x69\xb5\x02\x11\x02\x11\x02\x10\x02\x10\x02\x10\x02\x10\x02\x10\x02\x10\x02\x11\x02\x10\x02\x15\x03' > /tmp/zigate 

And socat for transfer:
socat -d pty,rawer,link=/tmp/zigate tcp:192.168.4.8:9999

Result:
[image]

Feedback from Zigate is nearly always different !

KiwiHC16, 2018-06-20:
Yep, I saw flush() and used in the past yield. I managed to compile the soft now trying to flash it, but I'm not managing. Don't know why ;-(
Sent a mail to akila to get guidance.

KiwiHC16, 2018-06-20:
I just managed to do it through the air. Investigations to continue...

doudz, 2018-06-20:
Could you share your code here ?

KiwiHC16, 2018-06-20:
https://github.com/fairecasoimeme/ZiGate/tree/master/Module%20WiFi/Firmware
No change yet. If I do I'll propose it on Akila's github.

doudz, 2018-06-20:
oh ok, I supposed that you made change

KiwiHC16, 2018-06-20:
Even with yield() everywhere I still have the issue.
If I log what is read on the serial coming from zigate, I can see the mistake. So issue is on serial to zigate.
Source of the code at
https://github.com/plerup/espsoftwareserial 
Stating:
"Please note that due to the fact that the ESP always have other activities ongoing, there will be some inexactness in interrupt timings. This may lead to bit errors when having heavy data traffic in high baud rates." 
Even if we are not in eavy traffic I think we should try with lower  speed.
I did the same test on a standalone ESP8266 and I have the same issue.

KiwiHC16, 2018-06-20:
Conclusion tonight: don't use softserial but HW serial.

doudz, 2018-06-21:
Esp has Hardware uart but I suppose it is used by the USB plug to flash the firmware

KiwiHC16, 2018-06-21:
Yes. So why not to use it in prod.
No CRC error this morning after 7h.

doudz, 2018-06-21:
But it require a new wiring since hw uart is on different PIN

KiwiHC16, 2018-06-21:
Exact. 

If you don't want to touch the HW, reduce the speed of the link (Should work). Need to recompile Zigate with reduced link speed but is not so easy, I tried yesterday and some lib are not happy with lower speed. 

HW, cut the track of GPIO 12/14 and connect Zigate connector to TX/RX.

Or do like me: https://github.com/KiwiHC16/Abeille/blob/master/Documentation/950_Wifi_Module.adoc

So I stopped as HW solution work well.

doudz, 2018-06-21:
lol nice hack
Well, @fairecasoimeme  should stop selling the wifi module since it's almost unsuable
maybe the firmware could be improve to give more priority to software serial, it will be not perfect but can save people having this revision of the wifi module

@fairecasoimeme in the next board revision maybe you can add a switch to redirect tx/rx to zigate or to USB

doudz, 2018-06-22:
Did you try to change the buffer size? Currently it's 128 ,maybe we could 256 or 64 to see if something change ?

KiwiHC16, 2018-06-22:
Yes.
This is not a buffer issue because even with very short messages the problem exist.
It's not a problem of // messages (Up/Down at the same time).

KiwiHC16, 2018-06-22:
It's probably a lib issue. Now the challenge is to find the right piece of SW...

doudz, 2018-06-22:
Or reduce zigate baudrate

KiwiHC16, 2018-06-22:
Akila has some ideas. Let see the results of his tests.

pdecat, 2018-06-23:
Same issue here, ZiGate Wifi module is really unstable right now (bad checksums and no response errors, then stops responding to ping until power cycled).

pipiche38, 2018-10-02:
I would like to report something that I discovered during the weekend with a user of the Domoticz/Zigate plugin.
It looks like if you flow 5 zigate command in the same second, the Zigate module is loosing nearly all except the 1st one .

Shall not the TCP stack to manage the communication ?

KiwiHC16, 2018-10-02:
Which TCP Stack ?

pipiche38, 2018-10-02:
Wifi ..

KiwiHC16, 2018-10-02:
TCP is between App/Plugin and WiFi module. It doesn't know about ZiGate. It doesn't how long the ZiGate will take to find the equipment (could be few second based on my experience). Up to the app to control the flow. SQN number send back by ZiGate is a start...

pipiche38, 2018-10-02:
Why is that to the App level to manage communication flow , this is exactly what is done from App <-> Plugin <-> Wifi

Wifi <-> Zigate should control the flow. 

Even on Serial communication you have RTS / CTS

jalme, 2018-12-01:
Hi,

Not the same symptom but sensor status stop to work (door was stuck on open state). Log bellow is related to a reboot and pair of the same device to see if it would reset status.

Currently the door sensor is stuck on open.

Regards,
João

2018-12-01 15:35:22 DEBUG (SyncWorker_0) [zigate] Saving persistent file
2018-12-01 15:35:22 DEBUG (SyncWorker_0) [zigate] File already existing, make a backup before
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] Try loading persistent file
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] Load success
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] ZiGate found on port 9999
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] REQUEST : 0x0010 
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] Msg to send b'0010000010'
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] Encoded Msg to send b'01021010021002101003'
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] Waiting for status message for command 0x0010
2018-12-01 15:36:08 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2018-12-01 15:36:08 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8000: b'00000010'
2018-12-01 15:36:08 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:16, error:b'', rssi:0
2018-12-01 15:36:08 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2018-12-01 15:36:08 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8010: b'0001030d'
2018-12-01 15:36:08 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8010 - Version list : major:1, installer:30d, rssi:0, version:3.0d
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] STATUS code to command 0x0010:0
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] Waiting for message 0x8010
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] Stop waiting, got message 0x8010
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] REQUEST : 0x0021 b'\x03\x18\xc8\x00'
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] Msg to send b'00210004f60318c800'
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] Encoded Msg to send b'0102102102100214f6021318c8021003'
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] Waiting for status message for command 0x0021
2018-12-01 15:36:08 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8000: b'00000021'
2018-12-01 15:36:08 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:33, error:b'', rssi:0
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] STATUS code to command 0x0021:0
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] REQUEST : 0x0023 b'\x00'
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] Msg to send b'002300012200'
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] Encoded Msg to send b'010210230210021122021003'
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] Waiting for status message for command 0x0023
2018-12-01 15:36:08 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8000: b'00000023'
2018-12-01 15:36:08 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:35, error:b'', rssi:0
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] STATUS code to command 0x0023:0
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] Check network state
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] REQUEST : 0x0024 
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] Msg to send b'0024000024'
2018-12-01 15:36:08 DEBUG (SyncWorker_1) [zigate] Encoded Msg to send b'01021024021002102403'
2018-12-01 15:36:09 DEBUG (SyncWorker_1) [zigate] Waiting for status message for command 0x0024
2018-12-01 15:36:09 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2018-12-01 15:36:09 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8000: b'00000024'
2018-12-01 15:36:09 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:36, error:b'', rssi:0
2018-12-01 15:36:09 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2018-12-01 15:36:09 DEBUG (SyncWorker_1) [zigate] STATUS code to command 0x0024:0
2018-12-01 15:36:09 DEBUG (SyncWorker_1) [zigate] REQUEST : 0x0009 
2018-12-01 15:36:09 DEBUG (SyncWorker_1) [zigate] Msg to send b'0009000009'
2018-12-01 15:36:09 DEBUG (SyncWorker_1) [zigate] Encoded Msg to send b'010210021902100210021903'
2018-12-01 15:36:09 DEBUG (SyncWorker_1) [zigate] Waiting for status message for command 0x0009
2018-12-01 15:36:09 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8000: b'00000009'
2018-12-01 15:36:09 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:9, error:b'', rssi:0
2018-12-01 15:36:09 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8009: b'000000158d0001edb71264f243a94141081980450b'
2018-12-01 15:36:09 DEBUG (SyncWorker_1) [zigate] STATUS code to command 0x0009:0
2018-12-01 15:36:09 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8009 - Network state response : addr:0000, ieee:00158d0001edb712, pan:25842, extend_pan:4875499819169841221, channel:11, rssi:0
2018-12-01 15:36:09 DEBUG (SyncWorker_1) [zigate] Waiting for message 0x8009
2018-12-01 15:36:09 DEBUG (SyncWorker_1) [zigate] Stop waiting, got message 0x8009
2018-12-01 15:36:09 DEBUG (SyncWorker_1) [zigate] REQUEST : 0x0015 
2018-12-01 15:36:09 DEBUG (SyncWorker_1) [zigate] Msg to send b'0015000015'
2018-12-01 15:36:09 DEBUG (SyncWorker_1) [zigate] Encoded Msg to send b'01021015021002101503'
[... cut ...] :, rssi:123
2018-12-01 15:37:38 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2018-12-01 15:37:38 DEBUG (ZiGate-Event Loop) [zigate] Receive Bad status
2018-12-01 15:37:38 DEBUG (ZiGate-Event Loop) [zigate] Received response 0x8102: b'015cdd010006800086ff0000'
2018-12-01 15:37:38 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8102 - Individual Attribute Report : sequence:1, addr:5cdd, endpoint:1, cluster:6, attribute:32768, status:134, data_type:255, size:0, data:, rssi:123

pipiche38, 2018-12-01:
Bonsoir,

Je confirme l'augmentation du nimbres de CRCs avec le nombres de Devices.
Pour moi il y a plusieurs problèmes
1- le flow entre le module zigate et le module wifi doit être contrôlé. Je suppose que les débits et durée de traitements ne sont pas égaux et donc on ne peut faire l’hypothèse que ça se passe bien.
2- le Wifi 2.4 et le ZigBee sont sur la même bande 2.4 Ghz, par conséquent chaque module passe du temps sur des messages qui ne lui sont pas destinés. Donc je me pose la question sur Wifi et Zigate.
3- Quand j'utilise le Network Management pour scanner les channels et voir le niveau d’interférences, les niveaux sur tous les channels sont relativement haut ( > 170 )
4- aujourdh'ui c'est acceptable avec quelques devices, mais quand on va atteindre les niveaux bien suppérieurs meme le fait de répéter les commandes n'aidera pas, car on saturera encore plus le traffic et de plus les messages async des devices aucun moyen de les faire re-emettre.

Bref, il faut absolument une gestion de flux aussi bien entre Wifi/Zigate que USB-TTL/Zigate

KiwiHC16, 2018-12-02:
1) Mon experience, ce n'est pas nécessaire, mais pourrait être mieux.
2) Wifi et Zigbee sur 2.4GHz sur la meme bande: Oui mais pas forcement sur la meme bande, faire une config ou les canaux ne se chevauchent pas. Par défaut 11, 15, 20 ou 25 sur la zigate en fonction du bruit et Wifi en fonction de votre AP Wifi. Canaux wifi et canaux zigbee ne correspondent pas un pour un, voir les plages de fréquence.
3) Dans ce cas, il faut augmenter les ratio signal sur bruit, donc changes mes fréquences, rapproche les équipements les eux des autres,...
4) On est loin de saturé le debit zigbee avec quelques messages par heures. D'après mes stats avec 40+ equipements , j'étais à 1kb/s on a de la marge.
Je ne partage que partiellement ta conclusion car de toute façon nous ne pouvons faire de la QoS bout en bout qu'en les équipements finaux ne le font pas et qu'on ne peut changer leur soft. Tout le modèle est en best effort avec aucune garanti de service. Peut être je me trompe....

jalme, 2018-12-08:
Hi all,

I have also serious wifi connection issues. Only with 4 sensors and wifi is very unreliable. Running wifi v1.3 and 3.0e.

My home router accepts wifi on 2.4 and 5 GHz so most equipments except ZiGate will connect on 5GHz.

Nevertheless ZiGates network quality deteriorates often (several times a day) and even drops out of the wifi network.

pipiche38, 2018-12-19:
You might have a look to that:
https://www.nxp.com/docs/en/application-note/JN-AN-1079.pdf

jalme, 2018-12-20:
> You might have a look to that:
> https://www.nxp.com/docs/en/application-note/JN-AN-1079.pdf

Thanks but Zigate is running on channel 11 and my wifi on channel 1 so if I understood the doc this should avoid the type of random quality drops

pipiche38, 2018-12-21:
Indeed, you should push your wifi to 13 Channel ; or move the Zigbee channel to 24.
What the paper is also recommended is to use the non-overallping channels and in Europe the best channels would be 15,16,21,22; so in your cas if you want to stay on Wifi 1 the best would 21 or 22 for Zigate

pipiche38, 2020-09-30:
solved with Wifi firmware 2.0. Please open a new issue if you still have a problem

closed 2020-09-30 by fairecasoimeme