fairecasoimeme/ZiGate #47: APS data confirm fail ; what are status codes & pending ?

nouknouk, 2018-07-11:
I recently faced 'APS data confirm fail' responses ; context was  just after having sent two "attribute request" in a row.

Unescaped & parsed data is:
| field | data |
| ---- | ---- |
| length | 00 0e |
| status | e9 |
| endpointSource | 01 |
| endpoint | 03 |
| addressmode | 02 |
| address | 6e 25 |
| seq number | 01 |
| ??? |  04 00 00 00 03 9c 4b |

Here are my questions:

1) what's the meaning of the status "e9" ; is there a place where we can get the full list of statuses ?

2) why do I get 7 bytes of data after the sequence field ?

Thanks in advance.

ISO-B, 2019-03-04:
Do you still get these?

pipiche38, 2019-03-04:
@ISO-B , For me there is a decoding issue from @nouknouk .
As from the doc, the address is always coded by the zigate as an IEEE despite the fact that the mode is specifying 0x02 (short address)

See #161

ISO-B, 2019-03-07:
I tried to find what status e9 means, but couldn't find anything. This might be manufacturer specific status code. Response format should now be correct so no more u64 address when address mode is not 0x03.

fairecasoimeme, 2019-03-07:
It seems that 0xe9 status is MAC_ENUM_NO_ACK --> No ack received when expected

ISO-B, 2019-03-08:
8702 now respects address mode so decoding should work correctly.

closed 2019-03-08 by ISO-B

sifus, 2019-03-28:
hello i have this error log 

```
[2019-03-28 21:57:44][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1dd9, sequence:72, lqi:0
[2019-03-28 22:07:13][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:2229, sequence:73, lqi:0
[2019-03-28 22:31:38][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:a79a, sequence:74, lqi:0
[2019-03-28 22:33:50][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:43d7, sequence:75, lqi:0
[2019-03-28 22:58:02][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1dd9, sequence:76, lqi:0
[2019-03-28 23:07:32][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:2229, sequence:77, lqi:0
[2019-03-28 23:31:51][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:a79a, sequence:78, lqi:0
[2019-03-28 23:34:11][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:43d7, sequence:79, lqi:0
[2019-03-28 23:58:17][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:1dd9, sequence:80, lqi:0
[2019-03-29 00:07:54][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:2229, sequence:81, lqi:0
[2019-03-29 00:24:51][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:43d7, sequence:82, lqi:0
[2019-03-29 00:32:02][ERROR][zigate](ZiGate-Decode data) : RESPONSE 0x8702 - APS Data Confirm Fail : status:240, source_endpoint:1, dst_endpoint:1, dst_address_mode:2, dst_address:a79a, sequence:83, lqi:0
```

ISO-B, 2019-03-29:
Status code 240 is 0xF0. Chapter 10.2.4 Mac codes in JN-UG-3113 v1.5 states:
>
>0xF0 Pending transaction has expired and data discarded

This means that node you are trying to send commands didn't receive command within time limit