fairecasoimeme/ZiGate #54: Checksum problem

pipiche38, 2018-08-20:
The issue is reported on the Zigate Plugin on Domoticz. A user using Wifi Zigate is reporting time to time soem Checksum error in his log. While no one on USB Zigate is.

The plugin use the same checksum function for sending and receiving Zigate frame.

Here are some date in order for you to see if there is an issue somewhere:

> Checksum computed based on the Message b4 
> Checksum received (extracted from the frame): 5d
> MsgType = 8102 
> MsgLength : 0032 
> RawData ( The FullFrame received) : '01810200325d27d98d010000ff01004200250121bd0b0421a84305210900062401000000006429e11cd9c86564adb68101000a210000a803'

> Checksum computed based on Message: 83 
> Checksum received (extracted from the frame) :  c8 
> MsgType = 8009 
> MsgLength : 0016 
> RawData (The Full Frame received) :  '0180090016c8000000158d0001ecbcf135371455922f9162220b0003'

Here after the  checksum()
```
def getChecksum(msgtype,length,datas) :
        temp = 0 ^ int(msgtype[0:2],16)
        temp ^= int(msgtype[2:4],16)
        temp ^= int(length[0:2],16)
        temp ^= int(length[2:4],16)
        for i in range(0,len(datas),2) :
                temp ^= int(datas[i:i+2],16)
                chk=hex(temp)
        return chk[2:4]
```

doudz, 2018-08-20:
related to https://github.com/fairecasoimeme/ZiGate/issues/45

pipiche38, 2018-08-27:
Je viens d'avoir une remonté de problème de checksum avec une Zigate USB (quickstarter) !!!

buragohf, 2018-11-04:
Hi, i'm facing as well the issue on domoticz with my USB Zigate
Error: (Zigate)
onMessage : Frame CRC is bad, computed = 124 
received = 122

pipiche38, 2018-11-14:
Still few CRC error with the new Wifi module

pipiche38, 2018-11-27:
Je confirme que sur la Zigate - Wifi, maintenant que j'ajoute des Devices, j'ai de plus en plus de CRC error !

pipiche38, 2018-11-27:
Si comme je l'ai compris il n'y a pas de gestion de flux entre l’étage Ziggate l'etage Wifi, cela ne me parait pas déconnant, et cela va s'aggraver plus le nombre de devices augmentera !

KiwiHC16, 2018-11-27:
Quel version de module wifi utilises tu ?

pipiche38, 2018-11-27:
Module reçu il y a 1 mois .. (v. 1.3) est affiché sur le paquet du module.

Le meme problème chez un autre internaute qui a reçu une mise à jour d'Akila

closed 2019-03-21 by pipiche38