fairecasoimeme/ZiGate #57: Temprature refresh rate/threshold

mce35, 2018-09-03:
Hello,

I bought a Zigate a few weeks ago. I am using it with a few xiaomi v1 temp/humidity sensors ((among other).

Sensors connected to the Zigate only report value change of at least 0.5°C (not sure about humidity change), whereas sensors connected to the original Xiaomi gateway report changes of 0.1°C. This leads to "hole" in graphs (there may be a few hours between 2 reports).

I have placed 2 sensors side by side and the graph of the one connected to the xiaomi gateway is far smoother than the one connected to the Zigate.

I am using the Zigate with Domoticz, maybe this is an issue with the Domoticz plugin and not the Zigate itself?

Someone could help solving this issue please?

Regards.

pipiche38, 2018-09-03:
Today we are just waiting for receiving information from the device. Either due to a big change ( and I would believe it is > 0.5° C) or every 50'

If now you have hole of few hours, there is more a communication issue between the device/Zigate/Plugin

Which Zigate are you using ? Wifi or USB ?

We are planning to push the "developement" branch by a week or 2 to the master , which is expecting to improve communication. You might want to try the "developement" branch for a while if you like, to see if there is an improvement ?

do move to this branch:
`git checkout developement`
to revert back
`git checkout master`

mce35, 2018-09-03:
Thanks for your quick reply. 
I'm already using the developement branch (pulled out yesterday).

I am using the USB Zigate on a Raspberry 3. The test sensor is ~1 meter away from the Zigate.

Here is an extract from the domoticz data table from last night:

2018-09-03 03:00:00 | 22.7
2018-09-03 03:05:00 | 22.7
2018-09-03 03:10:00 | 22.7
2018-09-03 03:15:00 | 22.7
2018-09-03 05:05:00 | 22.2
2018-09-03 05:10:00 | 22.2
2018-09-03 05:15:00 | 22.2
2018-09-03 05:20:00 | 22.2
2018-09-03 05:25:00 | 22.2
2018-09-03 05:30:00 | 22.2
2018-09-03 05:35:00 | 22.2
2018-09-03 05:40:00 | 22.2
2018-09-03 05:45:00 | 22.2
2018-09-03 05:50:00 | 22.2
2018-09-03 05:55:00 | 22.2
2018-09-03 06:00:00 | 22.2
2018-09-03 07:40:00 | 21.7
2018-09-03 07:45:00 | 21.7
2018-09-03 07:50:00 | 21.7
2018-09-03 07:55:00 | 21.7
2018-09-03 08:00:00 | 21.7

You can notice a hole between 3:15 and 5:05 and between 6:00 and 7:40. 

I can provide domoticz logs if needed.

pipiche38, 2018-09-03:
@mce35 please log an issue on https://github.com/sasu-drooz/Domoticz-Zigate/issues so we won't pollute this github which is more focus on the HW, and first lets investigate to see where the problem is.

mce35, 2018-09-03:
Hello,

I have created an issue on the domoticz plugin (https://github.com/sasu-drooz/Domoticz-Zigate/issues/153) with more detailed information.

I'm closing the issue here.

Thank you for your help !

closed 2018-09-03 by mce35