fairecasoimeme/ZiGate #58: Mise à jour du firmware en 3.0d sur raspberry

jeromevdl, 2018-09-07:
La documentation sur la page https://zigate.fr/wiki/mise-a-jour-de-la-zigate/ n'est pas très claire.

Pour la première partie, pas de soucis, git, make, ... se passent bien et j'obtiens bien le binaire JennicModuleProgrammer.

C'est ensuite où ce n'est pas très clair. 

> « &lt;chemin vers le bin&gt; »

De quel bin parle-t-on ? Est-ce ca : https://github.com/fairecasoimeme/ZiGate/blob/master/Module%20Radio/Firmware/bin/ZiGate/ZiGate_Coordinator_v3.0d.bin ?

Pas convaincu car j'ai l'erreur suivante (avec .`/JennicModuleProgrammer -V 6 -P 115200 -f "../ZiGate_Coordinator_v3.0d.bin" -s /dev/ttyUSB0`):

```
JennicModuleProgrammer Version: 0.6 (r56124)
Setting baudrate: 115200
Detected Chip: Unknown
MAC Address:   00:15:8D:00:01:EC:BC:42
Opened firmware file size 193852
Module uses Bootloader v2 Header
Erasing:   100%
Writing Program to Flash
Writing:     5%
Failed to write at address 0x00002b00
```

Et si je réexecute la commande, j'ai :
```
JennicModuleProgrammer Version: 0.6 (r56124)
Setting baudrate: 115200
Error setting baudrate
```

Je dois garder le bouton appuyé pendant que je lance la commande ?

jeromevdl, 2018-09-07:
J'ai recommencé et c'est passé... les aléas de l'informatique désolé...

closed 2018-09-07 by jeromevdl

GPierrot, 2019-03-31:
Je rencontre le meme problème pour flasher ma pizigate, et je reste bloqué avec l’eereur ‘Baudrate’

Des pistes de résolution ?