fairecasoimeme/ZiGate #61: Minimal hardware for debugging ?

Smanar, 2018-10-06:
Someone know what to use to have a minium to make debuging possible with a zigate ?
We can find firmwares with UART but no more connectors available on zigate.

I m searching the cheapest way not the powerful one.

pipiche38, 2018-10-09:
Hello,

Don't know if this could be helpfull, but it looks like you can find on AliExpress a Zigbee sniffer ( CC2531 USB sniffer ). This could then allow you to use the Zigate as a coordinator, while you would sniff the Zigbee part with the other keey.

Has the debugger part, I found the following
http://www.ti.com/lit/ug/swru197h/swru197h.pdf

Not sure if this is helpfull, this is really not by competence area ... But at least the keywords match !

pipiche38, 2018-10-09:
You might found quiet some information here:

https://github.com/Koenkk/zigbee2mqtt/wiki/Getting-started

Smanar, 2018-10-09:
Yep but all is for Texas instrument :(
I think it's exctly what I need, but I need a an usb key with JN5168 and a debug connector like this sniffer, even an UART connector.

Lot of user use this key, I think I will search is someone have started a custom firmware for it, but I rly can't compare the performance beetwwen these 2 keys.

KiwiHC16, 2018-10-09:
https://github.com/KiwiHC16/Butineuse

Smanar, 2018-10-10:
Je ne cherche pas a sniffer un reseau zigbee, mais a suivre le fonctionnement d'un firmware.
Par exemple je peux demander au firmware d'envoyer une commande texte a un port UART, mais je n'ai "plus" de port UART pour pouvoir recevoir les infos.
Il me faudrait un device a base de JN5168, avec un port connecte au pc pour m'en servir, et en meme temps un port pour pouvoir suivre le deroulement du firmware. Apres ce que le device "fait" sur le reseau zigbee c'est autre chose.
Il y a peut etre moyen de tout faire avec le port actuel, mais "ca pique" sans utiliser ZGWUI, et une fois qu'une appli est connecte sur la zigate, ca n'a pas l'air possible d'en connecter une seconde.

KiwiHC16, 2018-10-10:
[image]
Ma solution perso. Fonctionne très bien.

KiwiHC16, 2018-10-10:
Sinon y a ca: https://peeveeone.com/?p=187

Smanar, 2018-10-10:
Lol, ha ouais quand meme ^^.
Mais est ce que avec cette methode tu peux a la fois te connecter au pc pour utiliser ZGWUI (ou TESTGUI).
et t'y connecter en mode console via putty par exemple ?

Si tu regardes les modules de TI, tu as au moins 2 connecteurs, L'USB et 1 pour le debogguage > https://www.ebay.fr/itm/CC2531-USB-Sniffer-Stick-Protocol-Antenne-Zigbee-iobroker-FHEM-Openhab-Analyzer-/253912573254

En fait pour les TI, y a meme des cartes d'experimentations completes partout et pas trop cheres, mais NXP ...

qm3ster, 2018-11-13:
You can get a Texas Instruments clone on Aliexpress for $5

fairecasoimeme, 2018-11-14:
Je prévois de faire un article sur cette partie. La ZiGate est munie de 2 uarts. Il est possible d'envoyer des messages sur l'UART1 et d'avoir un fonctionnement normal (avec ZWGUI par exemple) sur l'UART0.
Il faut juste changer 2 directives de compilation dans le makefile (TRACE?=1 et DEBUG?=UART1) et avoir un autre module USB TTL

KiwiHC16, 2018-11-19:
Sur la photo tu as les fils Orange et Jaune sur la droite qui te donnent les logs.

Smanar, 2018-11-20:
Ha mince, j'avais pas du tout capte quand j'ai vu ta photo, meme pas regarde le brochage, je pensais que c'etait juste les pin de bases.
Vais me pencher sur le datasheet.

KiwiHC16, 2018-11-20:
Gauche: Boot selection
Bas Gauche: UART0
Bas Droite: Alim
Droite: UART1 Debug

ISO-B, 2018-12-12:
I am tring to build zigate binary from source code to get debugging working, but memsize tool doesn't work. Does anyone have working version of Memsize tool from JN-SW-4170? I have tried multiple different windows versions and 32 and 64 bit versions without luck.

ISO-B, 2018-12-13:
I finally managed to build firmware with debug on. There is good place to take UART1 tx and rx from zigate board. Find R7 and R8 And use those since there is place for resistor but no resistor so good place to use. On the other side of the board where is multiple pads numbered from 0 to 6, number 2 is ground.

pipiche38, 2020-09-30:
This issue has been closed due to no activity since while!

closed 2020-09-30 by fairecasoimeme