fairecasoimeme/ZiGate #62: Backup/Restore of Tables

pipiche38, 2018-10-06:
In the context of a firmware update, we have usually to erase the EEPROM and/or PDM.

Would be great to have 2 Zigates command, 1 for requesting a full backup and 1 for pushing it back

fairecasoimeme, 2018-11-14:
you can follow the instructions on :
https://zigate.fr/2018/10/27/sauvegardez-et-restaurez-votre-zigate/

pipiche38, 2018-11-14:
excellent, thanks

max5962, 2018-11-14:
@fairecasoimeme 
Hello :)
Cela signifie que nous pouvons faire cela  : 
1. Une zigate en 3d déjà bien utilisé
2. DUMP l'eeprom
3. flasher la zigate en 3e
4. Injecter l'EEPROM 

et cela marche tout seul ( tous les modules sont toujours appareillé ) ? 

merci beaucoup
maxime

KiwiHC16, 2018-11-14:
Non car le format du contenu entre version change.

max5962, 2018-11-14:
C'est bien ce que je pensais ... C'est ce qui me rebute à MAJ ... 
Une conversion automatique du format old vs format new est envisageable ? 
ou un soft qu'on pourrait  exécuter sur un DUMP.

Travaillant dans l'informatique, si je dis au client " faudra virer vos mails puis les re-ajouter" à chaque mise à jour ça râlerait je pense :D

doudz, 2018-11-14:
Ça pourrait être cool d'avoir un fichier pour analyser la structure et voir si on peut faire une conversion

KiwiHC16, 2018-11-14:
C'est probablement faisable mais très loin d'être simple. J'ai investigué le sujet, dump de la prom, analyse bit a bit, analyse du code et franchement y a des tables dans tous les sens. Certaines variables sont gérées par des librairies dont on a pas le code source. Il faut connaitre le format n et le format n+1, faire les macros de conversions... y a des infos introuvables comme la gestions faite par le trust center... Perso je reviendrai sur le sujet quand je n'aurai plus d'autre sujet que celui la.
On peut esperer qu'avec l experience, nous arrivions a une version stable de la prom/data et qu'il n'y ai que le code qui change.
Pour l instant il est bien plus simple de partir d une prom vierge et de refaire l appairage des equipements. 
Peut être que qq a une idée lumineuse et géniale...

fairecasoimeme, 2018-11-14:
En effet,
difficile de maitriser cette partie. De ce que j'ai vu, le binaire contient l'adresse de la structure PDM qui elle-même va pointer sur l'eeprom où sont stockés physiquement les devices.
Le problème est que l'adresse où se trouve la structure n'est apparemment fixe. Du coup, s'il y a beaucoup de changement dans le source, il est fortement probable que l'adresse change lors de la compilation.

La solution (ce que j'ai déjà eu l'occasion de faire sur d'autres micro arm) est de fixée cette adresse, mais je n'ai pas encore trouvé la section correspondante.

A force de chercher, on va bien finir par trouver...

KiwiHC16, 2018-11-14:
Sauf que le résultat de mes recherches le PDM utilisateur n'est pas utilisé par le soft du coordinator. Les deux tables qui sont dans le PDM restent vide. Par exemple les tables nib ne sont pas dans le PDM.
Par contre, en écrivant je viens peut être d'avoir un début d'idée lumineuse. Au lieu de comprendre la structure dans le détail, on peut probablement prendre un raccourci. Je vois deux cas de figure,
a) cas simple, on écrit un bout de soft et on gere une structure, dans ce cas on c'est que l on fait on doit etre capable de faire la conversion.
b) cas complexe, par exemple on change un paramètre de dimensionnement, on l'augmente (si c'est l inverse ca risque de ne pas marcher). Par exemple nombre d'équipements pour le trust center (Akila ce doit te rappeler des choses). On fait la manip suivante: on mets tout le source sous git, on change le parametre, on lance la compilation, les scripts refont toutes les structures et le binaire. On compare avec git les difference et on voit dans le code les changements (Je suis tombé par hasard sur ces changements en jouant avec git). Les tables sont étendues. On recupere la prom, on verse dans le nouvelle structure et la ou les tables sont plus grandes on complete avec des 0x00 ou autre. Je pense que ca doit marché car j'ai l impression que toutes les allocations sont faites dans les déclarations donc les tables ne changent pas de taille dans la prom.

Si qq veut investiguer plus avant cette approche ...

pipiche38, 2018-11-14:
A mon sens, c'est la Zigate qui devrait s'occuper de la sauvegarde/restore car seule elle connait la structure des données ..

KiwiHC16, 2018-11-14:
C est une autre solution, qui serait que la zigate lise la prom et envoie sur le port les infos. Mais la aussi je ne suis pas sure que nous ayons toutes les fonctions dispos pour le faire. On peut lire la nib, mais je n'est rien trouvé sur la table trust center ,si qq avait une piste ...

KiwiHC16, 2018-11-14:
Il est fort probable qu on n ai pas besoin de toutes les informations mais que celles critiques.

ISO-B, 2019-03-04:
Next release version should be able to automatically migrate to next version even if there is changes in structures. More information can be found from #141 and #166

pipiche38, 2019-03-04:
Closed to be followed on #166

closed 2019-03-04 by pipiche38