fairecasoimeme/ZiGate #64: Support de la télécommande ikea Tradfri

ChoOo7, 2018-10-12:
Bonjour

J'ai bien lu ton article
https://faire-ca-soi-meme.fr/test/2017/06/06/ikea-tradfri-zigbee-3-0-compatibilite-zigate/

Ce que je cherche à faire, ce n'est pas d'associer ma télécommande à une lampe, mais à associer la télécommande à ma zigate, pour ensuite binder les événements sur le broker pour lancer diverses actions.

Je n'arrive pas à associer la télécommande à la zigate / à recevoir des evenements

Lorsque je tiens le bouton appuyé 10 seconde,s rien coté zigate

Et si je fias un factory reset, j'arrive à avoir des choses
string(26) "zigate/device_changed/38b8"
string(128) "{"endpoints": [], "addr": "38b8", "info": {"addr": "38b8", "ieee": "d0cf5efffe05aa5d", "mac_capability": "10000000", "rssi": 3}}"
string(4) "XXXX"
string(26) "zigate/device_changed/38b8"
string(128) "{"endpoints": [], "addr": "38b8", "info": {"addr": "38b8", "ieee": "d0cf5efffe05aa5d", "mac_capability": "10000000", "rssi": 3}}"

Mais rien ensuite concernant les clics sur les boutons

Est-ce quelque chose de possible ?

Merci

ChoOo7, 2018-10-14:
A ce que j'ai compris, la télécommande Ikea ne diffuse pas son status alors que par exemple le bouton aqara si.

Dans tes articles, j'ai vu que en mode "sniffing", tu peut voir passer les trames entre la télécommande et l'ampoule. Donc au niveau du hardware, on pourrait bien détécter le "clic". 

Pourquoi ne pas laisser cette possibilité au niveau du firmware "normal" je voir toutes les trames ?

ChoOo7, 2018-10-15:
J'ai un autre souci plus ou moins lié, je ne sais pas trop
Voir https://github.com/doudz/zigate/issues/30#issuecomment-429790547
L'association de mon ampoule et ma télécommande à la zigate fait que la télécommande n'arrive plus à pilote la scène de l'ampoule (mais les autres boutons continuent de marcher)

KiwiHC16, 2018-11-07:
La telecommande Ikea envoie ses messages sur une adresse groupe. La Zigate recoit bien ce message mais ne le transmet pas. Actuellement pas de solution "simple" possible.

doudz, 2018-11-12:
Si effectivement la zigate voit les messages "passés", peut-être faudrait-il implémenter un nouveau type de réponse afin qu'une application tierce puisse exploiter ces messages, même si les messages sont destinés à un autre équipement.
@fairecasoimeme qu'en penses-tu ?

Par exemple un message type 0x9000 "routed message" avec en paramètre 
- source addr
-source endpoint
- (source cluster)
- dest addr
- dest endpoint
- dest cluster
- payload

Un peu comme si on pouvait faire du "sniffing" à tout moment

KiwiHC16, 2018-11-12:
Ne semble pas simple dans la mesure ou cette partie est dans les librairies NXP, donc pas accessible facilement. Une autre piste serait d'ajouter la zigate au groupe (comme une ampoule), je n'ai pas encore eu le temps de regarder plus avant cette solution.

doudz, 2018-11-12:
J'avais pensé à ça aussi mais je n'ai pas de télécommande pour essayer

fairecasoimeme, 2018-11-12:
Salut à tous,

Je confirme que la partie qui ignore le paquet envoyé par la télécommande se situe dans une librairie compilé (lib qui gère le protocole bas niveau).
J'ai aussi tenté d'associer la ZiGate à un groupe mais je pense que ce n'est pas possible non plus car c'est contraire à son rôle.... Il faudrait que la ZiGate soit aussi en mode "device".
Bref, je n'ai pas trouvé de solution qui ne remette pas en cause les principes élémentaires de la ZiGate.

Les télécommandes qui respectent la norme fonctionnent toutes de la même manière. Seul Xiaomi, à ma connaissance, utilise les télécommandes (ou plutôt interrupteur) comme actionneur.

Si vous avez d'autres idées je suis preneur ...

ISO-B, 2018-12-30:
I managed to get some progress to this problem. You can add ZiGate to group or at least I managed to do this twice. Currently using latest source code from github built with debug on and doudz/zigate library. Hardware wise Trådfri 980lm e27 bulb and 5 button remote both using latest firmware,

1. Reset ZiGate
2. Put ZiGate to permit join and reset ikea bulb to get it joining
3. Put ZiGate to permit join and reset Ikea remote remote
4. Pair Ikea bulb and remote using touchlink from remote
5. Request groups from bulb. There should be only one group
6. Add ZiGate to this group.

After that it is possible to see debug message every time remote send command. Only color temperature command is manufacturer specific. I added function to ZiGate to enable manufacturer specific commands for remote. It shows that color temperature command is for scenes cluster. I haven't yet have time to investigate is it possible to figure which buttons is pressed. After that it might be possible to catch remote press and give user information.

fairecasoimeme, 2018-12-30:
Sorry, but how could you add ZiGate to the group ?
ZiGate have a 0x0000 short address. I have tried many times but I have never succeeded.
You did this with ZWGUI ?

ISO-B, 2018-12-30:
No, I didn't use ZWGUI. I used python library(https://github.com/doudz/zigate).

fairecasoimeme, 2018-12-30:
Ok, I use ZWGUI only to tests some commands too. it's usefull... but could you share your command to group the ZiGate or the parameters you used because as I said, I have never succeeded to add a ZiGate to a group.

ISO-B, 2018-12-30:
This is how I do it with python. Included couple debug messages so you can see what commands are sent
```python
# Start library
import logging
logging.basicConfig()
logging.root.setLevel(logging.DEBUG)
import zigate
z = zigate.connect(port=None)

# Permit joining for 60 seconds
>> z.permit_join(60) 

# From debug messages find bulb short address. Mine was this time 35e6
>> z.get_group_membership('35e6', 0x01)
DEBUG:zigate:Msg to send b'00620006b50235e6010100'
DEBUG:zigate:Encoded Msg to send b'0102106202100216b5021235e602110211021003'

# Check group
>> z.groups
{'3880': {('35e6', 1)}}

# Add ZiGate to group
>> z.add_group('0000', 0x01, '3880')
DEBUG:zigate:Msg to send b'00600007dd02000001013880'
DEBUG:zigate:Encoded Msg to send b'0102106002100217dd02120210021002110211388003'
```

pipiche38, 2018-12-30:
Just a naive comment, it is showing progress, but adding the Zigate controller to the group is so far not bringing any features rather than just seen debug messages . Correct ?

As indeed , the remote can manage group and the Zigate can manage the same group too

ISO-B, 2018-12-30:
Yes, at the moment ZiGate is only able to see commands sent to group. There is possibility to add functionality to convert those commands to button presses which could be directed to user.

ZiGate can manage group without being member of group.

fairecasoimeme, 2018-12-30:
Ok, I just tested and ZiGate intercept messages from remote controller. I don't know if It's a good things (concerning the norm) but It's works !
I confirm the issue when ZiGate ask the group with get group command to 0x0000 address.

This solution is a kind of hack because the coordinator is not a device and mix roles is hassle. I will think about it and I'll see how to give this enhancement without make damages.

Concerning touchlink, It's the same, normally it's for device or router... 

I'll see how It's the best compromise and I'll inform to you.

Great thanks @ISO-B

ISO-B, 2018-12-30:
I watched couple of Ikea gateway and Phillips hub  videos and it seems that both are able to handle touchlink pairing. Also read somewhere that coordinator should be able to handle touchlink. So I think that we should try that too.

fairecasoimeme, 2019-01-02:
I don't know, it's weird because for me it is necessary that the initiator of the touchlink is within 50 cm of the object so even if the ZiGate is able to do a touchlink, it doesn't seem to me very useful.
I will look for the touchlink solution on my side too but in the meantime, I validate the operation with groups for the 3.0f version

ISO-B, 2019-01-02:
If I understand correctly ZiGate should work as client(initiator) for remotes. When remotes touchlink lights those should join to same network with ZiGate and remote. Not sure though is there any benefit using touchlink over regular pairing. Have anyone any experience of using Ikea or hue hubs with touchlink?

ISO-B, 2019-01-03:
Made little bit more research about this. To achieve programmable buttons from ikea remote will prevent remote working without ZiGate and library. This process is mostly done with library not with ZiGate firmware.

# Pairing
First thing is to get ZiGate and remote paired. I am not actually sure is touchlink between remote and ZiGate needed. If remote tells group address to ZiGate while touchlinking, it would be awesome to have touchlink. If remote wont tell group straight to ZiGate it's possible to get it from bulbs.

After that it is time to pair bulb to ZiGate and remote. This could be possible to do just using remote touchlink. It should add bulb to ZiGate's network. If it is not possible then pair bulb to ZiGate first and after that touchlink bulb with remote.

# Library
When everything is set up it's time to use ZiGate library of your choice.
1. Get group from bulb. (Not needed if remote can tell it while touchlinking ZiGate)
2. Add same group address for ZiGate. (Needed for getting button presses from remote)
3. Remove group from bulb. (This will prevent remote working without library)
4. Add new group to bulb.
5. Repeat steps 3 and 4 for all bulbs that you want control with remote.
After those steps use library to map button presses to desired commands.

# Firmware
Only thing that is needed firmware wise is way to pass commands(or decoded button presses) from remote to user library. Rest functionality is handled with library.

# Conclusion
This is only way that I could figure how to get it work. It feels bit hacky, but if you don't want to use it don't add ZiGate to same group with remote and you don't get extra messages for your library. Without removing group from bulbs your remote works as before. 

This can already be done since fairecasoimeme added functionality to intercept pushed buttons from remote using group on ZiGate.

# Things to figure
Things that still need more research are:
1. Does bulb or remote send any information for networks coordinator(ZiGate) when they touchlink?
2. What info ZiGate gets from remote when touchlink is used between them?
3. ZiGate should be client(initiator) when touchlinking with remote?
4. How can remote add bulbs to ZiGate's network? 
   - Is touchlink between remote and ZiGate needed for this or is it enough just pair them?
   - Does bulb need to be first paired with ZiGate before remote can touchlink with it? 

Does this all sound logical or stupid?

max5962, 2019-01-06:
Bonjour,
Je ne suis pas sûr de comprendre le but de ce ticket et donc de savoir si je suis hors sujet ^^.
Je souhaite 

Outlet Ikea <-> zigate (pour avoir le status et l'allumer) (appareillage classique)

Ikea remote <-> outlet (pour contrôler la prise)
(Appareillage Ikea)

1/ Si je fais ça, j'arrive a voir le statut de la prise et la contrôler via zigate ET le bouton ?

2/ si oui, dans quel cas il est utile d'avoir les "ordres" du bouton vu qu'on a le résultat (prise allumé ou éteinte" ?

Merci

ISO-B, 2019-01-07:
I am not completely sure that I understand what you are asking since I don't speak french. 

1. No, unless you also add ZiGate to remotes group with outlet.
2. In that case you should always see button pressed first and outlet state after that.

If it is now possible to pair ZiGate with both remote and outlet and you can already use remote to control outlet then you should get outlet new status from outlet when remote changes it. Only reason that I can imagine where you could need to get remote button presses is if you would like to use ikea outlet and other manufacturer outlet(unless it can be paired with ikea remote) and get them to switch state at the same time. If it's not that time sensitive you could just change other outlet to same state after receiving ikea outlet state.

Hopefully I answered questions that you asked 😄

ISO-B, 2019-01-08:
@fairecasoimeme tested 3.0f branch with bulb and remote and messages work fine. Only problem that I noticed is that if ZiGate thinks that light is turned off then lower brightness button won't do nothing. It's probably something to do with enabling ONOFF server on ZiGate that it wont show commands when ONOFF is on OFF state.

ISO-B, 2019-01-09:
Found solution for this. Going to make pull request later today.

fairecasoimeme, 2019-01-09:
Great ! I wait your update to include in the 3.0f version
Thx

ISO-B, 2019-01-12:
I might have managed to also decode color temperature buttons scene messages. Still need to test more and verify before making pull request.

KiwiHC16, 2019-01-12:
I haven’t looked yet at the details but will do soon. 
@ISO-B as said by @fairecasoitmeme, very very good finding as we were stuck with this topic. 
I guess we should split the topic at least in 2. First is for the zigate to process group command and transfer it to com port. Second how to configure with touch link. 
From plugin point of view (I think this is what you call library) all device could join the network with HA process (inclusion process), then informations collected and then Used by the plugin. 
Touch link would be used only once with a test bulb to get the group address used by the remote. 
Touch link is complex and for exemple a hue remote will not configure an ikea bulb even if they are both touchlink. Hue works with hue. Ikea works with Ikea. I haven t tried rémote Ikea with hue bulb.

ISO-B, 2019-01-16:
Testing is done and everything is looking good.  Going to make pull request tomorrow. 

@fairecasoimeme is it ok if I remove ^M line-endings from files that I am editing?

KiwiHC16, 2019-01-25:
Hi,
As I managed to compile it all yesterday, I'm looking at this topic now.
If I request group from the ZiGate (as I would do for any device), I get the answer without crash. looks that the correction works.
After sending the command the zigate replies with message 8000 telling us that the command has been received.
Then It sends message 8062 with 5 groups at 'ffff' as per the log below:

```
AbeilleParser 2019-01-25 17:50:24[DEBUG];type; 8000; (Status)(Not Processed); Length: 5; Status: 00-(Success); SQN: 46
AbeilleParser 2019-01-25 17:50:25[DEBUG];Type; 8062; (Group Memebership)(Processed->MQTT); SQN: 46; endPoint: 01; clusterId: 0004; Address: 0000; capacity: 00; group count: 05
AbeilleParser 2019-01-25 17:50:25[DEBUG];Type; 8062;group 0(addr:16): ffff
AbeilleParser 2019-01-25 17:50:25[DEBUG];Type; 8062;group 1(addr:20): ffff
AbeilleParser 2019-01-25 17:50:25[DEBUG];Type; 8062;group 2(addr:24): ffff
AbeilleParser 2019-01-25 17:50:25[DEBUG];Type; 8062;group 3(addr:28): ffff
AbeilleParser 2019-01-25 17:50:25[DEBUG];Type; 8062;group 4(addr:32): ffff
AbeilleParser 2019-01-25 17:50:25[DEBUG];Type; 8062;Groups: ->-ffff-ffff-ffff-ffff-ffff<-
```

Is it the right expectations ? With a device it would set the 'group count' value to the number of groups set. Here it should be 0. As I never send any group to configure.

Now if I send a command to add a group to the zigate as I do to a device I receive a 8000 message from the zigate telling me that the command has been received. Nothing else. With a real device I also receive command 8060 (Add a group response) to confirm that the group has been added.

If I trigger messages to the group from Ikea Remote, I don't see any message from the ZiGate on USB. Is there any message implemented yet ? or should I only see information in the ZIGate logs ?

ISO-B, 2019-01-25:
ZiGate's max group count is defined in some file(zcl_options.h maybe, don't remember right now). Might be that it sends all groups. I can check that tomorrow morning. I have been using debug interface uart1 and there it shows groups right. I can also check that should ZiGate send 8060 also.

There should be messages implemented. See my last message from https://github.com/doudz/zigate/pull/72

ISO-B, 2019-01-26:
Tested it and you are correct.
1. Sending add group(0x0060) to ZiGate will only generate 8000 response. It probably should also generate 8060 response. This is should be easy to fix.
2. 8062 returning always 5 groups even if no groups have been added might be something to do way NXP has decided to handle group register. It seems that they fill group capacity with incorrect group id (0xffff) and max length group name. This is probably to make sure that there is always room for adding groups. If wished this could be removed or modified not to send ffff groups. @fairecasoimeme do you have thoughts about this?
3. 8062 always returns 5 groups with ffff, even if there should be added groups.  Tested with using debug from uart1 which shows that there is group added and also remote commands are received using that group id. It should show added groups insted of ffff

1 and 3 are things that needs to be fixed.

KiwiHC16, 2019-01-26:
Thanks for the confirmation. Not yet plugged the UART1 from my side.

max5962, 2019-01-26:
Someone succeed to pair Ikea remote (two button) with zigate ?
"https://m2.ikea.com/fr/fr/p/tradfri-kit-prise-connectee-telecommande-80364794/"

Outlet - zigate : ok
Outlet - remote : ok
Remote - zigate : KO.

Thanks.

ISO-B, 2019-01-27:
Did you tried pairing it using touchlink or by resetting it? You should put zigate to permit join state and reset remote. IT should pair automaticly

max5962, 2019-01-27:
Using touchlink nothing happend even using remote link button.
Using permit join nothing too. Even remove and re-add the battery.

Someone succeed ?

ISO-B, 2019-01-28:
Touchlink should not work, but resetting remote and enabling joining should. Removing battery won't do anything. Pressing remote's touchlink button 4 times within couple seconds should make led on remote pulsating. This will also reset it and should make it join ZiGate. That's how it works with 5 button remote.

KiwiHC16, 2019-01-28:
Remote - zigate : KO

Juste 4x OO at the back. (There is no OO but a plastic button)

ISO-B, 2019-01-28:
So remote won't pair after resetting? Does it somehow indicate that it has been reseted? If this is case then we need to start looking how to enable touchlink on ZiGate

KiwiHC16, 2019-01-28:
_So remote won't pair after resetting?_ 
I don't know how you reset a remote. On zigbee message level I'm not aware of any way to know if a device as restarted.

My understanding is that 4xOO trigger a Leave or/and a Join for the remote.
A long 10s OO will trigger a touchlink.

Why do you want touchlink in zigate ?

ISO-B, 2019-01-28:
From Ikea manual
> FACTORY RESET YOUR DEVICE
For wireless ON/OFF switch:
Press the pairing button 4 times within 5 
seconds.

I assume OO means pairing Button which is backside of remote? If we can`t get remote to join network with resetting it then only other way to get them to same network id touchlink

ISO-B, 2019-01-31:
> Tested it and you are correct.
>
>     1. Sending add group(0x0060) to ZiGate will only generate 8000 response. It probably should also generate 8060 response. This is should be easy to fix.
>
>     2. [...]

Found fix to 1 and 3. Haven't yet had time to test if it broke how ZiGate handles groups. Now responses come as they should and also no ffff groups are showing up. Will make PR after more testing.

fairecasoimeme, 2019-01-31:
@max5962 

I found how to pair ikea remote control to oulet 

Ikea remote control talk only with 0x0000 group

So,

1- pair remote and outlet with zigate
2- add 0000 group to remote control
3- add 0000 group to outlet

And that's all

I'll add the device (with explanations) to compatibility devices on zigate.fr

pipiche38, 2019-02-01:
Is there a place where we could get a consolidated documentation for the IKEA Remote Pairing and the Group Management.

From what I see with the 3.0f, they are now 2 'reserved' groups 0x0000 and 0xffff. Correct ?

I see from the previous post how to pair the remote with an outlet, but how do you do with a Bulb.
Currently with 3.0f, the remote after reset and pairing it with a Bulb doesn't have anymore action to the bulb. And I do not see anything coming at the Zigate side.

Am I missing something ?

ISO-B, 2019-02-02:
> Ikea remote control talk only with 0x0000 group

I assume that you tested using other groups as well?

> From what I see with the 3.0f, they are now 2 'reserved' groups 0x0000 and 0xffff. Correct ?

You are partially correct. Both 0x0000 and 0xffff are invalid groups by Zigbee standard. You probably should be add those to any device that follows strictly Zigbee standard for group ids. For some reason Ikea has decided to use group 0x0000 with Outlet controller remote. So in this case I would say that it is 'reserved'. 0xffff group is what NXP has decided to use to reserve memory for groups. 

> I see from the previous post how to pair the remote with an outlet, but how do you do with a Bulb.
> Currently with 3.0f, the remote after reset and pairing it with a Bulb doesn't have anymore action to the bulb. And I do not see anything coming at the Zigate side.
> 
> Am I missing something ?

I don't have that remote so I can't say for sure but I think this is how it will work. First you pair remote with ZiGate and after that add both remote and ZiGate to group 0x0000. Now you should be able to see which remote button is pressed. Pair bulb to ZiGate. Write program that handles incoming remote presses and sends commands to bulb.

@KiwiHC16 I made PR https://github.com/fairecasoimeme/ZiGate/pull/137 which should fix those group command

pipiche38, 2019-02-02:
@ISO-B Thanks for that. 

I'll try again and will try directly with ZGUI. But right now, no way to get the Remote with membership group 0x0000. Do I have to do something on the remote ? As this is a battery device and I'm not sure that is on Receive when Idle .

Question: When adding Zigate to group 0x0000. I believe we use 0x0000 has the Network Adddress of the Zigate and 01 has the Endpoint with Address Mode 0x02, or should I use the IEEE addresse, or it doesn't matter ?

BTW, the Xiaomi Plug has 0x0000 group membership by default. This might explain why Ikea does the same .

pipiche38, 2019-02-02:
Making progress

I'm able to add 0x0000 to group 0x0000. However : 
I do not get 8062 to confirm the add .
When query 0000 for group membership, only ffff is reported .

I beleive this is related to the bug early reported. I'm going to handle that in the plugin code

pipiche38, 2019-02-02:
Stupid question. Cannot we make Zigate part of group 0x0000 by default ?

ISO-B, 2019-02-03:
I took better look for https://zigate.fr/produit-telecommande-ikea-tradfri-compatible-zigate/ and noticed that it doesn't have input cluster for Groups so we are not able to change its group unless it has hidden input input cluster for groups.. It however have ZLL: Commissioning cluster which can change devices group depending how it is implemented.

> I'll try again and will try directly with ZGUI. But right now, no way to get the Remote with membership group 0x0000. Do I have to do something on the remote ? As this is a battery device and I'm not sure that is on Receive when Idle .
> 

It won't receive when idle. 5 button remote receives commands if you press button before and after sending command. It really doesn't matter since there is not useful clusters to send commands. I don't know how @fairecasoimeme managed to add remote to group 0x0000 or is it already there.

> Question: When adding Zigate to group 0x0000. I believe we use 0x0000 has the Network Adddress of the Zigate and 01 has the Endpoint with Address Mode 0x02, or should I use the IEEE addresse, or it doesn't matter ?
> 

It should not matter, but I personally use address mode 0x02, because short addresses are easier to remember and faster to write.

> 
> 
> Making progress
> 
> I'm able to add 0x0000 to group 0x0000. However :
> I do not get 8062 to confirm the add .
> When query 0000 for group membership, only ffff is reported .
> 
> I beleive this is related to the bug early reported. I'm going to handle that in the plugin code

PR #137 should fix this.

> 
> 
> Stupid question. Cannot we make Zigate part of group 0x0000 by default ?

Yes we could, but I would not do it. Not all users will need ZiGate to be in group 0x0000 and might need 5 groups for something else. They would have to remote it from group 0x0000. For me it is the best solution to leave group table empty by default and let user add groups they need. You can make your code to add ZiGate to group 0x0000 every time you start it. If it is already in group then you will just get response with group already exist status code.

pipiche38, 2019-02-03:
@ISO-B thanks. Maybe the version I got from Akila doesn't include this PR. That could explain

KiwiHC16, 2019-02-04:
Hi All,
I have included this into the plugin. Command 8085 and 8095 work very well.
I'm stuck with command 80A7 as I don't find the format of the content. So can't decode it. 
Where could I find these information ?
If the format is the one in python's lib (https://github.com/doudz/zigate/blob/master/zigate/responses.py row 750)
then I have an issue because messages I received are not aligned.
Thanks for the info.

pipiche38, 2019-02-04:
Decode80A7:

SQN int8
EP int8
ClusterID Int16
Cmd int8
Direction int8
?? int16 + int8
Source Address: int16

ISO-B, 2019-02-05:
I couldn't figure what those extra values are so I left them undecoded.

uint8                       u8Direction;
uint8                     	u8Attr1;
uint8                     	u8Attr2;
uint8                     	u8Attr3;

KiwiHC16, 2019-02-05:
Very good, all working this morning !!!
The only thing I can't get the the hold state for Left and Right button (link to my remote firmware ?).
Thanks a lot @ISO-B

pipiche38, 2019-02-05:
It does work, but still (with the early announcement version from @fairecasoimeme  ), I do not get:

0x8060 when adding 0x0000 to a group

When requesting for Group Membership to 0x0000 , I'm getting 5 times 0xffff and not the network address which Zigate belongs too.

ISO-B, 2019-02-05:
@KiwiHC16 Can you test with doudz python library? It should show left and right holds. If it doesn't then it is probably firmware related.

@pipiche38 latest release on Github v3.0f_PR doesn't have that fix. You can build binary from my fork if you are in hurry. I think that fairecasoimeme will test my PR before accepting it.

pipiche38, 2019-02-05:
@ISO-B I do confirm that hold left/right works with the version I've got from Akila.

The V3.0f I'm refering is one I got from Akila end of last week, and would tend to agree that your PR is not in, but must be in before a full release.

KiwiHC16, 2019-02-06:
@fairecasoimeme told me that he will look at the PR probably before releasing 3.0f.
I can't see any message on air interface so I guess it's due to the ikea remote firmware. I will connect it to Ikea GW and see if an OTA will be done. If so I'll check again the hold button situation.

KiwiHC16, 2019-02-06:
I put the remote to the Ikea GW and no upgrade was done.
I'm back on ZiGate and I have the hold as expected.
I did probably shut my eyes yesterday when I was holding the buttons...

pipiche38, 2019-02-15:
Currently doing some implementation test, and I have some difficulties to decode the Left/Right Release message on message 0x80A7

As command is concerned I receive 0x09 - Which is inline with your work @ISO-B 
But as Direction and Unknown, this seems to be more challenging and not straightforward.

Like the Direction and Unknown value could be more related to the Time of the hold rather than if it was left or right.As well this message seems coming several time

ISO-B, 2019-02-15:
I couldn't figure release message payload. I also tough time but values seemed to be all over place, like long press one time had big value and sometimes short. Release is always related to button that is last pressed, since you can only press on button.  Maybe I need to switch release to use direction 2 so it would be unknown.

pipiche38, 2019-02-15:
Ok, so then we are on the same page !

[3 more replies]

closed 2019-05-07 by fairecasoimeme