fairecasoimeme/ZiGate #66: Salus and Metering

pipiche38, 2018-10-18:
Is there anyone who own a Salus smart plug with the corresponing hub been able to sniff what is/are the Read Attribute associated to the Cluster 0X0702 to retreive the metering informations ( Power ...)

We have currently done some test, and when using the 0x0702 cluster with 0X0000 attribute we get always a 0 value. May be there is something to be done before (as enabling some metering), but without the full Hub solution it is really hard to get it.

Help more than welcome

deennoo, 2018-10-20:
Surely WE have to bé inspired by what is done on Samsung smartthing :

https://docs.smartthings.com/en/latest/ref-docs/zigbee-ref.html#zigbee-configurereporting

This plug works as expected on this.

@Akila, Can you help to try it on Zgui.

Some élément are Missing on Samsung exemple

fairecasoimeme, 2018-11-14:
Update in progress. Indeed, the 24bits coding missing ...

pipiche38, 2019-03-06:
Implemented in 3.0f

closed 2019-03-06 by pipiche38