fairecasoimeme/ZiGate #71: What are 0x8003, 0x8004 and 0x8005 messages

pipiche38, 2018-11-07:
Is anyone knows what are those message and to which object do they relate ? I didn't see any related command, correct ?

Are those messages sent after an ERASE of the Persistent Data ( 0x0012 ) ?

doudz, 2018-11-12:
@pipiche38... , please read the doc before create issue
https://zigate.fr/wiki/commandes-zigate/

0x8003 | Liste des clusters de l’objet
0x8004 | Liste des attributs de l’objet
0x8005 | Liste des commandes de l’objet

vosmont, 2018-11-18:
Hello,
I'm searching how to retrieve the attributes linked to a cluster.

0x0045 command sends the cluster list, but I do not succeed in retrieving the attributes.

pipiche38, 2018-11-18:
I believe you are looking for 0x0140 get attribute discovery request.

pipiche38, 2019-02-21:
Those messages are triggered from the 0x0011 command when doing a Reset 0x0011

pipiche38, 2019-03-04:
I beleive this could be closed as they were triggered by 0x0011

closed 2019-03-04 by pipiche38