fairecasoimeme/ZiGate #74: 8024: Status: 04: Fragmented data requests must be sent with APS ack (fatal error)

pipiche38, 2018-11-09:
When sending command 0x0024 , I'm getting in return 0x8024 with a Status != 0

Status: 0x04

However the system seems to behave correctly !

Here after is my startup phase:

0x0010
0x0009
0x0015
0x0021 on channel 11
0x0023
0x0024
0x0040 ( optional )

pipiche38, 2018-11-09:
Not an issue Status Code 04 is Busy Node

doudz, 2018-11-16:
I'm facing the same problem, how do you handle this ? (status 4)

pipiche38, 2018-11-16:
@doudz I re-open the case and update the title.
So far I'm not handling ;-)

But what I found is that I'm receiving less data than expected. Basically the message is shorter than the 24bytes expected.

doudz, 2018-11-16:
I think it means that the network is already started so not forming a new one, not joining, so start_network could not be done since it's already done
But the response doesn't respect the doc since there'no short addr, no ieee addr and no channel

In previous firmware (3.0d) the command 0x0024 just return nothing
https://github.com/fairecasoimeme/ZiGate/issues/15

pipiche38, 2018-11-16:
Sébastien , while looking quickly at the source code, it looks like indeed it refers to a new network created after Reset. This is why also in my initial issue I put the sequence of command when starting as I suspect that some commands are not necessary needed when in steady state (outside of Software Reset, erase PDM ..), but hard to find the right start sequence.

doudz, 2018-11-16:
Ok, so the busy status 4 should be safely ignore since it's not an error.

ISO-B, 2019-03-06:
There is 4 possible status codes to get from 8024:

id | Description
--|--------------
0x00 | Success | 
0x02 | Error invalid parameter. If End device tries to form network. Should not be possible with ZiGate as it is Coordinator
0x04 | Node is on network. ZiGate is already in network so network is already format
0x06 | Commissioning in progress. If network forming is already in progress

Success message comes when network formation success event is triggered. There is also event for network formation failure, but it doesn't send any feedback for user.

When ZiGate starts  after with empty pdm it won't be part of any network. Sending 0x0024 will try to form network. That is straight forward case.

After Zigate starts with network details in pdm it will send same 8024 response as when it has formed network. This happens very quickly after startup so if you miss this you don't know that network is already formed. If you now send 0x0024 it will try forming network again, which should fail with code 0x04. Would it be desired to get same 24bytes long response that ZiGate sends at startup, but with status code 0x04?

pipiche38, 2019-03-06:
`Would it be desired to get same 24bytes long response that ZiGate sends at startup, but with status code 0x04?`

I would tend to say that it makes sense.

@KiwiHC16 , @doudz what do you think ?

doudz, 2019-03-06:
@pipiche38 I don't understand your difficulties with 0x8024... sorry

pipiche38, 2019-03-06:
@doudz you were mentioning having the same problem https://github.com/fairecasoimeme/ZiGate/issues/74#issuecomment-439336093

What @ISO-B is proposing seems to be logical, just wanted to check if you are also fine with that

doudz, 2019-03-06:
@pipiche38 but as I said : Ok, so the busy status 4 should be safely ignore since it's not an error.

pipiche38, 2019-03-06:
Fine with me, but then let's get the documentation updated, or get the fix proposed by @ISO-B

ISO-B, 2019-03-22:
@fairecasoimeme can you add status 0x04 (Node is on network. ZiGate is already in network so network is already format) for this commands documentation.

fairecasoimeme, 2021-03-10:
doc updated

closed 2021-03-10 by fairecasoimeme