fairecasoimeme/ZiGate #76: Join problem after firmware update 3.0e

Gemies, 2018-11-11:
Hi Guys,

Ik hope you can help me out. This evening I updated my Zigate firmware to the latest version 3.0e. Now when I permitted joining in domoticz for 254 seconds. The zigate is not recieving any devices. And it is not blinking as before. In my case tradfri bulbs who were working before. 
I downloaded the software (NXP JN51XX FLash program) and update ( ZiGate_Coordinator_v3.0e.bin). I followed the update instructions as descibed on the website of Zigate. After updating firmware i also flashed (JennicSniffer_JN5168_1000000.bin) to get it in sniffer modus again. 
When this did not work i did the flashing proces again, but nothing happens when i permit join in hardware (254 or 255) 

I also reinstalled the domoticz zigate python plugin. But nothing joins nor the zigate device leds are blinking.

Does anyone have a clue what to do?

pipiche38, 2018-11-11:
I suggest that you got to the Zigate Plugin forum as it probably not related to the firmware itself

pipiche38, 2018-11-11:
@Gemies As, I will respond to you on the Zigate side, make sure that when you start with the Zigate Plugin under Domoticz you also set Erase Permanent Memory

closed 2018-11-12 by Gemies