fairecasoimeme/ZiGate #79: Zigate disconnected

max5962, 2018-11-14:
Hello,

J'ai régulièrement des déconnections de ma zigate sur mon laptop ( ubuntu ).
L’occurrence est totalement aléatoire, ça peut arriver une fois tous les 3 jours, comme 1 fois par heure...

C'est plus un appel à la communauté qu'une vrai "issue"

```
Nov 14 13:00:25 ubuntu kernel: [2072728.426246] pl2303 ttyUSB1: usb_serial_generic_read_bulk_callback - urb stopped: -32
Nov 14 13:00:25 ubuntu kernel: [2072728.426647] pl2303 ttyUSB1: usb_serial_generic_read_bulk_callback - urb stopped: -32
Nov 14 13:00:26 ubuntu kernel: [2072728.802915] usb 2-1.2: USB disconnect, device number 11
Nov 14 13:00:26 ubuntu kernel: [2072728.804206] pl2303 ttyUSB1: pl2303_set_control_lines - failed: -19
Nov 14 13:00:26 ubuntu kernel: [2072728.804222] pl2303 ttyUSB1: error sending break = -19
Nov 14 13:00:26 ubuntu kernel: [2072728.804557] pl2303 ttyUSB1: pl2303 converter now disconnected from ttyUSB1
Nov 14 13:00:26 ubuntu kernel: [2072728.804640] pl2303 2-1.2:1.0: device disconnected
Nov 14 13:00:26 ubuntu upowerd[2573]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.0
Nov 14 13:00:26 ubuntu upowerd[2573]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2
Nov 14 13:00:26 ubuntu kernel: [2072729.026583] usb 2-1.2: new full-speed USB device number 12 using ehci-pci
Nov 14 13:00:26 ubuntu kernel: [2072729.136063] usb 2-1.2: New USB device found, idVendor=067b, idProduct=2303
Nov 14 13:00:26 ubuntu kernel: [2072729.136071] usb 2-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Nov 14 13:00:26 ubuntu kernel: [2072729.136076] usb 2-1.2: Product: USB-Serial Controller
Nov 14 13:00:26 ubuntu kernel: [2072729.136081] usb 2-1.2: Manufacturer: Prolific Technology Inc.
Nov 14 13:00:26 ubuntu kernel: [2072729.136787] pl2303 2-1.2:1.0: pl2303 converter detected
Nov 14 13:00:26 ubuntu kernel: [2072729.138744] usb 2-1.2: pl2303 converter now attached to ttyUSB0
Nov 14 13:00:26 ubuntu mtp-probe: checking bus 2, device 12: "/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2"
Nov 14 13:00:26 ubuntu mtp-probe: bus: 2, device: 12 was not an MTP device
Nov 14 13:00:26 ubuntu upowerd[2573]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.0
Nov 14 13:00:26 ubuntu upowerd[2573]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2
```
Une idée ? 
pour le moment soit un restart du container docker fonctionne, soit je suis obliger de débrancher puis redémarrer le docker.
merci.

KiwiHC16, 2018-11-14:
pl2303 semble poser des problèmes, essayes avec un FTDI232 aui fonctionne bien chez moi.

max5962, 2018-11-14:
Merci @KiwiHC16 
N'ayant pas du tout du domaine électronique, cela ne me parle pas :)
Finalement il a fallut que je demande de "l'aide" pour que je trouve une alternative à mon problème qui date de plusieurs mois ^^ : 

```
restartZigateUSB.sh : 
echo 1 > /sys/bus/usb/devices/usb2/../companion
echo -1 > /sys/bus/usb/devices/usb2/../companion

```
Où le "1" correspond au port USB et le "2" au numéro du HUB présent dans : 
```
 lsusb -t 
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 20, If 0, Class=Hub, Driver=hub/6p, 480M
        |__ Port 2: Dev 21, If 0, Class=Vendor Specific Class, Driver=pl2303, 12M
```

cela permet à distance de simuler un débranchement et branchement  de la zigate

KiwiHC16, 2018-11-15:
Interressante solution.
Dans Abeille, edger1er a ajouter une détection:
$cmd = "egrep 'pl2303' /var/log/syslog | tail -1 | egrep -c 'failed|stopped'";
associée à une alarme pour demander le débranchement/branchement de la zigate.

Est ce que tu as pu tester le solution ci-dessus ? Est ce qu'elle fonctionne ?

max5962, 2018-11-15:
@KiwiHC16 
Je suis sur home-assistant (pluging "doudz" qui se base sur une librairie commune à beaucoup de soft domotique de ce que j'ai compris)

Pour ma part, la chaîne est plus complexe et industrialisé ( pour apprendre ).
Tout est sous docker et j'ai câblé une chaîne ELK (https://www.elastic.co/fr/elk-stack) et ajouté le pluging sentiln (https://sentinl.readthedocs.io/en/latest/) pour l'alerting, car payant dans ELK.

Au final mes logs font : 
 `Home-assistant -> docker -> fileBeat -> logStash -> elasticSearch ( stockage) -> kibana (restitution) `

C'est un peu sortir le bazooka pour tuer une mouche, mais ça a plusieurs avantages : 
- Je découvre ELK et j'apprend à l'utiliser ( autodidacte)
- Scalable à d'autres besoins futurs ( détection d'intrusion sur mon réseau ect )  

je souhaite maintenant monter un serveur webhook qui ferait la passerelle entre requêtes HTTP  et actions sur le système : 
- Collection des logs
- Détection via sentinl des logs "Failed to connect, retry in 64 sec"
- Déclenchement d'un Webhook ( si plus de 2 occurrences ) qui va déclencher un script qui s'occupera de débrancher et rebrancher virtuellement la zigate et redémarrer mon docker home-assistant

Désolé c'est pas complètement la même problématique, mais ça peut servir à certains ayant le même genre de problème. Sachant qu'avec docker la chaîne elk se monte en 6 minutes, sentiln en 3 et webhook en 2 minutes... ça peut servir je pense :)

max5962, 2018-12-09:
Zigate is plug to a bad USB plug ...
I close the issue

closed 2018-12-09 by max5962