fairecasoimeme/ZiGate #80: Support of the new IKEA power socket Tradfri

yvin, 2018-11-21:
IKEA just launch a new power socket
Do you planned to support it ?
https://www.ikea.com/fr/fr/catalog/products/20364476/
BR

fairecasoimeme, 2018-11-22:
Hi,
I'll receive the device this week (normally) but it's seems to be supported like all other Tradfri devices.
BR,

pipiche38, 2018-12-01:
If that is this one, it works with Zigate and the Domoticz-Zigate plugin
'TRADFRI control outlet':{'Ep':{'02':{'1000':'','0021':''},'01':{'0006':'','fc7c':'','0004':'','0000':'','0003':'','0008':'','0005':''}},'Type':'Plug','ProfileID':'c05e','ZDeviceID':'010a'},

ruimarinho, 2018-12-01:
I bought one last night so maybe tomorrow or Monday I can give it a try as
well.

pipiche38, 2018-12-01:
Great and let us know

KiwiHC16, 2018-12-02:
Just integrated it in Jeedom plugin Abeille and works like a charme.

closed 2019-03-06 by ISO-B