fairecasoimeme/ZiGate #81: Problème avec ZiGate Wifi

Mific78, 2018-11-24:
Bonjour à tous,

Je viens de recevoir ma ZiGate Wifi et j’ai des soucis pour le faire fonctionner.

1/ Je n’arrive pas à modifier la configuration WiFi pour pouvoir connecter la ZiGate sur mon réseau. Je fais « Save » puis « Reboot » mais le configuration n’est pas prise en compte (le SSID reste en ZIGATE-XXXX, l’@IP rest à 192.168.4.1, etc.).

2/ Quand je fais un « Send » de « 01 02 10 10 02 10 02 10 10 03 » avec PacketSender (en restant en mode Point d’Accès), la commande est envoyée mais je n’ai pas de réponse de la ZiGate Wifi.

Merci d'avance pour votre aide !

pipiche38, 2018-12-01:
Ton réseau écoute-t-il sur le 2.4 G ? et pas que sur le 5 ?

En effet si on bout d'un certain temps l'etage Wifi ne s'accroche pas à l'AP donné il retombe sur sa comfig d'origine.

donc je chercherais : 
- Est-ce que l'AP est joiniable en 2.4
- Est-ce le bon mot de passe
- est-ce que l'AP n'a pas des securité ( autorise que certain devices sur leur MAC ...)
- ...

pipiche38, 2020-09-30:
This issue has been closed due to no activity since while!

closed 2020-09-30 by fairecasoimeme