fairecasoimeme/ZiGate #82: Firmware 3.0f

ruimarinho, 2018-11-25:
Hi,

Could you publish the binary for 3.0f please? 

Thank you,

Smanar, 2018-11-25:
It's note same ?
https://github.com/fairecasoimeme/ZiGate/tree/master/Module%20Radio/Firmware/src/ZiGate/Source

There is a chnaed file inside with "v3.0f" in comment.

ruimarinho, 2018-11-25:
Yes, but there is no binary yet (only [3.0e](
https://github.com/fairecasoimeme/ZiGate/blob/master/Module%20Radio/Firmware/bin/ZiGate/ZiGate_Coordinator_v3.0e.bin)).

pipiche38, 2018-11-25:
From the source I didn’t see yet the update to handle 24bits dataType so I suspect 3.0f is not yet finished

Smanar, 2018-11-25:
Ha yep, sorry.
But now the code is open source, so we will have more often modifications without official release for beta testers.

ruimarinho, 2018-11-25:
Ok, got it. Will close the issue. Do I need to Erase EEPROM and Complete EEPROM when upgrading to 3.0e?

Also, do I need to unpair devices before upgrading?

closed 2018-11-25 by ruimarinho