fairecasoimeme/ZiGate #84: Adding a CC2531 as a router

ruimarinho, 2018-11-27:
I am planning on adding a new router to the network, either by using a new socket device or by flashing a CC2531 in router mode.

Would that work seamlessly with Zigate? Do I need to pair the CC2531 with Zigate?

KiwiHC16, 2018-11-27:
"Would that work seamlessly with Zigate?" -> It should if ZHA, but with compatibility issus, there is no firm answer. Use device with 220V power supply and known to be zigate compatible will be the easies solution.
"Do I need to pair the CC2531 with Zigate?" -> Yes

ruimarinho, 2018-11-27:
Are all light bulbs routers or it depends on the manufacturer?

KiwiHC16, 2018-11-27:
All bulb I used are. (Ikea, Hue, Osram,...)

pipiche38, 2018-11-27:
My 2 cents, up to now, all Main Power devices I use are 'router'. I see that from the answer  read Attribute of those devices as well as from the Network Topology I'm getting out of LQI reports.

Those devices are : 
- Profalux shutters
- Xiaomi plug
- Salus plug
- Legrand plug
- IKEA Tradfri Bulbs

ruimarinho, 2018-12-02:
@pipiche38 how are you querying for LQI?

To have an existing device connect through a recently-paired device (ie.g. an IKEA Tradfri bulb), is there anything that needs to be done on the device or on zigate itself to start routing messages through it or is it part of the meshing routines of ZigBee?