fairecasoimeme/ZiGate #86: About OTA

ISO-B, 2018-11-28:
Should zigate be able to work as OTA server? I am using firmware 3.0e. I have been trying to get Ikea trådfri lamp to fetch ota updates from zigate. For me it seems that 0x0505 command doesn't do anything.

pipiche38, 2018-11-28:
That would be awesome. But where can we get the firmware?

ISO-B, 2018-11-29:
At the moment you can update Ikea stuff using deCONZ using their conbee or rasbee sticks. Their rest api repo has [python script](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/ikea-ota-download.py) that show how to download latest ota files from Ikea. 

As far as I understand OTA process works two ways.
1. Devices are quering server from time to time for new images
2. Server can send notify image command for device(s) to force it to ask ota image

So far I havn't seen any devices quering images on their own. Also using notify image command 0x0505 doesn't make any difference. Zigate gets status success message after that command but nothing else happens. Might be that I have wrong syntax and that why lamp doesn't ask for ota image. Figured it would be easier to ask that are these commands and features even implemented on current firmware rather than banging my head against wall for many hours.

ISO-B, 2018-12-14:
Just little update for this. 
Managed to build debug version of ZiGate firmware which makes debugging this easier. Noticed that devices actually request OTA files time to time and usually on startup. After that I had big facepalm moment when I found out that I have been using wrong OTA image. Now I am able to start updating process. Next things is to improve python library to get OTA file sent to client. I can confirm that OTA is implemented and it is working.

Would it be helpfully to get QUERY_NEXT_IMAGE_REQUEST messages passed to serial line?

pipiche38, 2018-12-14:
Excellent news !
Where do you find the OTA images ?

doudz, 2018-12-14:
great news !

ISO-B, 2018-12-17:
Managed to update Trådfrin 980lm E27 bulb firmware with ZiGate using python library. You can probably use TestGui to do same as well.

> 
> 
> Excellent news !
> Where do you find the OTA images ?
deCONZ have python script for downloading ota files from ikea. You can also use deCONZ to convert ikea ota files to standard ota files. Load ikea file using STD OTAU Plugin. Open it from OTAU File tab and save it and its converted to standard ota file.

closed 2018-12-17 by ISO-B

KiwiHC16, 2018-12-17:
Could you share a quick doc on how to do it (direct liks to source, steps to follow,...)

pipiche38, 2018-12-17:
@ISO-B 
> 
> 
> **Downloading and converting ikea ota files**
> 
Thanks a lot for sharing

Jsl-1, 2019-02-25:
Hi, 

I tried to update an ikea tradfri outlet using ota file.
Step : 
  - Download ota image from ikea http://fw.test.ota.homesmart.ikea.net/Tradfri_OTA_release_signed_2019_01_30_154434/bin/10005777-3.1-TRADFRI-control-outlet-2.0.019.ota.ota.signed
  - Convert ota file with deCONZ V2_05_20 (Windows)
  - Move file to my home assistant installation using smb
  - use python library to call ota_load_image('10005777-3.1-TRADFRI-control-outlet-2.0.019.ota.ota.signed')

I get this: Something wrong with ota file header.
I use python library 2.8.0 and zigate USB TTL with firmware 3.0f

Here is the debug log for command 0x0500: 

DEBUG:zigate:REQUEST : 0x0500 b'\x02\x00\x00\x0b\xee\xf1\x1e\x01\x00\x008\x00\x00\x11|\x11\x01 \x01\x96#\x00\x02EBL tradfri_control_outlet      \x00\x03\x19\xbe\x00\x00\x00\x00\x00\x03\x19\x80\x00\x02\x8c\xe3\x02'
DEBUG:zigate:Msg to send b'05000048a70200000beef11e010000380000117c110120019623000245424c20747261646672695f636f6e74726f6c5f6f75746c6574202020202020000319be000000000003198000028ce302'
DEBUG:zigate:Encoded Msg to send b'0102150210021048a7021202100210021beef11e0211021002103802100210117c11021120021196230210021245424c20747261646672695f636f6e74726f6c5f6f75746c65742020202020200210021319be0210021002100210021002131980021002128ce3021203'
DEBUG:zigate:Waiting for status message for command 0x0500
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15f\x02\x10\xd3\x02\x100\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'00d30030'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:211, packet_type:48, error:b'', lqi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0030:RESPONSE 0x8000 - Status response : status:0, sequence:211, packet_type:48, error:b'', lqi:0
DEBUG:zigate:Waiting for message 0x8030
DEBUG:zigate:Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15\x80\x02\x10\x02\x10\x02\x15\x02\x10\x02\x10\x03'
DEBUG:zigate:Dispatch ZIGATE_PACKET_RECEIVED
DEBUG:zigate:Received response 0x8000: b'00000500'
DEBUG:zigate:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:1280, error:b'', lqi:0
DEBUG:zigate:Dispatch ZIGATE_RESPONSE_RECEIVED
DEBUG:zigate:STATUS code to command 0x0500:RESPONSE 0x8000 - Status response : status:0, sequence:0, packet_type:1280, error:b'', lqi:0
WARNING:zigate:Something wrong with ota file header.
WARNING:zigate:Cannot read ota header. No ota file loaded.

I tried with other files from http://fw.ota.homesmart.ikea.net with same result
Did I miss something ?

Thank you for your help.