fairecasoimeme/ZiGate #89: Disparition des Modules

deennoo, 2018-12-01:
FW 3.0e+modif pour le 24bit
Plugin Zigate pour Domoticz

Il est assez fréquent que mes devices a piles Xiaomi ne soit plus reconnu après un certains temps (non determiné pour le moment), je suis obliger des les reinclures apres x heures sans utilisation : 

- lumi.sensor_switch.aq2
- lumi.sensor_86sw2
- lumi.sensor_86sw1
- lumi.sensor_cube
- lumi.sensor_wleak.aq1
- lumi.sensor_motion.aq2

le plus flagant etant pour les 86sw1 & 2 ou apres une nuit de non utilisation, si je clique l'un des bouton, les deux led clignotes plusieurs fois ce qui indique qu'ils n'appartiennet a aucun reseaux zigbee, je dois donc les reinclure.

Suis je seul a rencontrer ce problème tous contrôleur confondu ?

pipiche38, 2018-12-01:
A priori cela pourrait être identifiable au niveau de la Zigate quand elle gère la table de mesh ( LQI ).
Le tout est de voir comment reconnaître un router qui merde ;-)

KiwiHC16, 2018-12-02:
@deennoo, si tu pers le équipements il est fort probable qu'ils fasses un "Leave" (si la radio n'est pas bonne il est possible que tu ne le reçoives pas).

Reçois tu les Leave ?

Smanar, 2018-12-02:
Plutot le device, surement pour une fonction leave/rejoin.
Le clignotement sur le device indique qu'il n est plus "paire" alors qu'il est encore present sur le coordinateur.

KiwiHC16, 2018-12-02:
Mon experience depuis le debut est qu'on pert le équipements lorsqu'ils n'ont pas réussi à discuter avec le réseau pendant un certain temps (probablement 2 ou 3 heures). Problème de couverture radio, zigate HS du fait d'une coupure de jus,...  Ils envoient un message et ne reçoivent pas de message Ack d'un des équipements sur le réseau. Alors ils se déconnectent (Leave) et cherchent à se connecter mais la zigate/réseau n'est pas en mode inclusion, résultat équipement exclu du réseau. 
C'est l'équipement qui envoie un Leave, qui est transmis (si reçu) par la zigate au plug in.
Dans mon plugin Abeille, sur reception d'un Leave je désactive les équipements afin que cela soit visible dans Jeedom et j'envoie même un sms sur mon tel. Depuis que j'ai une bonne couverture radio, je n'ai plus de Leave.

deennoo, 2018-12-03:
@smanar

Effectivement j'ai recommencé mon appairage en commençant par : 
- une Ikea,
- la prise Ikea
- puis un Xiaomi ln2
- puis les modules a pile

Résultat : plus de deconnection

Il faut que je chech le mesh pour voir comment tout cela est en réseau.

pipiche38, 2018-12-03:
Je suis de plus en plus persuader que le niveau de mesh est important sur la stabilité du réseau et des devices.
Si je fais le parallèle avec Zwave, même les devices sur batterie participent au mesh. ce qui malheureusement impact la consommation des batteries!

KiwiHC16, 2018-12-03:
"Si je fais le parallèle avec Zwave, même les devices sur batterie participent au mesh" 
Je ne suis pas sur de comprendre. Indiques tu qu'en zwave le devices sur piles participent au routage ?

KiwiHC16, 2018-12-03:
"Il faut que je chech le mesh pour voir comment tout cela est en réseau." dans  le plugin Abeille/Jeedom j'ai développé une interface graphique pour le visualiser.

pipiche38, 2018-12-03:
> Mon experience depuis le debut est qu'on pert le équipements lorsqu'ils n'ont pas réussi à discuter avec le réseau pendant un certain temps (probablement 2 ou 3 heures). Problème de couverture [...]

Ca c'est dans le cas ou le Device ne reçoit plus les beacons du contrôleur. Mais ce que j'ai compris d'@Akila, c'est que c'est également vrai dans l'autre sens :  si la Zigate ne reçoit plus de message de l'objet alors celui-ci est retiré de la Zigate (après un Timeout) => D'où le firmware 3.0e avec une extension du TimeOut. 
Dans ce dernier cas, je doute que la Zigate génère un Leave. 
Et je pense que dans le cas des personnes qui perdent des devices, c'est celui-ci .

Ce que j'observe en échangeant avec ces internautes, c'est que : 
- Ils ont un réseau Zigbee pratiquement à 100% avec des objets sur batterie
- que comme tu l'indiquais , le niveau d’interférence (tel que rapporté par network management) et relativement élevé et qu'un changement de canal du Wifi semble avoir un effet positif.

Du reste, je suis perplexe sur l'utilisation du channel 0, car j'observe que la Zigate dans ce cas là ce positionne sur le channel 11 ce qui n'est pas forcément le plus optimum au regard du niveau d’interférence observé !

KiwiHC16, 2018-12-03:
"beacons du contrôleur.":  Beacon Zigate vers device: ca n'existe pas à ma connaissance. Jamais vu dans aucune trace.

"si la Zigate ne reçoit plus de message de l'objet alors celui-ci est retiré de la Zigate (après un Timeout)" : je suis en ligne. Si la zigate ne recoit pas d'info d'un device alors elle le retire de sa base.

Si le device revient apres trop longtemps, la ZiGate lui envoie un Leave, le device repond avec un Leave (pour dire qu'il a bien pris en compte la demande) puis cherche a se reconnecter mais le reseau est fermé. Le Leave venant du device est transmis au travers de la zigate vers le plugin. Ca je l'ai vu plein de fois dans les échanges.

"Ils ont un réseau Zigbee pratiquement à 100% avec des objets sur batterie" dans ce cas ils auront forcement des soucis si les distances sont un peu grande, ou presence de mur, de plantes,...

"channel 0": c'est la zigate qui choisi au demarrage entre 11, 15, 20 et 25 sur la base de sa mesure de bruit. Si tu regardes ces cannaux ne sont pas choisis par hasard, mais ils sont entre less canaux wifi 1, 6, 11 qui sont les plus utilisés, et/ou si tu fais un plan radio digne de ce nom.

pipiche38, 2018-12-03:
Un papier interessant qui donne pas mal de détail, sur le mesh, une des raisons du changement de Network address

doudz, 2018-12-03:
Je poste mon retour d'expérience
J'ai rarement eu des pertes d'équipement, mais ça vient de se produire à l'instant alors que je relançais pour 3 ou 4e fois mon soft, j'ai eu leave d'un équipement avec une demande de rejoin.
Ce qui m'étonne c'est que l'équipement en question est actuellement poser à 10cm de la zigate donc c'est pas un problème de réception...

Du coup j'ai pensé à qqc, combien de temps est valable de la demande rejoin ? si je reçois un Leave avec demande de rejoin, que se passe-t-il si j'active le permit join à ce moment là ?
Quelqu'un à essayer ? peut-être que l'équipement se ré-associe ?

Je ferai bien des tests mais ça m'était pas arrivé depuis 3 ou 4 mois de perdre un équipement et j'avais mis ça sur le compte de la distance car l'équipement se trouve loin de la zigate

doudz, 2018-12-03:
C'est la loi de murphy, 5 Leave en quelques minutes, alors que tout allait bien depuis des mois
```
2018-12-03 13:42:29 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8048 - Leave indication : ieee:00158d00016c47f5, rejoin_status:1, rssi:0
2018-12-03 13:50:51 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8048 - Leave indication : ieee:00158d0001b7b04a, rejoin_status:1, rssi:0
2018-12-03 13:57:40 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8048 - Leave indication : ieee:00158d0002a229fb, rejoin_status:1, rssi:0
2018-12-03 14:02:22 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8048 - Leave indication : ieee:00158d0002564e08, rejoin_status:1, rssi:0
2018-12-03 14:04:42 DEBUG (ZiGate-Event Loop) [zigate] RESPONSE 0x8048 - Leave indication : ieee:00158d000277338c, rejoin_status:1, rssi:0
```

pipiche38, 2018-12-03:
> "beacons du contrôleur.": Beacon Zigate vers device: ca n'existe pas à ma connaissance. Jamais vu dans aucune trace.
> 
Normal, car ça n'a aucune raison d'etre remonté. C'est la stack ZigBee qui gère cela. Voir l'implémentation faite par NXP, mais à mon sens c'es le propre du controleur et des routers, car sinon comment s'adapteraient-il à la topologie du réseau.

> "Ils ont un réseau Zigbee pratiquement à 100% avec des objets sur batterie" dans ce cas ils auront forcement des soucis si les distances sont un peu grande, ou presence de mur, de plantes,...
> 
CQFD

> "channel 0": c'est la zigate qui choisi au demarrage entre 11, 15, 20 et 25 sur la base de sa mesure de bruit. Si tu regardes ces cannaux ne sont pas choisis par hasard, mais ils sont entre less canaux wifi 1, 6, 11 qui sont les plus utilisés, et/ou si tu fais un plan radio digne de ce nom.

C'est effectivement ce que j'attend, mais en utilisation j'ai des doutes. Pourquoi:
 J'ai 2 Zigate à la maison, une pour ma prod et une pour mes tests et dev/ La prmière est bien sur le 11, car fixe , la deuxieme est sur 0, donc devrait prendre celui avec le moins d'interférence, et je ne le vois pas. J'ai plutot l'impression que lorsque l'objet fait son broadcast pour un Annouce celui arrive sur le channel qui était précèdement et la zigate suis.

pipiche38, 2018-12-03:
@doudz, Sébastien je peu juste partager avec toi ce qui est fait sur le plugin Domoticz, quand on reçoit un Leave d'un objet, l'objet dans le plugin est simplement désactivé. Ce qui fait que lorsque après le Leave on reçoit le device annouce on a juste à refaire la connexion avec la nouvelle Short address. Du coup on s’affranchit du Permit to Join dans la mesure où l'ensemble des informations à déjà été obtenues.

doudz, 2018-12-03:
@pipiche38 oui je comprends mais tu dois refaire l'association de l'objet (je veux dire appuyer physiquement sur le bouton de l'objet)
mon idée était qu'en activant le permit join, l'objet revenait sans intervention humaine

KiwiHC16, 2018-12-03:
"objets n'ont pas d'adresse IEEE" c est normal. En considerant qu'il n'y a pas de bug. La commande t'envoie la liste des IEEE qu'elle connait/utilise (c est un peu la table arp dans la stack IP). Si l IEEE ni est pas il est fort probable que le device soit routé.

Ce qui est important, ce n'est pas la distance mais qualité radio. Donc meme si ton device est à 20cm de la zigate, peut etre passe t il par l ampoule qui est dans une autre piece pour dialoguer. J'ai vu des choses tres surprenantes... si cette ampoule disparait alors tu perds ton device... et autre scenarii.
5 en meme temps, peut etre as tu eteints un routeur...

"combien de temps est valable de la demande rejoin ?" quelques secondes. Tu peux tenter le coup de mettre en inclusion sur reception d'un Leave... Mais ca ne resoudra pa le probleme de fond qui est que la radio n'est pas bonne.

Un autre scenarion de $ç!s!è§ (pour etre poli) est l'interrupteur double Xiaomi qui envoie des beacon incomplete et qui provoque un PAN ID change et alors la c'est pas beau à voir dans beaucoup de cas.

"sinon comment s'adapteraient-il à la topologie du réseau": la zigate n'a pas besoin de la topology du réseau, tout s'organise en fonction des besoins. routage dynamique.

"J'ai plutot l'impression que lorsque l'objet fait son broadcast pour un Annouce celui arrive sur le channel qui était précèdement et la zigate suis." La zigate scan au démarrage les canaux 11, 15, 20 et 25, elle choisi le canal le moins brillant (de son point de vue) et y reste tout le temps (Sauf scenario que je n'ai pas encore rencontré). Un device qui arrive fait un beacon sur tous les canaux les uns apres les autres jusqu'a ce qu'il recoive un message lui indiuant qu'il y a un reseau sur le canal et qu'il est ouvert. La il s'associe. C est le device qui s'aligne sur la zigate.

pipiche38, 2018-12-03:
j'ai eu un cas similaire ce week-end ou il y avait des IEEE en 0x00000000 ! La zigate etait en 3.0e !

doudz, 2018-12-03:
Je suis aussi en 3.0e.
Le seul router que j'ai c'est une ampoule OSRAM, qui fonctionne nickel et je n'éteins jamais (d'autant que je ne suis pas sur place :D)
En fait voici ce que j'ai fait aujourd'hui :
Je testais la commande LQI Management et j'ai demandé un index plus grand que le nombre d'"entry" (en demandant l'index 0, j'avais obtenu entry:15 et j'ai demandé index: 0F) et là la zigate n'a plus répondu à plus rien, j'ai donc simplement relancé mon soft (je ne suis pas surplace donc pas de débranchement physique de la clé)
C'est à ce moment là, au redémarrage que les Leave ont commencé.

D'où question :
Est-il possible qu'en l'absence temporaire de la zigate (absence relative puisque c'est simplement une déconnexion/reconnexion du port série), les objets ont tenté de passer par l'Ampoule (Router) et ont envoyé un Leave ?

doudz, 2018-12-03:
> Si l IEEE ni est pas il est fort probable que le device soit routé.
Et comment trouver le router ?

pipiche38, 2018-12-03:
@doudz  A mon sens, t'es sur le bon chemin avec LQI ( si tu veux on peux echanger, je suis joiniable sur keybase.io @pipiche)

KiwiHC16, 2018-12-03:
Difficile a dire mais qq éléments:
- si les equipements perdent leur parent: un routeur (ca peut etre la zigate) car enlevé du réseau, éteint, sans réponse temporairement, ils vont chercher à se reconnecter en s'attachant a un autre routeur.
- si il cherche a se reconnecter, il envoie de memoire un Leave suivi d'un beacon. Il faudrait que je revérifie.
- exemple, ils sont connectés à la zigate qui ne leur repond pas, ils vont chercher à se raccrocher ailleurs. Apres ce va dependre des timers dans les devices. J'ai vu des comportement différents.
- Si la zigate ne repondait plus peut etre ne repondait elle pas non plus coté zigbee et donc les devices n'avait pas de réponse.
- Avec LQI tu va pouvoir avoir les parents mais (car il y a un mais) tu n'auras pas le routage des messages. Tu auras le premier routeur utilisés uniquement. (Dans ton cas si tu en n'a qu'un, c est facile...
- J'ai dans ma todo list de couper le jus de la zigate pour voir ce qu'il se passe coté device.

doudz, 2018-12-03:
Merci @KiwiHC16 

Vous avez lu les pages 87 et 88 de ce document ? https://www.nxp.com/docs/en/user-guide/JN-UG-3048.pdf
ça parle de Leaving and Rejoining the Network, il semblerait qu'on puisse demander un rejoin à un objet qui n'a pas réussi à faire le rejoin tout seul (ZPS_eAplZdoRejoinNetwork)

doudz, 2018-12-03:
Ce que je comprends :
ZPS_eAplZdoRejoinNetwork : Permet de demander à un node de faire un rejoin (c'est intéressant pour récupérer un node perdu je pense)
ZPS_eAplZdpMgmtDirectJoinRequest : Permet de demander à un parent (un coordinateur) d'accepter un node en particulier (peut être utile dans le cas où le coordinateur n'est pas la zigate)
ZPS_eAplZdpMgmtPermitJoiningRequest : Permet d'activer/désactiver la possibilité de faire un join sur un node (je vois pas trop l'intérêt, peutêtre blacklister un node)

KiwiHC16, 2018-12-03:
Je suis d'accord sauf pour le premier. Ma comprehension est différente. Prenons un capteur de porte xiaomi. Si le capteur veut faire un rejoin alors le firmware xiaomi appel ZPS_eAplZdoRejoinNetwork (s'ils utilisent la stack NXP). Mais dans notre cas, depuis la zigate on ne peut pas utiliser cette fonction.

doudz, 2018-12-03:
En tout cas la doc est claire :
- Si le node fait un `rejoin`, il n'y a pas besoin de faire un `permit join`
- Sur un `rejoin`, le coordinateur doit faire un `Device Announce`

```
Note 1: When a device rejoins a network, the ‘permit
joining’ status on the potential parent is ignored.
Note 2: When a device joins the network, its application
may call ZPS_eAplZdpDeviceAnnceRequest() to
announce the device’s membership and network
address to the rest of the network. The information is
sent in a Device_annce announcement, which must be
collected by the recipient nodes using the RTOS
function OS_eCollectMessage().
```

KiwiHC16, 2018-12-03:
Je suis perdu dans qui fait quoi dans tes explications et je ne pense pas être aligné (peut être que je loupe un truc). Si on regarde le protocole et le messages échangés pour un appairage normal:
Node->broadcast: beacon
ZiGate->Node: network open (si inclusion mode)
(échange divers comme adresses)
Node->Broadcast: annonce
...

Pour le rejoin je dois regarder des traces car je ne me souviens pas des details.

Smanar, 2018-12-03:
Pour avoir fait quelques tests, avant de laisser tomber car j'etais tomber sur un truc qui tourne.

Pour avoir tester un device xiaomi (le fameux double inter).
Non, quand un device a des prbs avec un routeur, il n'essaye  pas forcement de se connecter a un autre.
Non, lors du commisssioning il n'essaye pas forcement de se connecter avec le device qui a la meilleure connexion.

Systematiquement il essaye de se connecter a un inter legrand, alors qu'il est a 
80 cm de la zigate
3m de l'ampoule ikea
5m de ce put1 d'inter.
Evidement ca marche quelques heures puis out.

Tandis qu'en le forcant a passer par l'ampoule ikea, plus de prbs, meme avec des coupures ponctuelles de l ampoule (he oui j'ai encore l'inter de fonctionnel, donc qui passe l'ampoule a off, mais c'est pas des coupures d'1h non plus)

max5962, 2018-12-16:
Hello, 
La perte de module ne se contente pas du xiaomi mais la prise ikea est sortie de mon réseau (n'interagit plus) pour aucune raison  : (
J'ai l'impression que cest plus recurrent depuis le firmware 3f.

pipiche38, 2018-12-16:
Tu veux dire 3e ?

deennoo, 2019-01-02:
De mon côté je suis passer en Fw 3.0f et c'est pas mieux.

Quasiment tout mes modules a piles ont disparus après 4 jours d'absence.

Comme si il était en sommeil (certains au bout de 2 heures des fois). Cela arrivé également sur des modules Xiaomi 230v (interrupteur) ils perdent le réseau environ 5 secondes par jour ( je le vois car l'ampoule qui est relier a linterrupteur flash tant que le device n'est pas reinclu dans le réseau)

Lors du réveil, ils changent de short ID.

Aucun message de leave dans les logs de la Zigate avec le plugin pour Domoticz.

max5962, 2019-01-11:
Hello,

Idem, 11 modules de perdu ( je suis en 3f_pre_release ) xiaomi ou ikea ( pile ou sur secteur )

doudz, 2019-01-11:
@max5962 est-ce que tu perds les modules après coupure d'alimentation la zigate ? (via un débranchement physique ou redémarrage de la machine ce qui coupe l'alimentation des ports USB)
Ou est-ce que lors d'un redémarrage du soft (Home Assistant ou Jeedom ou autre)
Ou est-ce en fonctionnement normal ?

Perso je n'ai "rien" perdu depuis le passage en 3.0f, un seul capteur de porte Xiaomi mais comme avant peut importe la version du firmware, et c'est un capteur situé (trop) loin à l'extérieur. Par contre je n'ai fait aucune coupure d'alimentation de la zigate depuis, uniquement de redémarrage de Home Assistant.

doudz, 2019-01-11:
Attention issue en double https://github.com/fairecasoimeme/ZiGate/issues/38

max5962, 2019-01-11:
@doudz
Je ne pourrai pas te dire. Je ne touche pas/peu a la zigate (mise a part mes erreurs récurrents de mon port USB ce qui la deconnecte 2 ou 3 fois / semaine)

J'ai une cron qui reboot le port serial.
J'ai la même supposition que toi, c'est dommage que nous ne puissions pas mettre en "full debug" la zigate le temps de la rendre fiable de ce point de vue.

Finalement un simple broker comme zigbeeMQTT aurait été plus efficace pour du debug.

max5962, 2019-01-19:
@fairecasoimeme  
Vous avez une piste pour ce comportement ? Je sais c'est très complexe à résoudre sans log et erreur précise, mais ça rend le système peu fiable ce qui rend ma zigate un peu inutile :( ( le WAF passe moins bien :) ) 

Un poolling régulier serait pas la solution ( temporaire) ça spammerait le réseau mais un dirty fix c'est toujours un fix.

Avec mon problème de port USB, il est possible que la zigate soit indispo pendant maximum 1minute30 ( j'ai une cron qui détecte et corrige ) 

@doudz Il y aurait moyen d'avoir une méthode dans home-assistant de ce re-synchro avec ce qu'il y a dans la zigate et faire abstraction du cache/fichier zigate ect. Histoire de connaitre la liste des module à ré associer ?

:)

merci pour vos réponses.

fairecasoimeme, 2019-01-20:
@max5962 : je pense qu'il y a 2 problèmes:
- le premier lié à la version 3.0e qui engendre un problème mémoire dans le stockage des données. Pour exemple, quand vous faites un discover devices, il n'est pas rare que le tableau passe (par exemple) de l'id 4 à 40/41/50/Etc... avec des données incohérentes sur l'adresse IEEE

- le deuxième lié à une qualité de réseau / mesh moyenne et des devices avec une gestion spécifique. Et c'est là que c'est dur de gérer parce que souvent, c'est une sorte d'incompatibilité avec le device (comme Xiaomi, par exemple, qui a un comportement étrange sur les leave/rejoin). Sur ce problème, je n'ai pour le moment pas trouvé d'autres solutions que d'avoir un mesh plus important. Mais j'y travaille :)

max5962, 2019-01-20:
Hello,

Effectivement la version 3.0e n'était pas super comme release ^^

Concernant le mesh, c'est étrange dans le sens où j'ai autant de disparitions avant qu'après l'ajout de prises Ikea dans mon réseau. Je pense qu'une vrai visualisation, comme on a sur du zwave, du réseau serait un vrai plus. (Là c'est au développeur des box de bosser :) )

pipiche38, 2019-01-20:
Dispo avec le plugin Domoticz-Zigate sur la branche 'beta' actuellement

sifus, 2019-03-28:
Hello je suis en 3f et après un redémarrage de Jeedom tous mes modules Xiaomi ont perdu leurs affectations (je me retrouve avec plein de modules avec un nom par defaut et aucun en objet)

J'ai du faire un backup de mon Jeedom pour retrouver le tout comme avant.

nouknouk, 2019-08-15:
> Je suis de plus en plus persuader que le niveau de mesh est important sur la stabilité du réseau et des devices.
> Si je fais le parallèle avec Zwave, même les devices sur batterie participent au mesh. ce qui malheureusement impact la consommation des batteries!

Je déterre et pollue un peu le thread, mais juste pour info: 

(à moins qu'il existe quelques exceptions pour des devices **très** spécifiques qui me sont inconnus jusque là) **non** les devices zwave sur pile **ne** participent **pas** au routage dans le réseau ; seuls les modules alimentés en permanence sur secteur sont des routeurs.

biscwii, 2019-11-03:
> Par contre pas de solution technique, j'ai du prendre le device qui deco, le mettre contre la lampe, l’appairer et le remettre a sa place apres.

On peut forcer le mesh de cette façon? Il faut être en permit to join sur la zigate?

pipiche38, 2019-11-03:
A Mon sens non, car même si tu forces la route au premier appairage, au fil du temps cela va évoluer et il n'y aucun controle dessus.

@biscwii , pour répondre à ta question, oui il faut etre en permit to join, car tu te mets en situation d'appairage de ce device dans le réseau zigbee

Smanar, 2019-11-03:
@biscwii Yep.

Comme l'a dit @pipiche38 tu dois etre en permit join.
Et cela ne marche (testé de mon coté) que avec xiaomi, car ces appareils ne changent jamais de parent, une fois "meshé" a une lampe, il ne bouge plus.
Autre chose chose, avec la pratique j'ai remarqué autre chose, c'est que les capteurs Xiaomi sont plus actifs durant les premières heures, par exemple la detection du capteur de presence est plus rapide, et surtout durant cette période il peut encore changer de parent. Donc il te faut mettre en permit join, forcer le maillage, et le laisser en place quelques heures, apres ca, il ne bougera plus.

biscwii, 2019-11-03:
Merci beaucoup @pipiche38 et @Smanar pour vos réponses, je vais tester ça.
J'ai effectivement des détecteurs de porte Xiaomi qui se déconnectent régulièrement, et j'ai acheté un certain nombre de prises Ikéa pour augmenter le maillage du réseau. J'attends en vain depuis plus d'une semaine que les capteurs de porte se raccrochent à une prise située dans la même pièce.

Question bête : J'ai quand même l'impression que depuis que j'ai installé les prises Ikéa, le LQI des détecteurs à chuté. Est ce que si un détecteur n'utilise pas le routeur à proximité, mais plutôt la zigate située plus loin, ce routeur va brouiller le signal entre la zigate et le détecteur?

Smanar, 2019-11-03:
Nope, je ne pense pas, jamais vu de brouillage.

Par contre, je ne peux pas dire pour les autres capteurs a piles, mais Xiaomi sont tres mauvais niveau routage, ca ne sert a rien d'attendre que les capteurs se raccrochent a un routeur plus pres, une fois qu'ils ont choisit, ils ne bougent plus, meme si le parent disparait. Pour avoir une chance de les voir bouger, il te faut remettre la passerelle en permet join, et refaire une procédure d'inclusion.

biscwii, 2019-11-03:
Un de mes détecteurs d'ouverture Xiaomi vient de se déconnecter (celui qui est à 1m50 de la zigate, bizarre), alors j'en profite pour tester ta méthode. Je lance le mode Permit to Join, je place le détecteur d'ouverture juste à coté d'une prise Ikéa, je lance l'inclusion avec un trombone sur le détecteur. L'inclusion se passe bien, je lance un get LQI et un Network graph sous abeille : le détecteur est toujours accroché à la zigate, et pas à la prise Ikéa qui se trouvait juste sous son pif. Je dois louper un truc, mais je ne vois pas quoi.

Smanar, 2019-11-04:
Non non, je pense que tu es bon, désolé, mais quand ca veux pas, ca veux pas ^^.
Moi je fais ca avec des lampes, d'ikea aussi. Peut etre que a cet endroit ta passerelle est deja trop puissante comparativement au plug, la je peux pas te dire.

biscwii, 2019-11-04:
C'est bon, ouf !
Je ne devais pas être suffisamment prêt la premiere fois. Je l'ai vraiment refait avec le capteur touchant littéralement la prise et ça a fonctionné (après tout, dans la notice Ikéa, lors des appairages de télécommande, il faut être à moins de 5cm).
A voir si ça tient dans le temps !

pipiche38, 2020-09-30:
This issue has been closed due to no activity since while!

closed 2020-09-30 by fairecasoimeme