fairecasoimeme/ZiGate #92: 0x8701 - Route Discovery

pipiche38, 2018-12-03:
I'm wondering if the 0x8701 message documented in zigate.fr is not inversed in terms of Status and Network Status

Indeed when receiving such messages, the first uint8 looks to be the Network Status, while the Status looks to be the 2nd uint8

For instance I'm getting several 0x8701 where status is equal to 0xd0, which is a NWK Codes and not a MAC Codes as per the documentation ( ZigBee PRO Stack User Guide.

Here after is one exemple of recurring Status Code I'm getting and which looks as a Network Status:
 d0 - Attempt at route discovery has failed due to any reason except lack of table space

doudz, 2018-12-04:
@pipiche38 the good thing with source on github is that you can get the answer just looking the code...
https://github.com/fairecasoimeme/ZiGate/blob/8948430358c587bd5afd2e89bec5d89d6710dc03/Module%20Radio/Firmware/src/ZiGate/Source/ZigbeeNodeControlBridge/app_general_events_handler.c#L904

pipiche38, 2018-12-04:
I'm not sure I understand the answer , as the source you are referring do not explain the mix

doudz, 2018-12-04:
Just to show that there's no mix in ZiGate code.
Maybe it would help if you could provide the full zigate response message

pipiche38, 2018-12-20:
Here are few examples : 

018701000333d0006603
018701000301d0005403

So what I'm suspecting is that the 2 uint8 are grouped in a 1 uint16 may be @fairecasoimeme could clarify

pipiche38, 2019-03-06:
up

doudz, 2019-03-06:
01 8701 0003 33 d0 00 66 03
01 8701 0003 01 d0 00 54 03
where's the problem ?
0xd0 : status (0xd0 ZPS_NWK_ENUM_ROUTE_DISCOVERY_FAILED)
0x00 : network status

pipiche38, 2019-03-06:
My issue is : 
0xd0 is part of the NWK codes in the NXP Zigbee stack documentation. So from my understanding
Network Status should be 0xd0 and status 0x00

I do not understand why getting a network code in the status field.

ISO-B, 2019-03-22:
@pipiche38 You are right, there is mixup between those values. NXP documentation (JN-UG-3113 v1.5 section 7.2.2.14 ZPS_tsAfNwkRouteDiscoveryConfEvent) determinates that:
>
> - u8Status is one of the status codes from the MAC layer, detailed in Section 10.2.4
> - u8NwkStatus is one of the status codes from the NWK layer, detailed in Section 10.2.3

There is no MAC code 0x0d, but there is NWK code 0x0d. Problem is that that we can't fix this correctly since mix up happens in code that is part of library and we only have headers for this. What we can do is to switch parameters to wrong order in response. This has to be well commented to code so we can switch them back to correct order if NXP decides to fix this. 

Does this happen with every 0x8701 or is this some special case?

pipiche38, 2019-03-22:
No need to follow-up on that one. This was for my purpose and the used in my code. No need anymore

ISO-B, 2019-03-22:
@fairecasoimeme Do you think that it would be wise to change code or just correct documentation until NXP fixes code?

closed 2019-11-27 by pipiche38