fairecasoimeme/ZiGate #93: 4noks plugs support

pouicr, 2018-12-04:
Hi,

I had [4noks plugs](https://www.4-noks.com/shop/elios4you-accessories/smart-plug/?lang=en) at home and it would be great to have them supported by the Zigate!

I currently have an error at bind time :cry: 

Is it somethings feasible?

pipiche38, 2018-12-12:
Is the bind mandatory ? Did you try without ?
Fro what I see you don't need to always do a bind, and some devices doesn't allow bind

pouicr, 2018-12-13:
Well, I don't receive anything from those plugs otherwize...?

pipiche38, 2018-12-17:
If you can bind them, that mean you have pair it with Zigate
Can you share the list of Clusters ?

pouicr, 2018-12-17:
Hi, I fact the issue is happening at binding.
It is never bind...

However I read something interesting (and a bit disappointing) on 4noks plugs: 
http://www.smartofthehome.com/2014/03/4-noks-zb-connection/

It says: "4-Noks have developed their own propriety protocol  based on an older serial communication protocol called Modbus which they have re-branded ‘ZB-Connection’. This means the 4-Noks ZigBee devices are only compatible with other devices in their ZB-Connection range"....

Maybe not so open...

ISO-B, 2019-03-06:
Have you been able to pair them to ZiGate?

pouicr, 2019-03-06:
I did not retried 
But once again, I am not sure those devices are well designed for an open solution...

pipiche38, 2020-09-30:
This issue has been closed due to no activity since while!

closed 2020-09-30 by fairecasoimeme