fairecasoimeme/ZiGate #94: Add Group via ZGUI

pipiche38, 2018-12-05:
In ZGUI, 

Add Group doesn't look like the documented API in Zigate.fr 
ZGUI seems to expect Name Len Max Len and Group Name (which is not supported on Zigate)
Due to that if you try to add a device to a group via ZGUI with only the 4 first parameters you get an error
[image]

Smanar, 2018-12-05:
And if you put '0' on the 3 lasts case ?

KiwiHC16, 2018-12-05:
I'm surprised because this paprt went smoothly in my implementation based on NXP doc (zigate is a copy of it).

pipiche38, 2018-12-05:
For instance 0x8060 got more data than the only specified sqn, EP and cludterId

pipiche38, 2018-12-17:
@Smanar I do confirm puting '0' in the 3 last boxes make the trick

closed 2018-12-17 by pipiche38