fairecasoimeme/ZiGate #95: Write Attribute 0x0110 command is wrongly documented in Zigate.fr

pipiche38, 2018-12-10:
The https://zigate.fr/wiki/commandes-zigate/ is given wrong informations as regards to 0x0110 Write Attribute command 

Is missing for instance the value we want to write , and if I'm not mistaken the datatype of this value

pipiche38, 2018-12-10:
Here is what is available on ZGUI and which looks much complete.

[image]

doudz, 2018-12-10:
You're right
Extracted from ZGUI :

```
private void sendWriteAttribRequest(UInt16 u16ShortAddr, byte u8SrcEndPoint, byte u8DstEndPoint, UInt16 u16ClusterID, byte u8Direction, byte u8ManuSpecific, UInt16 u16ManuID, byte u8AttribCount, UInt16 u16AttribID, byte u8AttribType, byte[] au8Data, byte u8DataLen)
        {
            byte[] commandData = null;
            commandData = new byte[32];
            int u8Len = 0;

            // Build command payload   
            commandData[u8Len++] = 0x02; // Short address mode
            commandData[u8Len++] = (byte)(u16ShortAddr >> 8);
            commandData[u8Len++] = (byte)u16ShortAddr;
            commandData[u8Len++] = u8SrcEndPoint;
            commandData[u8Len++] = u8DstEndPoint;
            commandData[u8Len++] = (byte)(u16ClusterID >> 8);
            commandData[u8Len++] = (byte)u16ClusterID;
            commandData[u8Len++] = u8Direction;
            commandData[u8Len++] = u8ManuSpecific;
            commandData[u8Len++] = (byte)(u16ManuID >> 8);
            commandData[u8Len++] = (byte)u16ManuID;
            commandData[u8Len++] = u8AttribCount;
            commandData[u8Len++] = (byte)(u16AttribID >> 8);
            commandData[u8Len++] = (byte)u16AttribID;
            commandData[u8Len++] = u8AttribType;

            int i;
            for (i = 0; i < u8DataLen; i++)
            {
                commandData[u8Len] = au8Data[i];
                u8Len++;
            }

```

doudz, 2018-12-10:
```

<address mode: uint8_t> | Data Indication
-- | --
  |   |   | <target short address: uint16_t> |  
  |   |   | <source endpoint: uint8_t> |  
  |   |   | <destination endpoint: uint8_t> |  
  |   |   | <Cluster id: uint16_t> |  
  |   |   | <direction: uint8_t> |  
  |   |   | <manufacturer specific: uint8_t> |  
  |   |   | <manufacturer id: uint16_t> |  
  |   |   | <number of attributes: uint8_t> |  
  |   |   | <attributes data list: 

attributes data list is :
Attribute id : uint16_t
Attribute type : uint8_t
Attribute data

```

fairecasoimeme, 2018-12-10:
Salut,
En effet, j'ai corrigé
Merci !

closed 2018-12-10 by fairecasoimeme

KiwiHC16, 2018-12-10:
Was right in original NXP doc as I checked my code and is aligned to doudz comments.