fairecasoimeme/ZiGate #96: Add Group response in Zigate.fr is uncomplete

pipiche38, 2018-12-11:
The documentation of the Add group response 0x8060 is uncomplete, 

The API is returning more and especiall a Status as well as the GroupID

Obj-> ZiGate | 0x8060 | Add Group response | <Sequence number: uint8_t> | Status
-- | -- | -- | -- | --
  |   |   | <endpoint: uint8_t> |  
  |   |   | <Cluster id: uint16_t>
```

        # ZNC_BUF_U8_UPD   ( &au8LinkTxBuffer [u16Length],pCustom->uMessage.psAddGroupResponsePayload->eStatus,       u16Length );
        # ZNC_BUF_U16_UPD  ( &au8LinkTxBuffer [u16Length],pCustom->uMessage.psAddGroupResponsePayload->u16GroupId,    u16Length );

        # 0180600008a98d0100048a00014203
        # Start: 01
        # Command: 8060
        # Leght: 0008
        # Checksum: a9
        # SQN: 8d
        # EP:        01
        # ClusterID        0004
        # Status:        8a
        # GroupID        0001
        # RSSI: 42
        # End: 03
```

doudz, 2018-12-11:
yes you're right (again)

Maybe it could be useful to add the short addr in the response ?

pipiche38, 2018-12-11:
@doudz if the information is available on the Zigate side, it would be indeed helpfull. 

I'm BTW wonder why the cluster is part, can we have something else than cluster 0x0004 ?

doudz, 2018-12-12:
https://github.com/fairecasoimeme/ZiGate/pull/97

pipiche38, 2018-12-12:
Excellent, thanks !

doudz, 2018-12-12:
not tested since I can't compile the firmware...

fairecasoimeme, 2018-12-14:
Fixed in 3.0f version

closed 2018-12-14 by fairecasoimeme

pipiche38, 2019-01-08:
In the addgroupresponse, adding the Source Address is a good thing, but then the source addr must be added in order to be consisent.

doudz, 2019-01-08:
What to you mean ? Source Address or source addr is the same thing, no ?

pipiche38, 2019-01-08:
@doudz , my mistake ... no issue at all. I was under the impression that Source EP was missing. But that is not the case

doudz, 2019-01-08:
ok...