fairecasoimeme/ZiGate #100: channel !

pipiche38, 2018-12-17:
Using ZGUI I put 
0x00000000 and do Set CMSK

Then I do Network State
got 0x8009 and Channel: 11

Then I do Start NWK
Got 0x8025 and Channel is 7

Redo Network State and git the same 0x8009 with Channel: 11

What are the difference between Channel from 0x8009 and channel from 0x8025 ?

fairecasoimeme, 2018-12-17:
Hi,
exact, it's a mistake.
I'll fix that in the next firmware.
thanks

pipiche38, 2018-12-18:
@fairecasoimeme Mistake where, in ZGUI, or in one of the Command 0x0024/0x8024 or 0x0009/0X8089 ?

pipiche38, 2018-12-19:
I'm not sure I'm following you . As all the steps were done with ZGUI

(1) Set CMSK : 0x00000000 

So I can understand your point  of the 0x04 status aand so the channel is not good, but where the channel 7 came from ?

pipiche38, 2018-12-19:
@fairecasoimeme could you kindly comment where this restrictions on the channels are coming from ?
If you set other channel than 11,15,19,20,25 ZiGate set default channel like 11.

For me the ZigBee channels supported by the NXP stack are 11 to 26, and in Europe we would benefit to use those ''11,15,16,21,22' as there overall with the Wifi are limited

ISO-B, 2019-03-07:
I can't answer why channels are restricted, but restrictions are defined at ZigbeeNodeControlBridgeCoordinator.zpscfg

pipiche38, 2019-03-07:
However, it works with Channel 26 in order to get the Livolo connected. 
So I don't feel the restriction really is that one

ISO-B, 2019-03-07:
Channels available changed to version 3.0f. Currently available channels can be found:
https://github.com/fairecasoimeme/ZiGate/blob/76c328bb44a6fdd1774c86807b101c9c15edfebc/Module%20Radio/Firmware/src/ZiGate/Source/ZigbeeNodeControlBridge/ZigbeeNodeControlBridgeCoordinator.zpscfg#L405

pipiche38, 2019-03-07:
And we agree that channel 0 is interpreted as all of those one .

closed 2019-03-08 by pipiche38