fairecasoimeme/ZiGate #103: Support Aqara light bulb

spairault, 2018-12-19:
Hello,

Aqara launch few weeks ago a new zigbee light bulb. It's seems very nice.
Do you planned to support it ?

https://www.gearbest.com/smart-bulb/pp_009892485950.html?wid=1433363
https://www.youtube.com/watch?v=SoLRTNLrh-o

Thanks a lot for your hard work.

Regards,

spairault, 2018-12-19:
Great news! Thx a lot.

cpainchaud, 2019-01-10:
I have one connected to Zigate -> HomeAssistant via https://github.com/doudz/homeassistant-zigate plugin

Everything works fine BUT color temperature selection. The light bulb does not offer RGB colros but only temperature

doudz, 2019-01-11:
Maybe it's related to my homeassistant component.
Could you share the zigate.json ?

doudz, 2019-01-11:
Could you call the service "discover_device" on that bulb, I think it will solve the problem. If not try to restart HA

cpainchaud, 2019-01-11:
discover_services has triggered a hundreds of errors like this:
2019-01-11 12:44:51 DEBUG (ZiGate-Listen) [zigate] Raw packet received, b'\x01\x80\x02\x10\x02\x10\x02\x15?\x15\xee\x02\x11@\x02\x10\x03'
2019-01-11 12:44:51 DEBUG (ZiGate-Event Loop) [zigate] Dispatch ZIGATE_PACKET_RECEIVED
2019-01-11 12:44:51 DEBUG (ZiGate-Decode data) [zigate] Received response 0x8000: b'15ee0140'
2019-01-11 12:44:51 DEBUG (ZiGate-Decode data) [zigate] RESPONSE 0x8000 - Status response : status:21, sequence:238, packet_type:320, error:b'', rssi:0
2019-01-11 12:44:51 DEBUG (ZiGate-Decode data) [zigate] Dispatch ZIGATE_RESPONSE_RECEIVED
2019-01-11 12:44:51 ERROR (ZiGate-Decode data) [zigate] Command 0x0140 failed Failed (ZigBee event codes) 21 : b''
2019-01-11 12:44:51 DEBUG (Thread-13) [zigate] STATUS code to command 0x0140:21

I am now trying a restart of ha

cpainchaud, 2019-01-11:
Restart didnt help. For odd reasons I had to re-join the bulb.
After rejoining, the bulb is working (on+off) but still no option to pick a color temperature.

updates json:

```json
{
    "devices": [
        {
            "addr": "fedd",
            "discovery": "",
            "endpoints": [
                {
                    "clusters": [
                        {
                            "attributes": [
                                {
                                    "attribute": 65281,
                                    "data": "01213b0b0328150421a80105210c0006240100010000082108030a21000098212800992100009a25c8ff79009c04",
                                    "name": "battery",
                                    "unit": "V",
                                    "value": 2.875
                                }
                            ],
                            "cluster": 0
                        },
                        {
                            "attributes": [
                                {
                                    "attribute": 1288,
                                    "data": "\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"
                                },
                                {
                                    "attribute": 1283,
                                    "data": 0.0,
                                    "expire": 2,
                                    "name": "rotation",
                                    "type": "float",
                                    "unit": "\u00b0",
                                    "value": 0.0
                                },
                                {
                                    "attribute": 85,
                                    "data": "",
                                    "expire": 2,
                                    "expire_value": "",
                                    "name": "movement",
                                    "type": "str",
                                    "value": ""
                                },
                                {
                                    "attribute": 1285,
                                    "data": 1769472
                                }
                            ],
                            "cluster": 257
                        }
                    ],
                    "device": 0,
                    "endpoint": 1,
                    "in_clusters": [],
                    "out_clusters": [],
                    "profile": 0
                }
            ],
            "generictype": "",
            "info": {
                "addr": "fedd",
                "id": 2,
                "ieee": "00158d0002b716c6",
[... 275 lines cut ...]
                    "profile": 260
                }
            ],
            "generictype": "sensor",
            "info": {
                "addr": "80ed",
                "bit_field": "0100000000000010",
                "descriptor_capability": "00000000",
                "id": 1,
                "ieee": "00158d000204402b",
                "last_seen": "2019-01-11 12:56:54",
                "mac_capability": "10000000",
                "manufacturer_code": "1037",
                "power_type": 0,
                "rssi": 84,
                "server_mask": 0
            }
        }
    ],
    "groups": {},
    "scenes": {}
}```

screenshot of homeassistant:
![image](https://user-images.githubusercontent.com/6696638/51035136-50c11c00-15a9-11e9-83ae-28727370b33a.png)

pipiche38, 2019-01-11:
I'm waiting for receiving one in the next couple of days.
But it looks like we are in the Xiaomi world and it looks that Cluster 12 is used ... I got a similar informations ...

cpainchaud, 2019-01-11:
ok for the record, model number from xiaomi is : ZNLDP12LM

cpainchaud, 2019-01-14:
Doudz's patch in his library fixed my Colour Temperature issue BUT i have a new one : after a few hours/days the bulb becomes unesponsive : it wont turn on/off anymore. 

What kind of logs can I forward? I can provided Doudz's Zigate python lib output eventually....

pipiche38, 2019-01-16:
Interesting Bulb. 
Seems that Aqara is using Cluster 0000 with Attribute ff01 to send some additional information on regular basic.

Tag: 6420 give you OnOff state 00 or 01
Tag: 6520 give you the Level state 01 to FF
Tag 6621 don't know yet. It is most-likely related to the WW 

In addition there is also attribute f000 on cluster 0300

cpainchaud, 2019-01-17:
Yes @pipiche38 , it's actually the only way I can see to get state of the lamp if someone takes it manually on/off with an electric switch

pipiche38, 2019-01-17:
The tag 6621 looks strange for me, and then I need also to understand the 0x0300/f000 , which seems to be in relation with WW variation. 2 messages are sent.

Need to see the pattern. but have also some other challenges ;-)

pipiche38, 2019-03-06:
From my end this one can be close, as the Aqara builb is fully supported, no ?

pipiche38, 2019-07-14:
Aqara bulb works very well with Zigate. This should be close

closed 2019-07-15 by ISO-B